mkuf/prind #1: 404 on klippy.log

frankbags, 2021-11-25:
In fluidd or mainsail I can't download the klippy.log file, I received a 404 error.   

2021-11-25 04:28:49,537 [app.py:log_request()] - 404 GET /server/files/klippy.log?token=Q2KBS22TXR46W3O4EADRPI63NCAZQCND&date=1637814525084 (192.168.2.13) [_TRUSTED_USER_] 1.49ms

mkuf, 2021-11-25:
Hi there, 

this is to be expected, as none of the Services is configured to create Logfiles.
All Logs are to be aquired via `docker logs`.

That said, usability wise, it is probably better to have them served via Moonraker. 
This can be achieved by updating the service definitions in `docker-compose.yaml` and adding a shared log volume.

I created a branch that reconfigures Klipper and Moonraker to write logs to a common tmpfs volume. 
This prevents the underlying local storage to wear down, as I'm expecting this to be an SD-Card in most cases and purges the Logs with every Reboot of the base system. 

Have a go at it and let me know if this works for you: https://github.com/mkuf/prind/tree/klippy-logs-1

-Markus

frankbags, 2021-11-25:
Yes, Thank you. It works as expected, It will probably make debugging easier. 

I tried to add a volume for logs but did not have the time to look into docker compose files.  The moonraker log worked fine before these changes.  I understand the concerns about wearing down SD-cards but I set this up on an old Mac Mini running debian.  

I tried getting the logs using the Docker Logs  command, which works for Moonraker and the other containers but not the Klipper container.

mkuf, 2021-11-25:
> I tried to add a volume for logs but did not have the time to look into docker compose files. The moonraker log worked fine before these changes. I understand the concerns about wearing down SD-cards but I set this up on an old Mac Mini running debian.

You can simply remove `driver_opts` from the log volume if you'd like to persist your logs.

> I tried getting the logs using the Docker Logs command, which works for Moonraker and the other containers but not the Klipper container.

You mean, with this Branch you don't get any output for `docker logs klipper`, or with `main`?
With `klippy-logs-1` you won't get any logs via docker, as the klippy process is now logging to file. 
If you did not get any logs while on `main`, there might be some error in your docker installation. 

-Markus

mkuf, 2021-11-29:
Changes have been merged into main.

closed 2021-11-29 by mkuf