mkuf/prind #2: No such file or directory: 'ip

jhuizingh, 2021-12-30:
I am trying to use your containers to set up this stack on my local machine. I'm running ubuntu 20.04 LTS so I'm not using it as-is since the docker version that comes with the OS doesn't have the newer docker-compose features that you're using here.  I also can't use the traefik configuration b/c I already have another service using port 80 on the system.

I'm got a compose file started with containers for klipper and moonraker. I've worked through a couple volume mount issues, but those seem to be settled. 

Now I'm running into the error that the moonraker service can't find a command `ip`. I'd not worked with this tool before, but when I tried running it on my ubuntu server it existed.  It doesn't work in the container though. That's kind of weird because I would expect it to be part of the base container if this image worked previously.   Any idea what's going on?

```
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/components/machine.py", line 423, in parse_network_interfaces
    resp = await scmd.run_with_response(log_complete=False)
  File "/opt/moonraker/moonraker/components/shell_command.py", line 241, in run_with_response
    raise ShellCommandError(
components.shell_command.ShellCommandError: Error running shell command: '['ip', '-json', 'address']'
2021-12-30 00:00:18,544 [shell_command.py:_create_subprocess()] - shell_command: Command (ip -json address) failed
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/components/shell_command.py", line 256, in _create_subprocess
    transport, protocol = await loop.subprocess_exec(
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1652, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/local/lib/python3.10/asyncio/unix_events.py", line 207, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/local/lib/python3.10/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/local/lib/python3.10/asyncio/unix_events.py", line 799, in _start
    self._proc = subprocess.Popen(
  File "/usr/local/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'ip
```

closed 2021-12-30 by mkuf

mkuf, 2021-12-30:
Hi Jonathan, 
thanks for raising this issue. 

Moonraker has added a Method to get the Hosts IP-Address via the `ip` command in https://github.com/Arksine/moonraker/commit/b33183f60f42e134d2990a31291ae7fc5106aab6. 

As the runtime Image for Moonraker is built upon the `python:3-slim` image, it is missing the `ip` command. 
Adding the `iproute2` package to the container fixes this and brings the `ip` command.
I added this to the official image in https://github.com/mkuf/prind/commit/7d52210d25d9d5dcc8f41fe82a5fc920e7ee091b

This feature of moonraker will probably have no use when running within a container, but at least we got rid of the Stacktrace. 

-Markus

mkuf, 2021-12-30:
The Image for `moonraker:9d7baa1` (and nightly) is currently being rebuilt.
Check https://github.com/mkuf/prind/runs/4670033831 for progress.
When the build is complete, you'll have to remove your local moonraker images and re-pull from Dockerhub. 

-Markus

jhuizingh, 2021-12-31:
Thanks for the quick response and getting it taken care of! I will try the new image in the next day or two and report back whether it works for me.

jhuizingh, 2021-12-31:
It looks like moonraker is up and running without crashing any more. Thanks again!