mkuf/prind #4: Upstream Moonraker address is configurable

sbonnick, 2022-01-20:
Currently it looks like the upstream server is hardcoded to be the same host. but in a docker environment it would be ideal if the docker instance can be passed as an env the address of moonraker. for example, this would work better if your moonraker is in a independent docker and only exposed to other docker instance within the same docker-compose. you could reference the address just by its hostname:7125/api.

sbonnick, 2022-01-20:
I'm not too familiar with VUE builds, but It looks like during the mainsail build, its hardcoding the address and port - I expected to see a NGINX config in their repo somewhere similar as they have in their docs on manual install, but I'm guessing VUE is dynamically creating this at build intended for development purpose. anyways, all guessing still at this point, but there is a config file in the mainsail repo that I'm keying off of  `.env.development.local.example`,

sbonnick, 2022-01-20:
yep: https://docs.mainsail.xyz/development/standalone

looks like the address and port are being set at build time, which of course makes this a lot harder to change at run time using an environment variable in a docker-compose file. I have tried the following in my docker-compose to see if the hosting app will pick it up - which it did not, still showing "localhost" in the logs:

```
environment:
      - VUE_APP_HOSTNAME=moonraker
```

sbonnick, 2022-01-20:
Correction, you are using NGINX, however copying the artifacts VUE generates it looks like... so ya that would explain that the Host is not dynamic, since its not VUE.  

I would have to see the output of the VUE build, but one solution could be to modify the NGINX upstream apiserver if there is a ENV provided. Should be a upstreams.conf file that's generated:

```
# upstreams.conf

upstream apiserver {
    ip_hash;
    server 127.0.0.1:7125;
}

....
```

sbonnick, 2022-01-20:
according to this article, looks like it could potentially be hardcoded in the upstream file to use a ENV. if so, could be set right after build to a ENV, then in the dockerfile you assign a reasonable default. a user can then override it:

https://stackoverflow.com/a/63215308

mkuf, 2022-01-20:
Hi there, 

I'm not quite sure, what you are trying to achive.  
<br>

> for example, this would work better if your moonraker is in a independent docker and only exposed to other docker instance within the same docker-compose. you could reference the address just by its hostname:7125/api.

This is already the case, the moonraker container is accessible via its hostname within the docker network.
So any container started on the same host can call moonraker:7125.
Via traefik, moonrakers urls are accessible from outside of the docker network so that Clients outside of the docker network also can access it.

This is necessary because Mainsail/Fluidd are PWAs, which run entirely on the Client.
Therefore, your actual client (pc, phone, etc.) needs to access the moonraker api.  

https://github.com/mkuf/prind/blob/815cc3762f4b440b677529511dc1298ef849ac85/docker-compose.yaml#L87-L91

<br>

> yep: https://docs.mainsail.xyz/development/standalone
>
> looks like the address and port are being set at build time, which of course makes this a lot harder to change at run time using an environment [...]

This setup is only for development purposes.
Starting Mainsail via 'npm run serve' takes a lot of time compared to serving static files via nginx.

<br>

> Correction, you are using NGINX, however copying the artifacts VUE generates it looks like... so ya that would explain that the Host is not dynamic, since its not VUE.
>
> I would have to see the [...]

The Mainsail Image is based on Nginx, but this config snippet does not apply here, as it is solely used to serve static files and not proxy moonraker.
The actual proxying is done via Traefik (see the linked snippet above).

-Markus

sbonnick, 2022-01-20:
ah right, ya let me try to give a more specific example. in your config, Traefik is settign everything up to be from the same host and all the services are paths on that host:

moonraker:   http://HOST/websocket
mainsail:       http://HOST

in my particular case, I use the same host for many services (but this case would apply for remote hosts too). I use subdomains per service in traefik:

moonraker:   moonraker.HOST/webservice
mainsail:       mainsail.HOST
glances:        glances.HOST

The problem is in the mainsail PWA, its effectively hardcoded to look at the same host domain then the path prefix. Id ideally like to configure via a ENV the "HOST" the PWA will use to be anything where moonraker may reside. Could be HOST, moonraker.HOST, or could even reside on a separate physical machine called MOONRAKER.

Hope that adds clarity. I see all the points you made, very helpful - I was trying to revises engineer some of that to understand.

sbonnick, 2022-01-20:
However, If you aren't doing any proxying in NGINX, then it could's like this may be hardcoded right in Mainsail code. to look at a relative path */webcomponents for example.  So, perhaps there is nothing you can do here unless that's exposed via mainsail itself.

mkuf, 2022-01-20:
OK, so you'd just want to make the containers available via subdomains. 
This is already supported by mainsail. Moonraker does not have to reside on the same subdomain or host as mainsail.

Using traefik, you'd have to change the containers labels in the following manner:

```
mainsail:
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.mainsail.loadbalancer.server.port=80"
      - "traefik.http.routers.mainsail.rule=Host(`mainsail.host`)"
      - "traefik.http.routers.mainsail.entrypoints=web"

moonraker:
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.moonraker.loadbalancer.server.port=7125"
      - "traefik.http.routers.moonraker.rule=Host(`moonraker.host`)"
      - "traefik.http.routers.moonraker.entrypoints=web"
```

After your Services are online, head to mainsail.host, open the settings in the top right corner, click on 'printers' and add a new printer with the hostname 'moonraker.host' and port 80.

The default behaviour of Mainsail is to use the current hostname to look for the moonraker api endpoints. 
If your moonraker api is not hosted under the same hostname as mainsail, you'll have to manually tell it where to look for it.

One thing I noticed, you specifically wrote 'moonraker.HOST/webservice'.
Moonraker (at least in my experience) does not work under a subpath. 
This is why traefik is configured to only redirekt certain links to the moonraker container. 
https://github.com/mkuf/prind/blob/815cc3762f4b440b677529511dc1298ef849ac85/docker-compose.yaml#L90

sbonnick, 2022-01-21:
ok interesting. My Traefik is set up more or less like that, but when I navigate to my mainsail host it says it cannot connect. However it also wont let me click anything else on screen - So I cannot navigate to settings. Fluidd on the other hand does let me ignore the error and go add my own server address. I just assumed it did not have that feature. 

Tried again, latest docker image and sure enough under chrome or edge I get the connection failed error dialog and cannot interact with anything else on screen. 

[image]

So, looks like a bug for mainsail. Regardless though, do you know where mainsail intern stores a server address once entered? if is some json file, it would be convenient for re-deploys to be able to immediately reconfigure mainsail settings - including any custom server addresses.

mkuf, 2022-01-21:
Oh, I see. 
It seems that for the initial Setup, moonraker and mainsail need to reside on the same host. 
Unfortunately, I don't know anything about the inner workings of mainsail. Raising an issue in https://github.com/mainsail-crew/mainsail should be your best way forward. 

I'd guess there is already a mechanism for this in place which is used by http://my.mainsail.xyz/

-Markus

closed 2022-01-21 by mkuf

sbonnick, 2022-02-16:
Oh this is great news. Thanks for comimg.back to this issue with an update! I will give it a try

Thermionix, 2022-03-10:
finally got this working, overwrote nginx.conf for mainsail
````
$ cat docker-compose.yml 
version: '3.4'

services:
  klipper:
    image: mkuf/klipper:nightly
    container_name: klipper
    volumes:
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0:rmw
    command:
      - "-I"
      - "run/klipper.tty"
      - "-a"
      - "run/klipper.sock"
      - "cfg/printer.cfg"
      - "-l"
      - "log/klippy.log"

  moonraker:
    image: mkuf/moonraker:nightly
    container_name: moonraker
    hostname: moonraker.local
    depends_on:
      - klipper
    volumes:
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - moonraker-db:/opt/db
    command:
      - "-c"
      - "cfg/moonraker.cfg"
      - "-l"
      - "log/moonraker.log"

  mainsail:
    image: mkuf/mainsail:nightly
    container_name: mainsail
    restart: unless-stopped
    volumes:
      - ./mainsail.conf:/etc/nginx/conf.d/default.conf
    depends_on:
      - moonraker
    ports:
      - 8011:80

volumes:
  run:
    driver_opts:
      type: tmpfs
      device: tmpfs
  gcode:
  moonraker-db:

$ cat mainsail.conf 
server {
    listen 80 default_server;

    access_log /var/log/nginx/mainsail-access.log;
    error_log /var/log/nginx/mainsail-error.log;

    # disable this section on smaller hardware like a pi zero
    gzip on;
    gzip_vary on;
    gzip_proxied any;
    gzip_proxied expired no-cache no-store private auth;
    gzip_comp_level 4;
    gzip_buffers 16 8k;
    gzip_http_version 1.1;
    gzip_types text/plain text/css text/xml text/javascript application/javascript application/x-javascript application/json application/xml;

    # web_path from mainsail static files
    root /usr/share/nginx/html;

    index index.html;
    server_name _;

    # disable max upload size checks
    client_max_body_size 0;

    # disable proxy request buffering
    proxy_request_buffering off;

    location / {
        try_files $uri $uri/ /index.html;
    }

    location = /index.html {
        add_header Cache-Control "no-store, no-cache, must-revalidate";
    }

    location /websocket {
        proxy_pass http://apiserver/websocket;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection $connection_upgrade;
        proxy_set_header Host $http_host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_read_timeout 86400;
    }

    location ~ ^/(printer|api|access|machine|server)/ {
        proxy_pass http://apiserver$request_uri;
        proxy_set_header Host $http_host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Scheme $scheme;
    }
}

upstream apiserver {
    ip_hash;
    server moonraker.local:7125;
}

map $http_upgrade $connection_upgrade {
    default upgrade;
    '' close;
}
````