mkuf/prind #7: Ensure minimal ressources for klipper (and mainsail)

speendo, 2022-02-05:
I enjoy this project because it allows me to easily drive my 3D printer on my home server (which is not a dedicated raspberry pi).

However, the Klipper documentation makes clear that this setup has it's drawbacks:

> 
If you are looking to run the Klipper host software on a shared general-purpose desktop or server class machine, then note that Klipper has some real-time scheduling requirements. If, during a print, the host computer also performs an intensive general-purpose computing task (such as defragmenting a hard drive, 3d rendering, heavy swapping, etc.), then it may cause Klipper to report print errors.

https://www.klipper3d.org/FAQ.html#can-i-run-klipper-on-something-other-than-a-raspberry-pi-3

A way to overcome this issue could be to guarantee a reasonable amount of resources to the klipper container (and maybe also to the mainsail container?)

This could possibly be achieved by reserving cpu shares and memory by adding

    cpu_shares: 2048
    mem_reservation: 256m

to the services `klipper` (and `moonraker`). See https://docs.docker.com/compose/compose-file/compose-file-v2/#cpu-and-other-resources

I apology if this is not a useful idea - I used klipper with octopi for a long time on my raspberry pi and I just start moving things to my home server.

mkuf, 2022-02-06:
Hi There, 

thanks for your suggestion, that would actually be a great feature. 

Unfortunately docker does not work that way. 
Even though it's called `reservation`, these values work as upper limits for containers. 
https://docs.docker.com/config/containers/resource_constraints/

That means, even if those values are set, other processes will still be able to hog resources and slow klipper/moonraker down.
It is solely a mechanism to limit resource consumption of single containers. 

-Markus

closed 2022-02-06 by mkuf

speendo, 2022-02-07:
Thanks for your insightful response.

It's actually quite hard to understand, what the docker manual means. Maybe it's just me, but in the documentation I don't find a hint on what the container or the host actually will do, once the host has less memory than `mem_reservation` available.

I therefore searched for alternative sources of information (which of course are not necessarily correct) and found this page (https://programmer.ink/think/docker-resource-limitation.html) that provides the following explanation on `mem_reservation`:

> Set a soft limit for memory usage and perform an OOM operation if docker finds that the host is out of memory. This value must be less than the value set by --memory

This is actually wouldn't be too bad: Let's say in a certain state klipper needs 100MB. When `memory_reservation` is set to 200 MB, the container would ask the host to perform Out Of Memory Management (or in other words to kill processes that eat up too much memory). There is the slight chance that klipper itself would be killed, but in fact that's quite unlikely as the kernel prefers to kill "young" processes that use a lot of memory (https://www.kernel.org/doc/gorman/html/understand/understand016.html).

So the advantage of `memory_reservation` would be that OOM is called precociously, when klipper still has some memory space left to "grow" and not when it's too late and your print is ruined because real time operations couldn't be performed anymore.

However, it also seems that one would need to set `memory` in order for `memory_limit` to have an effect (at least that is what I understand from the docs...)

mkuf, 2022-02-08:
Your explanation sounds reasonable. 
If you test edgecases, let me know how it behaves. 

This compose file should be considered as a "one size fits all"/"keep it simple" approach. 
Anything not covered by this example is up to the individual user to configure. 

The short term benefit this would bring, might cause more headache in the future. 
Especially because Moonraker and Klipper are growing quickly and might exceed the given limits sooner rather than later. 

-Markus

speendo, 2022-02-15:
I just leave this here as a note to myself for how much memory Klipper might use.

https://www.reddit.com/r/klippers/comments/ss6len/running_klipper_on_a_256mb_ram_device/

I will try out some settings later