mkuf/prind #11: Fluidd can't reach Moonraker

Okywan, 2022-02-25:
When starting the stack without the -d option, this is the log:

> Attaching to fluidd, klipper, moonraker, traefik, ustreamer
> ustreamer  | -- INFO  [10493.930      main] -- Using internal blank placeholder
> ustreamer  | -- INFO  [10493.931      main] -- Listening HTTP on [0.0.0.0]:8080
> ustreamer  | -- INFO  [10493.931      http] -- Starting HTTP eventloop ...
> ustreamer  | -- INFO  [10493.932    stream] -- Using V4L2 device: /dev/webcam
> ustreamer  | -- INFO  [10493.932    stream] -- Using desired FPS: 30
> ustreamer  | ================================================================================
> traefik    | time="2022-02-25T21:00:24Z" level=info msg="Configuration loaded from flags."
> ustreamer  | -- INFO  [10494.020    stream] -- Device fd=8 opened
> ustreamer  | -- INFO  [10494.021    stream] -- Using input channel: 0
> ustreamer  | -- ERROR [10494.022    stream] -- Requested resolution=1280x960 is unavailable
> ustreamer  | -- INFO  [10494.022    stream] -- Using resolution: 1280x720
> ustreamer  | -- INFO  [10494.023    stream] -- Using pixelformat: MJPEG
> ustreamer  | -- INFO  [10494.024    stream] -- Using HW FPS: 30
> ustreamer  | -- ERROR [10494.024    stream] -- Device does not support setting of HW encoding quality parameters
> ustreamer  | -- INFO  [10494.024    stream] -- Using IO method: MMAP
> ustreamer  | -- INFO  [10494.037    stream] -- Requested 5 device buffers, got 5
> ustreamer  | -- INFO  [10494.270    stream] -- Capturing started
> ustreamer  | -- INFO  [10494.270    stream] -- Switching to HW encoder: the input is (M)JPEG ...
> ustreamer  | -- INFO  [10494.270    stream] -- Using JPEG quality: encoder default
> ustreamer  | -- INFO  [10494.270    stream] -- Creating pool JPEG with 1 workers ...
> ustreamer  | -- INFO  [10494.270    stream] -- Capturing ...
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> fluidd     | /docker-entrypoint.sh: /docker-entrypoint.d/ is not empty, will attempt to perform configuration
> fluidd     | /docker-entrypoint.sh: Looking for shell scripts in /docker-entrypoint.d/
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/10-listen-on-ipv6-by-default.sh
> fluidd     | 10-listen-on-ipv6-by-default.sh: info: Getting the checksum of /etc/nginx/conf.d/default.conf
> fluidd     | 10-listen-on-ipv6-by-default.sh: info: /etc/nginx/conf.d/default.conf differs from the packaged version
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/20-envsubst-on-templates.sh
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/30-tune-worker-processes.sh
> fluidd     | /docker-entrypoint.sh: Configuration complete; ready for start up
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: using the "epoll" event method
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: nginx/1.21.1
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: built by gcc 10.3.1 20210424 (Alpine 10.3.1_git20210424)
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: OS: Linux 5.10.63-v7l+
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: getrlimit(RLIMIT_NOFILE): 1048576:1048576
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker processes
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 31
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 32
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 33
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 34
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> traefik    | time="2022-02-25T21:00:27Z" level=error msg="service \"moonraker\" error: unable to find the IP address for the container \"/moonraker\": the server is ignored" providerName=docker container=moonraker-prind-3c4940dd2b1c9573739041869997c142031cd9277d8ebf8cecc9c42032914062
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()

These last 7 lines are then repeated over and over again.

mkuf, 2022-02-25:
Hi there,

The provided Log implies that moonraker can not create its logfile.

```
moonraker | Traceback (most recent call last):
moonraker | File "/opt/moonraker/moonraker/moonraker.py", line 16, in
moonraker | import logging
moonraker | File "/usr/local/lib/python3.10/logging/init.py", line 57, in
moonraker | _startTime = time.time()
moonraker | PermissionError: [Errno 1] Operation not permitted
```

Did you modify the docker-compose.yaml in regards to the log volume? 
Please share your `docker-compose.yaml` and your `moonraker.cfg` file.

-Markus

Okywan, 2022-02-25:
Only modified what is on the guide.

moonraker.cfg:

> 
> [server]
> host: 0.0.0.0
> port: 7125
> klippy_uds_address: /opt/run/klipper.sock
> 
> [file_manager]
> config_path: /opt/cfg
> 
> [machine]
> provider: none
> 
> [database]
> database_path: /opt/db
> 
> [authorization]
> trusted_clients:
>   0.0.0.0/0
> cors_domains:
>   *
> 
> [octoprint_compat]
> 
> [history]
> 
> 

docker-compose.yaml:
> ## Common Templates
> x-klipper-svc: &klipper-svc
>   image: mkuf/klipper:nightly
>   container_name: klipper
>   restart: unless-stopped
>   logging:
>     driver: none
>   command:
>     - "-I"
>     - "run/klipper.tty"
>     - "-a"
>     - "run/klipper.sock"
>     - "cfg/printer.cfg"
>     - "-l"
>     - "log/klippy.log"
> 
> x-ustreamer-svc: &ustreamer-svc
>   image: mkuf/ustreamer:nightly
>   restart: unless-stopped
>   command:
>     - "--host=0.0.0.0"
>     - "--port=8080"
>     - "--slowdown"
>     - "--device=/dev/webcam"
>     - "--resolution=1280x960"
>     - "--format=MJPEG"
>     - "--desired-fps=30"
>   depends_on:
>     - traefik
> 
> ## Service Definitions
> version: "3.4"
> services:
> 
>   ## Klippy Services
>   ##
>   klipper:
>     <<: *klipper-svc
>     volumes:
>       - ./config:/opt/cfg
>       - run:/opt/run
>       - gcode:/opt/gcode
>       - log:/opt/log
>     devices:
>       - /dev/ttyUSB0
>     profiles:
>       - fluidd
>       - mainsail
> 
>   klipper-priv:
>     <<: *klipper-svc
>     privileged: true
>     volumes:
>       - /dev:/dev
>       - ./config:/opt/cfg
>       - run:/opt/run
>       - gcode:/opt/gcode
>       - log:/opt/log
>     profiles:
>       - octoprint
> 
>   ## WebApi
>   ##
>   moonraker:
>     image: mkuf/moonraker:nightly
>     container_name: moonraker
>     restart: unless-stopped
>     logging:
>       driver: none
>     command:
>       - "-c"
>       - "cfg/moonraker.cfg"
>       - "-l"
>       - "log/moonraker.log"
>     depends_on:
>       - klipper
>       - traefik
>     volumes:
>       - ./config:/opt/cfg
>       - run:/opt/run
>       - gcode:/opt/gcode
>       - log:/opt/log
>       - moonraker-db:/opt/db
>     profiles:
>       - fluidd
>       - mainsail
>     labels:
>       - "traefik.enable=true"
>       - "traefik.http.services.moonraker.loadbalancer.server.port=7125"
>       - "traefik.http.routers.moonraker.rule=PathPrefix(`/websocket`,`/printer`,`/api`,`/access`,`/machine`,`/server`)"
>       - "traefik.http.routers.moonraker.entrypoints=web"
> 
>   ## Frontends
>   ##
>   octoprint:
>     image: octoprint/octoprint:minimal
>     container_name: octoprint
>     restart: unless-stopped
>     depends_on:
>       - klipper-priv
>       - traefik
>     privileged: true
>     volumes:
>       - /dev:/dev
>       - run:/opt/run
>       - octoprint:/octoprint
>     profiles:
>       - octoprint
>     labels:
>       - "traefik.enable=true"
>       - "traefik.http.services.octoprint.loadbalancer.server.port=5000"
>       - "traefik.http.routers.octoprint.rule=PathPrefix(`/`)"
>       - "traefik.http.routers.octoprint.entrypoints=web"
> 
>   fluidd:
>     image: cadriel/fluidd:latest
>     container_name: fluidd
>     restart: unless-stopped
>     depends_on:
>       - moonraker
>       - traefik
>     profiles:
>       - fluidd
>     labels:
>       - "traefik.enable=true"
>       - "traefik.http.services.fluidd.loadbalancer.server.port=80"
>       - "traefik.http.routers.fluidd.rule=PathPrefix(`/`)"
>       - "traefik.http.routers.fluidd.entrypoints=web"
> 
>   mainsail:
>     image: mkuf/mainsail:nightly
>     container_name: mainsail
>     restart: unless-stopped
>     depends_on:
>       - moonraker
>       - traefik
>     profiles:
>       - mainsail
>     labels:
>       - "traefik.enable=true"
>       - "traefik.http.services.mainsail.loadbalancer.server.port=80"
>       - "traefik.http.routers.mainsail.rule=PathPrefix(`/`)"
>       - "traefik.http.routers.mainsail.entrypoints=web"
> 
>   ## Accompanying Services/Infra
>   ##
> 
>   ## Webcam Service
>   ustreamer:
>     <<: *ustreamer-svc
>     container_name: ustreamer
>     devices:
>       - /dev/video0:/dev/webcam
>     labels:
>       - "traefik.enable=true"
>       - "traefik.http.services.ustreamer.loadbalancer.server.port=8080"
>       - "traefik.http.routers.ustreamer.rule=PathPrefix(`/stream`)"
>       - "traefik.http.routers.ustreamer.entrypoints=web"
> 
>   ## Loadbalancer/Proxy
>   traefik:
>     image: "traefik:v2.5"
>     container_name: "traefik"
>     hostname: "traefik"
>     command:
>       - "--accesslog"
>       - "--providers.docker=true"
>       - "--providers.docker.exposedbydefault=false"
>       - "--entrypoints.web.address=:80"
>     ports:
>       - "80:80"
>     restart: unless-stopped
>     volumes:
>         - "/var/run/docker.sock:/var/run/docker.sock:ro"
> 
> volumes:
>   run:
>     driver_opts:
>       type: tmpfs
>       device: tmpfs
>   gcode:
>   octoprint:
>   moonraker-db:
>   log:
>     driver_opts:
>       type: tmpfs
>       device: tmpfs
>

mkuf, 2022-02-25:
Please upload the configs as files to your reply. 
Pasting them into the reply field makes them unreadable.

Okywan, 2022-02-25:
> Please upload the configs as files to your reply. Pasting them into the reply field makes them unreadable.

Changed their extensions as otherwise it wouldn't let me upload them.
[moonraker.txt](https://github.com/mkuf/prind/files/8145047/moonraker.txt)
[docker-compose.txt](https://github.com/mkuf/prind/files/8145048/docker-compose.txt)
d

mkuf, 2022-02-25:
Thanks, your Configs seem to be allright. 
Stop your stack, remove the log volume and partially start your stack again.
The last command attaches to moonraker so you can read its logs.

```
docker compose --profile <profilename> down
docker volume rm prind_log
docker compose --profile <profilename> up -d klipper
docker compose --profile <profilename> up moonraker
```

Okywan, 2022-02-25:
Identical result:

> pi@raspberrypi:~/Klipper/prind$ docker volume rm prind_log
> prind_log
> pi@raspberrypi:~/Klipper/prind$ docker compose --profile fluidd up -d klipper
> [+] Running 3/3
>  ⠿ Network prind_default  Created                                                                                                                                  0.4s
>  ⠿ Volume "prind_log"     Created                                                                                                                                  0.0s
>  ⠿ Container klipper      Started                                                                                                                                  1.4s
> pi@raspberrypi:~/Klipper/prind$ docker compose --profile fluidd up moonraker
> [+] Running 3/3
>  ⠿ Container traefik    Created                                                                                                                                    0.2s
>  ⠿ Container klipper    Running                                                                                                                                    0.0s
>  ⠿ Container moonraker  Created                                                                                                                                    0.2s
> Attaching to moonraker
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> ^CGracefully stopping... (press Ctrl+C again to force)
> [+] Running 1/1
>  ⠿ Container moonraker  Stopped

mkuf, 2022-02-25:
It does not seem to be the logging volume that causes the issues. 
The only other possible cause I can think of, would be that the config directory is not writable by the user that is used within the container. 

Please execute the following commands from within your prind directory.

```
ls -an .
ls -an config/
```

Okywan, 2022-02-25:
> pi@raspberrypi:~/Klipper/prind$ ls -an .
> total 84
> drwxr-xr-x 7 1000 1000  4096 feb 25 22:35 .
> drwxr-xr-x 4 1000 1000  4096 feb 25 22:15 ..
> drwxr-xr-x 2 1000 1000  4096 feb 23 22:09 config
> drwxr-xr-x 7 1000 1000  4096 feb 23 22:09 docker
> -rw-r--r-- 1 1000 1000  4223 feb 23 22:12 docker-compose.yaml
> drwxr-xr-x 8 1000 1000  4096 feb 23 22:09 .git
> drwxr-xr-x 3 1000 1000  4096 feb 23 22:09 .github
> -rw-r--r-- 1 1000 1000    25 feb 23 22:09 .gitignore
> -rw-r--r-- 1 1000 1000 35149 feb 23 22:09 LICENSE
> -rw-r--r-- 1 1000 1000  7492 feb 23 22:09 README.md
> drwxr-xr-x 2 1000 1000  4096 feb 23 22:09 scripts
> pi@raspberrypi:~/Klipper/prind$ ls -an config/
> total 20
> drwxr-xr-x 2 1000 1000 4096 feb 23 22:09 .
> drwxr-xr-x 7 1000 1000 4096 feb 25 22:35 ..
> -rw-r--r-- 1 1000 1000 1256 feb 23 22:09 build.config
> -rw-r--r-- 1 1000 1000  271 feb 23 22:09 moonraker.cfg
> -rw-r--r-- 1 1000 1000 4035 feb 25 23:07 printer.cfg
> 

I have noticed I cannot cd to the log volume location:
pi@raspberrypi:~/Klipper/prind$ docker volume inspect prind_log

> [
>     {
>         "CreatedAt": "2022-02-25T23:22:17+01:00",
>         "Driver": "local",
>         "Labels": {
>             "com.docker.compose.project": "prind",
>             "com.docker.compose.version": "2.2.3",
>             "com.docker.compose.volume": "log"
>         },
>         "Mountpoint": "/var/lib/docker/volumes/prind_log/_data",
>         "Name": "prind_log",
>         "Options": {
>             "device": "tmpfs",
>             "type": "tmpfs"
>         },
>         "Scope": "local"
>     }
> ]
>pi@raspberrypi:~/Klipper/prind$ cd /var/lib/docker/volumes/prind_log/_data
>-bash: cd: /var/lib/docker/volumes/prind_log/_data: Permiso denegado

Which means permission denied. Is this normal under debian raspberry pi?

mkuf, 2022-02-25:
The permissions seem to be ok. While developing this stack, I ran into an issue where I had to chown the config directory to uid/gid 1000, otherwise moonraker would not start. 
From your output this already seems to be the case.

It is normal that the user `pi` cannot access these directories, as they belong to `root`

Could you try starting the stack as user `root` instead of `pi`?

Okywan, 2022-02-25:
Did start it as root, had to move compose to /usr/local/lib/docker/cli-plugins, but the outcome is the same:

> root@raspberrypi:/home/pi/Klipper/prind# docker compose --profile fluidd up
> [+] Running 5/5
>  ⠿ Container traefik    Running                                                                                                                                    0.0s
>  ⠿ Container ustreamer  Created                                                                                                                                    0.3s
>  ⠿ Container klipper    Running                                                                                                                                    0.0s
>  ⠿ Container moonraker  Created                                                                                                                                    0.0s
>  ⠿ Container fluidd     Created                                                                                                                                    0.3s
> Attaching to fluidd, klipper, moonraker, traefik, ustreamer
> ustreamer  | -- INFO  [4335.168      main] -- Using internal blank placeholder
> ustreamer  | -- INFO  [4335.169      main] -- Listening HTTP on [0.0.0.0]:8080
> ustreamer  | -- INFO  [4335.169    stream] -- Using V4L2 device: /dev/webcam
> ustreamer  | -- INFO  [4335.169    stream] -- Using desired FPS: 30
> ustreamer  | -- INFO  [4335.169      http] -- Starting HTTP eventloop ...
> ustreamer  | ================================================================================
> ustreamer  | -- INFO  [4335.254    stream] -- Device fd=8 opened
> ustreamer  | -- INFO  [4335.254    stream] -- Using input channel: 0
> ustreamer  | -- ERROR [4335.255    stream] -- Requested resolution=1280x960 is unavailable
> ustreamer  | -- INFO  [4335.256    stream] -- Using resolution: 1280x720
> ustreamer  | -- INFO  [4335.256    stream] -- Using pixelformat: MJPEG
> ustreamer  | -- INFO  [4335.256    stream] -- Using HW FPS: 30
> ustreamer  | -- ERROR [4335.256    stream] -- Device does not support setting of HW encoding quality parameters
> ustreamer  | -- INFO  [4335.256    stream] -- Using IO method: MMAP
> ustreamer  | -- INFO  [4335.268    stream] -- Requested 5 device buffers, got 5
> ustreamer  | -- INFO  [4335.494    stream] -- Capturing started
> ustreamer  | -- INFO  [4335.495    stream] -- Switching to HW encoder: the input is (M)JPEG ...
> ustreamer  | -- INFO  [4335.495    stream] -- Using JPEG quality: encoder default
> ustreamer  | -- INFO  [4335.495    stream] -- Creating pool JPEG with 1 workers ...
> ustreamer  | -- INFO  [4335.495    stream] -- Capturing ...
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1

mkuf, 2022-02-25:
We'll have to look inside the Container to check for permissions. 
As the moonraker container is restarting, we'll use klipper instead. 

`docker exec -it klipper ls -na /opt`

Okywan, 2022-02-25:
> pi@raspberrypi:~/Klipper/prind$ docker exec -it klipper ls -na /opt
> total 24
> drwxr-xr-x 1    0    0 4096 Feb 25 22:22 .
> drwxr-xr-x 1    0    0 4096 Feb 25 22:22 ..
> drwxr-xr-x 2 1000 1000 4096 Feb 23 21:09 cfg
> drwxr-xr-x 2 1000 1000 4096 Feb 25 02:55 gcode
> drwxr-xr-x 1 1000 1000 4096 Feb 23 02:34 klipper
> drwxrwxrwt 2    0    0   60 Feb 25 23:41 log
> drwxrwxrwt 2    0    0   80 Feb 25 23:41 run
> drwxr-xr-x 1 1000 1000 4096 Feb 23 02:34 venv

Okywan, 2022-02-26:
Hm, did a full upgrade of my install by switching to bullseye repo (was on buster) and doing apt upgrade. Stack starts ok now and im looking at the fluidd web!

mkuf, 2022-02-26:
I just pushed a change for Klipper and Moonraker images and rebuilt the most recent images. (https://github.com/mkuf/prind/commit/8055c187bfaff153a9d48bc9bb35bf005c21c642)
Stop your stack, prune all images and explicitly set die Tags for klipper and moonraker in docker-compose.yaml

```
mkuf/klipper:7c964e5
mkuf/moonraker:ed20223
```
```
docker compose --profile fluidd down
docker image prune -af
docker compose --profile fluidd up -d klipper
docker compose --profile fluidd up moonraker
```

mkuf, 2022-02-26:
Well, that is great timing.
You should already have received the new images I was talking about.

mkuf, 2022-02-26:
> Think might be related with this: [docker-library/python#674](https://github.com/docker-library/python/issues/674)
Great find.
Timing-wise we now can't quite pin down, wich change fixed this issue. 
Imho your change fixed the issue, as logs were working before on other installs.

closed 2022-02-26 by mkuf