mkuf/prind #9: Moonraker is starting over and over

Jodameister, 2022-02-25:
Hello,
the moon raker is starting over and over. The docker runs on a raspy 3B+ with buster.

Here are the log:

`--------------------Moonraker Log Start--------------------
config_file: cfg/moonraker.cfg
log_file: log/moonraker.log
software_version: ?
python_version: 3.10.2 (main, Jan 29 2022, 05:43:04) [GCC 10.2.1 20210110]
2022-02-25 18:39:59,921 [utils.py:setup_logging()] - config_file: cfg/moonraker.cfg
2022-02-25 18:39:59,922 [utils.py:setup_logging()] - log_file: log/moonraker.log
2022-02-25 18:39:59,922 [utils.py:setup_logging()] - software_version: ?
2022-02-25 18:39:59,922 [utils.py:setup_logging()] - python_version: 3.10.2 (main, Jan 29 2022, 05:43:04) [GCC 10.2.1 20210110]`

mkuf, 2022-02-25:
Hi there, 

I could not reproduce the described behaviour with the current nightly build, which is [0a2c190](https://hub.docker.com/layers/mkuf/moonraker/0a2c190/images/sha256-9bb07772e75919432994c4649a0e30391a06d3947b3612e35b8d892601500908?context=explore) as of writing this comment.
Please execute the following commands to find out which Image Tags you are actually using and add them to your reply:
```
docker exec klipper cat /opt/klipper/.git/HEAD | cut -c 1-7
docker exec moonraker cat /opt/moonraker/.git/HEAD | cut -c 1-7
```

Please also attach your klippy.log and moonraker.log.
These Files should be located in `/var/lib/docker/volumes/prind_log/_data/` on your machine.

-Markus

Jodameister, 2022-02-25:
Thank you for the fast reply.

docker exec klipper cat /opt/klipper/.git/HEAD | cut -c 1-7:
> 7c964e5

docker exec moonraker cat /opt/moonraker/.git/HEAD | cut -c 1-7:
> Error response from
 daemon: Container 2a62ea877bfb45cef5864497c56de4b7038b3ea29217c33ca4b0aab510a39194 is restarting, wait until the container is running

[klippy.log](https://github.com/mkuf/prind/files/8144712/klippy.log)
[moonraker.log](https://github.com/mkuf/prind/files/8144714/moonraker.log)

Jodameister, 2022-02-25:
Image from moonraker was created on 2022-02-25 03:55:45 (Info from Portainer)

Okywan, 2022-02-25:
I'm having the same behaviour, when starting the profile without -d option this is shown:

> Attaching to fluidd, klipper, moonraker, traefik, ustreamer
> ustreamer  | -- INFO  [10493.930      main] -- Using internal blank placeholder
> ustreamer  | -- INFO  [10493.931      main] -- Listening HTTP on [0.0.0.0]:8080
> ustreamer  | -- INFO  [10493.931      http] -- Starting HTTP eventloop ...
> ustreamer  | -- INFO  [10493.932    stream] -- Using V4L2 device: /dev/webcam
> ustreamer  | -- INFO  [10493.932    stream] -- Using desired FPS: 30
> ustreamer  | ================================================================================
> traefik    | time="2022-02-25T21:00:24Z" level=info msg="Configuration loaded from flags."
> ustreamer  | -- INFO  [10494.020    stream] -- Device fd=8 opened
> ustreamer  | -- INFO  [10494.021    stream] -- Using input channel: 0
> ustreamer  | -- ERROR [10494.022    stream] -- Requested resolution=1280x960 is unavailable
> ustreamer  | -- INFO  [10494.022    stream] -- Using resolution: 1280x720
> ustreamer  | -- INFO  [10494.023    stream] -- Using pixelformat: MJPEG
> ustreamer  | -- INFO  [10494.024    stream] -- Using HW FPS: 30
> ustreamer  | -- ERROR [10494.024    stream] -- Device does not support setting of HW encoding quality parameters
> ustreamer  | -- INFO  [10494.024    stream] -- Using IO method: MMAP
> ustreamer  | -- INFO  [10494.037    stream] -- Requested 5 device buffers, got 5
> ustreamer  | -- INFO  [10494.270    stream] -- Capturing started
> ustreamer  | -- INFO  [10494.270    stream] -- Switching to HW encoder: the input is (M)JPEG ...
> ustreamer  | -- INFO  [10494.270    stream] -- Using JPEG quality: encoder default
> ustreamer  | -- INFO  [10494.270    stream] -- Creating pool JPEG with 1 workers ...
> ustreamer  | -- INFO  [10494.270    stream] -- Capturing ...
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> fluidd     | /docker-entrypoint.sh: /docker-entrypoint.d/ is not empty, will attempt to perform configuration
> fluidd     | /docker-entrypoint.sh: Looking for shell scripts in /docker-entrypoint.d/
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/10-listen-on-ipv6-by-default.sh
> fluidd     | 10-listen-on-ipv6-by-default.sh: info: Getting the checksum of /etc/nginx/conf.d/default.conf
> fluidd     | 10-listen-on-ipv6-by-default.sh: info: /etc/nginx/conf.d/default.conf differs from the packaged version
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/20-envsubst-on-templates.sh
> fluidd     | /docker-entrypoint.sh: Launching /docker-entrypoint.d/30-tune-worker-processes.sh
> fluidd     | /docker-entrypoint.sh: Configuration complete; ready for start up
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: using the "epoll" event method
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: nginx/1.21.1
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: built by gcc 10.3.1 20210424 (Alpine 10.3.1_git20210424)
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: OS: Linux 5.10.63-v7l+
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: getrlimit(RLIMIT_NOFILE): 1048576:1048576
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker processes
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 31
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 32
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 33
> fluidd     | 2071/07/11 08:16:40 [notice] 1#1: start worker process 34
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()
> moonraker  | PermissionError: [Errno 1] Operation not permitted
> moonraker exited with code 1
> traefik    | time="2022-02-25T21:00:27Z" level=error msg="service \"moonraker\" error: unable to find the IP address for the container \"/moonraker\": the server is ignored" providerName=docker container=moonraker-prind-3c4940dd2b1c9573739041869997c142031cd9277d8ebf8cecc9c42032914062
> moonraker  | Traceback (most recent call last):
> moonraker  |   File "/opt/moonraker/moonraker/moonraker.py", line 16, in <module>
> moonraker  |     import logging
> moonraker  |   File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
> moonraker  |     _startTime = time.time()

These last 7 lines are then repeated over and over again.

mkuf, 2022-02-25:
Moonraker restarting is a symptom of a nonworking klipper process. 
Looking at the provided klippy.log, we can see the following error in line 7 and no further logs.

```
[...]
Unable to build C code module (error=32512)
```

This is unusual, as the C code modules are compiled when building the klipper image. 
https://github.com/mkuf/prind/blob/df3199fb9f2a9f1e78f5cb4a83de515691aa5386/docker/klipper/Dockerfile#L15-L16

I started `mkuf/klipper:7c964e5` (the same image as yours) and it worked as expected. 
I'm currently not able to reproduce this error. 

You could try to stop your stack, prune all Images and re-pull them again but this is grasping for straws.

mkuf, 2022-02-25:
@Okywan 

Your log states an error with permissions when trying to create the `moonraker.log` file.
Please raise a separate issue for your concern. 

```
moonraker | Traceback (most recent call last):
moonraker | File "/opt/moonraker/moonraker/moonraker.py", line 16, in
moonraker | import logging
moonraker | File "/usr/local/lib/python3.10/logging/init.py", line 57, in
moonraker | _startTime = time.time()
moonraker | PermissionError: [Errno 1] Operation not permitted
```

mkuf, 2022-02-25:
@Jodameister 
Are you using a 64bit OS on your Pi?

Jodameister, 2022-02-25:
32bit

mkuf, 2022-02-25:
Try to run the Image without any volumes or config.

```
docker run --rm mkuf/klipper:7c964e5
```

This should yield the following error message, without any mention of C builds.
```
INFO:root:Starting Klippy...
WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Start printer at Fri Feb 25 21:56:27 2022 (1645826187.4 7756.3)
ERROR:root:Unable to open config file cfg/printer.cfg
Traceback (most recent call last):
  File "/opt/klipper/klippy/configfile.py", line 153, in _read_config_file
    f = open(filename, 'r')
IOError: [Errno 2] No such file or directory: 'cfg/printer.cfg'
ERROR:root:Config error
Traceback (most recent call last):
  File "klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "klipper/klippy/klippy.py", line 134, in _read_config
    config = pconfig.read_main_config()
  File "/opt/klipper/klippy/configfile.py", line 274, in read_main_config
    data = self._read_config_file(filename)
  File "/opt/klipper/klippy/configfile.py", line 159, in _read_config_file
    raise error(msg)
Error: Unable to open config file cfg/printer.cfg
```

mkuf, 2022-02-25:
For the sake of testing, I removed the two lines from the klipper dockerfile that are responsible for compiling the C code in my local branch and built an image for `7c964e5`.
Running this image without any config yields the following error

```
markus@colossus:~/prind$ docker run --rm -it 11ea364c0312
INFO:root:Starting Klippy...
WARNING:root:No log file specified! Severe timing issues may result!
INFO:root:Building C code module c_helper.so
sh: 1: gcc: not found
ERROR:root:Unable to build C code module (error=32512)
Traceback (most recent call last):
  File "klipper/klippy/klippy.py", line 369, in <module>
    main()
  File "klipper/klippy/klippy.py", line 351, in main
    main_reactor = reactor.Reactor(gc_checking=True)
  File "/opt/klipper/klippy/reactor.py", line 288, in __init__
    SelectReactor.__init__(self, gc_checking)
  File "/opt/klipper/klippy/reactor.py", line 99, in __init__
    self.monotonic = chelper.get_ffi()[1].get_monotonic
  File "/opt/klipper/klippy/chelper/__init__.py", line 270, in get_ffi
    do_build_code(cmd % (destlib, ' '.join(srcfiles)))
  File "/opt/klipper/klippy/chelper/__init__.py", line 246, in do_build_code
    raise Exception(msg)
Exception: Unable to build C code module (error=32512)
```
The actual errorcode seems to be the same, but your provided log is missing the line about gcc that is used to compile the modules.
```
[...]
INFO:root:Building C code module c_helper.so
sh: 1: gcc: not found
ERROR:root:Unable to build C code module (error=32512)
[...]
```
Could you share your `printer.cfg`?

Jodameister, 2022-02-26:
Here is my printer.cfg:

```
# This file is an example config file.
# See https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md for a description of parameters.

###### Printer Config
[stepper_x]
step_pin: PC2
dir_pin: PB9
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA5
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PB8
dir_pin: PB7
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA6
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB6
dir_pin: !PB5
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
endstop_pin: ^PA7
position_endstop: 0.0
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB4
dir_pin: PB3
enable_pin: !PC3
microsteps: 16
rotation_distance: 34.406
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA2
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA0

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

###### Klipper
[virtual_sdcard]
path: /opt/gcode

[display_status]

[pause_resume]

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
# change this if you need more or less extrusion
variable_extrude: 1.0
gcode:
  ##### read E from pause macro #####
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  ##### set park positon for x and y #####
  # default is your max posion from your printer.cfg
  {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
  {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
  ##### calculate save lift position #####
  {% set max_z = printer.toolhead.axis_maximum.z|float %}
  {% set act_z = printer.toolhead.position.z|float %}
  {% if act_z < (max_z - 2.0) %}
      {% set z_safe = 2.0 %}
  {% else %}
      {% set z_safe = max_z - act_z %}
  {% endif %}
  ##### end of definitions #####
  PAUSE_BASE
  G91
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G1 E-{E} F2100
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}
  {% if "xyz" in printer.toolhead.homed_axes %}
    G1 Z{z_safe} F900
    G90
    G1 X{x_park} Y{y_park} F6000
  {% else %}
    {action_respond_info("Printer not homed")}
  {% endif %}

[gcode_macro RESUME]
description: Resume the actual running print
rename_existing: RESUME_BASE
gcode:
  ##### read E from pause macro #####
  {% set E = printer["gcode_macro PAUSE"].extrude|float %}
  #### get VELOCITY parameter if specified ####
  {% if 'VELOCITY' in params|upper %}
    {% set get_params = ('VELOCITY=' + params.VELOCITY)  %}
  {%else %}
    {% set get_params = "" %}
  {% endif %}
  ##### end of definitions #####
  {% if printer.extruder.can_extrude|lower == 'true' %}
    G91
    G1 E{E} F2100
  {% else %}
    {action_respond_info("Extruder not hot enough")}
  {% endif %}  
  RESUME_BASE {get_params}

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
  TURN_OFF_HEATERS
  CANCEL_PRINT_BASE
```

mkuf, 2022-03-05:
Is this still relevant @Jodameister ?

closed 2022-03-12 by mkuf