mkuf/prind #13: Device share between containers

zilion2000, 2022-03-05:
Hi,

I'm trying to use Octoprint frontend with Klipper.
I can connect with Klipper container to my printer, but Octoprint container can't because USB device is busy.
And if I connect Octoprint container with USB device then Klipper container is unable to connect.

So my question is: 
How to connect Klipper container with printer and use this connection in Octoprint container.

mkuf, 2022-03-05:
Hi there,

Which device are you trying to access via octoprint? 
The actual MCU (e.g. `/dev/ttyUSB0`) is used by Klipper, and Klipper itself provides a new device File, that should be consumed by Octoprint. 

So, you'll have to set up Octoprint to use `/opt/run/klipper.tty` to connect to your printer. 

-Markus

zilion2000, 2022-03-05:
Thank you for your prompt response.
`/opt/run/klipper.tty` solved my issue.

closed 2022-03-05 by mkuf