mkuf/prind #14: Warning since last update

speendo, 2022-03-08:
Since my recent pull I get these Warnings in Mainsail

[image]

I couldn't tack down the source of this message but I *think* that they are again imposed by recent changes in Moonraker.

mkuf, 2022-03-08:
I'll reopen this, as i might have a possible solution to monkey patch the running Moonraker instance to contain the  requested directories.

-Markus

mkuf, 2022-03-09:
Have a look at this Branch: [moonraker-init](/mkuf/prind/tree/moonraker-init)  
The `init-moonraker` service copies the klipper directory from the klipper service, so it should always contain the right files for your current deployment.   

Imho this is just a dirty hack. 
The directories are not essential for using moonraker and therefore only generate a warning.  
I guess there was a change in Mainsail, so that these warnings now get propagated to the UI. Which might be useful if they had any impact on the functionality of moonraker.

speendo, 2022-03-10:
Well, this sounds better than my approach: I was about to just create the directories at their default location (where they make absolutely no sense).

I will try that soon.

mkuf, 2022-03-12:
With https://github.com/fluidd-core/fluidd/pull/505 this is now also present in fluidd.

speendo, 2022-03-14:
Do you still need me to test it? It's not as easy to install as I would have to change my docker-compose file.

So, if you are convenient it doesn't break anything I would wait until this comes to upstream. However, if you need someone to test it, I can try.

mkuf, 2022-03-16:
It would be great if you could give it a try.  
Have a look at the diff here https://github.com/mkuf/prind/commit/13f31e299d9294b318920a372e3681dc3848b11a  
It's basically just an extra service and a new dependency for the moonraker service. The other lines just rename the existing init service, which is not necessary for testing this change. 

-Markus

speendo, 2022-03-17:
I just tried it out. If I didn't make a mistake, the warnings are still there. Maybe I misconfigured something?

mkuf, 2022-03-18:
Mind sharing your compose file?
Did the init-moonraker service produce any logs?
```
docker logs init-moonraker
```

Have any klipper files been copied to the moonraker container?
```
docker exec -it moonraker ls -al /opt
```

speendo, 2022-03-19:
Here's my compose file
```
~/docker-compose/3dprinter/prind2/prind$ cat docker-compose.yaml 
## Common Templates
x-klipper-svc: &klipper-svc
  image: mkuf/klipper:nightly
  container_name: klipper
  restart: unless-stopped
  logging:
    driver: none
  depends_on:
    init-config:
      condition: service_completed_successfully
  command:
    - "-I"
    - "run/klipper.tty"
    - "-a"
    - "run/klipper.sock"
    - "cfg/printer.cfg"
    - "-l"
    - "log/klippy.log"

## Service Definitions
services:

  ## Klippy Services
  ##
  klipper:
    <<: *klipper-svc
    volumes:
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
    profiles:
      - fluidd
      - mainsail

  klipper-priv:
    <<: *klipper-svc
    privileged: true
    volumes:
      - /dev:/dev
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
    profiles:
      - octoprint

  ## WebApi
  ##
  moonraker:
    image: mkuf/moonraker:nightly
    container_name: moonraker
    restart: unless-stopped
    logging:
      driver: none
    command:
      - "-c"
      - "cfg/moonraker.cfg"
      - "-l"
      - "log/moonraker.log"
    depends_on:
      init-config:
        condition: service_completed_successfully
      klipper:
        condition: service_started
      init-moonraker:
        condition: service_completed_successfully
    volumes:
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
      - moonraker-db:/opt/db
    profiles:
      - fluidd
      - mainsail
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.moonraker.loadbalancer.server.port=7125"
      - "traefik.http.routers.moonraker.rule=PathPrefix(`/websocket`,`/printer`,`/api`,`/access`,`/machine`,`/server`)"
      - "traefik.http.routers.moonraker.entrypoints=web"

  ## Frontends
  ##
  octoprint:
    image: octoprint/octoprint:minimal
    container_name: octoprint
    restart: unless-stopped
    depends_on:
      klipper-priv:
        condition: service_started
    privileged: true
    volumes:
      - /dev:/dev
      - run:/opt/run
      - octoprint:/octoprint
    profiles:
      - octoprint
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.octoprint.loadbalancer.server.port=5000"
      - "traefik.http.routers.octoprint.rule=PathPrefix(`/`)"
      - "traefik.http.routers.octoprint.entrypoints=web"

  fluidd:
    image: cadriel/fluidd:latest
    container_name: fluidd
    restart: unless-stopped
    profiles:
      - fluidd
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.fluidd.loadbalancer.server.port=80"
      - "traefik.http.routers.fluidd.rule=PathPrefix(`/`)"
      - "traefik.http.routers.fluidd.entrypoints=web"

  mainsail:
    image: mkuf/mainsail:nightly
    container_name: mainsail
    restart: unless-stopped
    profiles:
      - mainsail
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.mainsail.loadbalancer.server.port=80"
      - "traefik.http.routers.mainsail.rule=PathPrefix(`/`)"
      - "traefik.http.routers.mainsail.entrypoints=web"

  ## Accompanying Services/Infra
  ##

  ## Config dir needs to be writable by uid/gid 1000
  ## This container sets the right permissions and exits
  init-config:
    image: busybox:latest
    container_name: init-config
    volumes: [".:/prind"]
    command: chown -R 1000:1000 /prind/config

  ## Moonraker warns, if certain klipper directories are not accessible
  ## This container copies /opt/klipper from the klipper service to moonraker
  ## https://github.com/mkuf/prind/issues/14
  init-moonraker:
    image: docker:latest
    container_name: init-moonraker
    entrypoint: []
    volumes: ["/var/run/docker.sock:/var/run/docker.sock"]
    command: |
      /bin/sh -c "
        docker cp klipper:/opt/klipper /tmp
        docker cp /tmp/klipper moonraker:/opt
      "

  ## Loadbalancer/Proxy
  traefik:
    image: "traefik:v2.5"
    container_name: "traefik"
    hostname: "traefik"
    command:
      - "--accesslog"
      - "--providers.docker=true"
      - "--providers.docker.exposedbydefault=false"
      - "--entrypoints.web.address=:80"
    ports:
      - "3434:80"
    restart: unless-stopped
    volumes:
        - "/var/run/docker.sock:/var/run/docker.sock:ro"

volumes:
  run:
    driver_opts:
      type: tmpfs
      device: tmpfs
  gcode:
  octoprint:
  moonraker-db:
  log:
    driver_opts:
      type: tmpfs
      device: tmpfs

```

I also would like to provide docker logs, however, after pulling the images again, I cannot get moonraker to start:

```
$ docker-compose --profile=mainsail up
Starting init-config ... 
Starting init-moonraker ... 
Starting init-config    ... done
Starting init-moonraker ... done
klipper is up-to-date

ERROR: for moonraker  Container "be24d9379702" exited with code 1.
ERROR: Encountered errors while bringing up the project.

```

Luckily it does still work In my default configuration (without the init-containers).

mkuf, 2022-03-22:
Your compose file looks fine to me.  
I can't recreate this issue and can't pin down what is causing this behaviour for you. 

Here are a few things you could try.

**Option 1:** (preferred) Trigger the init Service again
1. Start the init-moonraker service after the stack has started
```
docker compose start init-moonraker
```
2. Inspect the init services' logs and check if klipper directory has been copied
```
docker compose logs init-moonraker
docker compose exec moonraker ls /opt
```
3. If the files have been copied, restart the moonraker service
```
docker compose restart moonraker
```

**Option 2:** Manually Copy Files  
1. Copy Files to your local disk and then into the moonraker service
```
docker compose cp klipper:/opt/klipper /tmp
docker compose cp /tmp/klipper moonraker:/opt
```
2. Restart the moonraker service
```
docker compose restart moonraker
```

---
On an unrelated note, I noticed that the `docker:dind` Image is only available for `x64` and `arm64` architectures. This would manifest in an error, if you were on a unsupported arch. I am personally using this Stack on `arm/v7` which would not be able to start with this init service.
To circumvent this, we could probably build an image that contains docker ourselves.

speendo, 2022-03-26:
hm. I use my home server which runs a amd64 architecture.

mkuf, 2022-03-26:
The options I mentioned above are meant as troubleshooting steps, as I could not reproduce your described behaviour. Manually copying the Files is not the correct long-term solution.

At the moment I would like to bother the Maintainers of the packaged software as little as possible as long as there are viable Workarounds for running their projects in Docker.

gminadak, 2022-03-26:
Sorry to jump in but i have similar problems after updating to latest docker [20.10.14](https://github.com/moby/moby/security/advisories/GHSA-2mm7-x5h6-5pvq)  today, everything worked perfectly before the update.
I also had problem with Pi-hole (irrelevant but worth mentioning)
I'm using debian (bullseye) on an old laptop.

And i have the following errors after replacing my `docker-compose.yaml` and adding `docker-compose.override.yaml`
[image]
And this with my old `docker-compose.yaml`
[image]
[docker-compose.yaml.old.txt](https://github.com/mkuf/prind/files/8356416/docker-compose.yaml.old.txt)
[docker-compose.yaml.current.txt](https://github.com/mkuf/prind/files/8356417/docker-compose.yaml.current.txt)
[docker-compose.override.yaml.current.txt](https://github.com/mkuf/prind/files/8356418/docker-compose.override.yaml.current.txt)

mkuf, 2022-03-27:
@gminadak please share your moonraker.cfg and a full moonraker log.

gminadak, 2022-03-27:
@mkuf thank you for your reply
[moonraker.cfg.txt](https://github.com/mkuf/prind/files/8358239/moonraker.cfg.txt)
[moonraker.log.txt](https://github.com/mkuf/prind/files/8358240/moonraker.log.txt)

mkuf, 2022-03-27:
@gminadak 
those Pilicykit Warnings are probably caused my a missing setting.  
Have a look at the example:
https://github.com/mkuf/prind/blob/081cfaaff51d251675a220350c6631419bd2125f/config/moonraker.cfg#L9-L10

Also, make sure to use [mkuf/moonraker:4b27e5e](https://hub.docker.com/layers/moonraker/mkuf/moonraker/4b27e5e/images/sha256-83de5a9b7119a28f1ff2a96c5215dbf154d1d465e9f4fa798ac33ce9be8929f6) or later, as described in the v0.5.0 Releasenotes

gminadak, 2022-03-27:
@mkuf by adding
[machine] 
provider: systemd_cli

The PolicyKit warnings have gone, after some researching i come across this on [GitHub](https://github.com/Arksine/moonraker/issues/349#issue-1117683218) 
but as mentioned this feature will be deprecated in the future and the correct solution is to run the 
`~/moonraker/scripts/set-policykit-rules.sh
`
I won't lie to you i tried to execute this from docker but it asked for root password for moonraker and I stopped the procedure.
I'm by no means an expert in Docker, klipper or linux.

Thank you for your help.
ps: The suppliedpath errors are still in place

mkuf, 2022-03-27:
Thanks for pointing this out. I did not consider this when implementing the features for v0.5.0.
I'll have to look into handling this with dbus and policykit to be compatible with future moonraker releases, but this will be a separate issue.

Did you test the moonraker-init branch to get rid of the directory warnings, @gminadak?

gminadak, 2022-03-27:
@mkuf 
Just now but with no luck

> gminadak@linux-server:~/prind-init$ docker-compose --profile fluidd up
> Starting init-moonraker ...
> ustreamer is up-to-date
> Starting init-config    ...
> Starting init-moonraker ... done
> Starting init-config    ... done
> klipper is up-to-date
> 
> ERROR: for moonraker  Container "c25c9310faed" exited with code 1.
> ERROR: Encountered errors while bringing up the project.
> gminadak@linux-server:~/prind-init$

gminadak, 2022-03-28:
@mkuf 

> gminadak@linux-server:~/prind-init$ uname -a
> Linux linux-server 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux
> 
> gminadak@linux-server:~/prind-init$ docker logs init-moonraker
> 
> Error: No such container:path: moonraker:/
> Error: No such container:path: moonraker:/
> gminadak@linux-server:~/prind-init$

mkuf, 2022-03-28:
The errors in `init-moonraker`s logs indicate that the containers are not available.
From your output, it seems that the moonraker container is missing at the time the init container tries to copy the files.

```
markus@colossus:~/prind$ docker ps -a
CONTAINER ID   IMAGE                           COMMAND                   CREATED              STATUS                          PORTS                               NAMES
908b26fae6a8   mkuf/moonraker:nightly          "/opt/venv/bin/pytho…"    About a minute ago   Up About a minute               7125/tcp                            moonraker
34204551e883   docker:latest                   "/bin/sh -c '\n  set …"   About a minute ago   Exited (0) About a minute ago                                       init-moonraker
5832ca2cadf6   mkuf/klipper:nightly            "/opt/venv/bin/pytho…"    2 minutes ago        Up 2 minutes                                                        klipper
ee652c8071a9   busybox:latest                  "chown -R 1000:1000 …"    3 minutes ago        Exited (0) About a minute ago                                       init-config
5511e9b5ab99   cadriel/fluidd:latest           "/docker-entrypoint.…"    11 minutes ago       Up 11 minutes                   80/tcp                              fluidd
32901bf69963   simulavr                        "klipper/scripts/avr…"    11 minutes ago       Up 11 minutes                                                       simulavr
5510e44de016   traefik:v2.5                    "/entrypoint.sh --ac…"    11 minutes ago       Up 11 minutes                   0.0.0.0:80->80/tcp, :::80->80/tcp   traefik
96337df5dd3a   busybox:latest                  "chown -R 1000:1000 …"    11 minutes ago       Exited (0) 8 minutes ago                                            init
```

Which docker compose version are you using?

gminadak, 2022-03-28:
@mkuf 

gminadak@linux-server:~/prind-init$ docker-compose version
docker-compose version 1.29.2, build 5becea4c
docker-py version: 5.0.0
CPython version: 3.7.10
OpenSSL version: OpenSSL 1.1.0l  10 Sep 2019

mkuf, 2022-03-28:
Thats quite old. v1 might handle container creation in a different way, so that moonraker does not exist at the time we try to copy files into it.
Please give it a try with v2, as mentioned in the Readme: https://github.com/mkuf/prind#getting-started

gminadak, 2022-03-28:
I'm still wondering how did i miss this step.
Thank you for your help, the problem is now solved. 
Worth to mention that the main version still have this problem even after the update.

gminadak@linux-server:~/prind-init$ docker-compose version
Docker Compose version v2.3.4

gminadak@linux-server:~/prind-init$ docker-compose --profile fluidd up -d

[+] Running 9/9
 ⠿ Network prind-init_default  Created                                                                                                                   0.2s
 ⠿ Volume "prind-init_run"     Created                                                                                                                   0.0s
 ⠿ Container fluidd            Started                                                                                                                   4.4s
 ⠿ Container traefik           Started                                                                                                                   4.8s
 ⠿ Container ustreamer         Started                                                                                                                   4.7s
 ⠿ Container init-config       Exited                                                                                                                    3.6s
 ⠿ Container init-moonraker    Exited                                                                                                                   15.2s
 ⠿ Container klipper           Started                                                                                                                   4.1s
 ⠿ Container moonraker         Started                                                                                                                  14.9s
gminadak@linux-server:~/prind-init$ docker ps -a

> CONTAINER ID   IMAGE                             COMMAND                   CREATED             STATUS                       PORTS                                                                                                                                       NAMES
> 246f3474b6cf   mkuf/moonraker:nightly            "/opt/venv/bin/pytho…"    20 seconds ago      Up 5 seconds                 7125/tcp                                                                                                                                    moonraker
> f553409ca5e9   mkuf/klipper:nightly              "/opt/venv/bin/pytho…"    20 seconds ago      Up 15 seconds                                                                                                                                                            klipper
> fdc98100f47d   docker:latest                     "/bin/sh -c '\n  dock…"   21 seconds ago      Exited (0) 6 seconds ago                                                                                                                                                 init-moonraker
> d0d87b42086c   mkuf/ustreamer:nightly            "/opt/ustreamer/ustr…"    21 seconds ago      Up 16 seconds                8080/tcp                                                                                                                                    ustreamer
> eb01f16469e5   traefik:v2.5                      "/entrypoint.sh --ac…"    21 seconds ago      Up 16 seconds                0.0.0.0:81->80/tcp, :::81->80/tcp                                                                                                           traefik
> 15296a5827df   cadriel/fluidd:latest             "/docker-entrypoint.…"    21 seconds ago      Up 16 seconds                80/tcp                                                                                                                                      fluidd
> 7011fdfb6f4e   busybox:latest                    "chown -R 1000:1000 …"    21 seconds ago      Exited (0) 18 seconds ago                                                                                                                                                init-config

mkuf, 2022-03-29:
Great, these changes will be merged into main soon. 
Thanks for testing this @gminadak 

@speendo
I noticed you use the `docker-compose` command instead of `docker compose`. Are you also on compose v1?

mkuf, 2022-03-30:
I gave this a try on my printer, which is equipped with a fairly old udoo quad.  
Thanks to its very limited computing power paired with a micro-sd, starting the stack was painfully slow.
The init-moonraker service took almost two minutes to complete. This is way to long, considering it is executed every time the stack is brought up.

So I went ahead and mounted `/dev/null` in those places where moonraker expects a directory.  
This renders the docs and example endpoints from moonraker useless, but gets rid of the Warnings.

commit 78f39da9 (2022-03-30): Create Dummy directories for Moonraker

Fixes #14