mkuf/prind #17: Broken webcam config

flynny75, 2022-04-16:
Fluidd & Mailsail can not serve the ustreamer stream in the provided configuration. Mailsail & Fluidd are reaching out to '/webcam', but the traefik path prefix is '/stream'.

Changing the path prefix to '/webcam' is not enough, as the '/webcam' path reaches ustreamer, which doesn't know what to do with it. Changing the default ustreamer service config to:

  ustreamer:
    <<: *ustreamer-svc
    container_name: ustreamer
    devices:
      - /dev/video0:/dev/webcam
    ports:
      - "8080:8080"
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.webcam.loadbalancer.server.port=8080"
      - "traefik.http.routers.webcam.rule=PathPrefix(`/webcam`)"
      - "traefik.http.routers.webcam.entrypoints=web"
      - "traefik.http.middlewares.webcam.stripprefix.prefixes=/webcam"
      - "traefik.http.routers.webcam.middlewares=webcam"

fixes the issue, and Mailsail and serve the first webcam. I can submit a PR for this if required.

mkuf, 2022-04-18:
Hi There, 

thanks for mentioning this. 

Just to clarify - the default ustreamer Service is not broken.  
Moonraker as well as Fluidd are able to display the Stream that is served via `/stream`, when set up to do so via their respective Settings. 

I'm okay with exposing ustreamer as a whole via a subpath as you suggest, instead of just the `/stream` endpoint.  
From there, Users have the freedom to use ustreamers full functionality or rely on the default paths set in the various Frontends. 

-Markus

closed 2022-04-18 by mkuf

flynny75, 2022-04-18:
Ah, I see my mistake. I had only attempted to reconfigure Mainsail's `URL Stream` setting, not its `URL Snapshot` setting. It seems Mainsail is using the snapshot URL rather than the stream URL, which I never changed from the default `/webcam?action=snapshot`. I guess that also explains my poor webcam framerate.

I am still new to the klipper environment, thank you for your efforts with this repository. This is the third installation / deployment I have tried, and is by far the easiest

commit 27e33d5d (2022-04-18): Serve ustreamer via subpath

Resolves #17