mkuf/prind #19: mcu container in docker-compose cannot find unknown in $PATH

tkodev, 2022-05-08:
I'm looking to put the mcu build process in a simple docker-compose:
```
version: "3.9"

services:
  app:
    image: "mkuf/klipper:722ad4a-mcu"
    restart: "always"
    stdin_open: true
    tty: true
    depends_on:
      - "perms"
    devices:
      - "/dev/serial/by-id/usb-Klipper_stm32f429xx_3C0032001350304738313820-if00:/dev/ttyUSB0"
    volumes:
      - "$PWD/../../configs/klipper-mcu/build.cfg:/opt/klipper/.config"
      - "$PWD/../../configs/klipper-mcu/out:/opt/klipper/out"
    command:
      - "bash -c 'cd /opt/klipper; make menuconfig && make && make flash'"

  perms:
    image: "busybox:latest"
    volumes:
      - "$PWD/../../configs/klipper-mcu/build.cfg:/opt/klipper/.config"
    command: "chown 1000:1000 /opt/klipper/.config"
```

As you can see, it's the same as running the `docker run` command in the README.md, but instead I run it with `docker-compose up`. The `stdin_open` and `tty` acts like `-it` flag for `docker run`.

However. I'm unable to run this container. This is the error message returned 

````
Starting klipper-mcu_perms_1 ... done
Starting klipper-mcu_app_1 ... error

ERROR: for klipper-mcu_app_1  Cannot start service app: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/bin/bash -c 'cd /opt/klipper; make menuconfig && make && make flash'": stat /bin/bash -c 'cd /opt/klipper; make menuconfig && make && make flash': no such file or directory: unknown

ERROR: for app  Cannot start service app: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "/bin/bash -c 'cd /opt/klipper; make menuconfig && make && make flash'": stat /bin/bash -c 'cd /opt/klipper; make menuconfig && make && make flash': no such file or directory: unknown
ERROR: Encountered errors while bringing up the project.
```

Any idea why this would happen?

tkodev, 2022-05-08:
I've solved this issue by removing the extra quotes inside the command prop, as well as splitting the lines into each argument
```
    command:
      - "bash"
      - "-c"
      - "cd /opt/klipper && make menuconfig && make && make flash"
```

tkodev, 2022-05-08:
Also I had to run it via `docker-compose run app bash -c "cd /opt/klipper && make menuconfig && make && make flash"` instead of just `docker-compose up`. Not sure why it couldn't capture the std input.

closed 2022-05-08 by tkodev

mkuf, 2022-05-08:
Hi There, 

this is a working compose file for building the mcu code. See the further explanations on how you should set `command`.

```yaml
services:
  mcu:
    image: mkuf/klipper:nightly-mcu
    volumes:
      - ./config/build.config:/opt/klipper/.config
      - ./out:/opt/klipper/out
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```
**Option 1:**
If you need to make changes to your build config, you'll have to run `make menuconfig`, therefore the service needs a tty.
Add this to the above snippet.
```
tty: true
command: bash -c "cd /opt/klipper; make menuconfig && make && make flash"
```
With this, your service _could_ be brought up with `docker compose up`, but (!) you won't be able to interact with it, as the `up` command does not connect your current session the the service.
Therefore, you'll have to use `docker compose run mcu`.

**Option 2:**
If your build config is already set and does not need to be changed, you're able to skip `make menuconfig` and set the command like this. Therefore `tty: true` is also not requrired. Add this to the above snippet.
```
command: bash -c "cd /opt/klipper; make && make flash"
```
With this, you'll be able to bring up the service via 'docker compose up mcu' to build the mcu code.

---
In both cases, the defined service exits after the operation is completed. 
You should remove the `restart: always` directive, as this would cause the container to be restarted, when the command is completed and therefore would build the code and flash your board over and over again.

Also, make sure to use the `depends_on` statements in their long form, otherwise the default behaviour is to wait for the container to be started. In the case of the `init` (or in your case `perm`) we require the service to be completed before starting other services.

-Markus

tkodev, 2022-05-08:
Thank you for the detailed follow up! Much appreciated