mkuf/prind #22: Multiple printers

appleimperio, 2022-06-02:
is there a way to add multiple printers?

mkuf, 2022-06-03:
Hi, 

you'll have to create a `klipper` and `moonraker` service for each printer that you'd like to run. Each printer also needs separate docker volumes for storing gcode, etc.

Each moonraker Service should be hosted under a separate subdomain or port.
It is probably the easiest to add portbindings directly to the moonraker services instead of proxying them via traefik.

The Webfrontends (mainsail/fluidd) do not need to be duplicated, as they can accomodate multiple printers.

---

Alternatively, you could start the whole stack multiple times.
Clone prind into a different subdirectory and reconfigure the portbindings for `traefik` in your `docker-compose.override.yaml`.
You can then access all Services via a different Port (in this case 81). Keep using different Ports for each printer that you'd like to run.
```yaml
services:
  traefik:
    ports:
      - 81:80
```
This comes with the Downside that supporting Services such as traefik or the web frontends run multiple times. This requires more resources but is probably the easiest way if you do not want to deal with writing compose files yourself.

-Markus

appleimperio, 2022-06-04:
Thanks for the explanation. I will try it and see how it goes.

closed 2022-06-04 by mkuf

mkuf, 2022-07-08:
@appleimperio you might want to have a look at https://github.com/mkuf/prind/issues/28#issuecomment-1179405557, if you're still interested in this. 

-Markus