mkuf/prind #23: ability to build klipper firmware

ciekawy, 2022-06-04:
I'm super excited to see this project leveraging docker. I'm only staring with klipper and just managed to use klipper image found here to run `./scripts/debian-install.sh` - I think it may be worth to support optional pre-built image, or at least to install `sudo` and `systemd` required by `make` process.

mkuf, 2022-06-04:
Hi there, 

the klipper image intentionally does not ship with any build tools to keep its size low.
If you'd like to build the MCU code, you'll have to use the klipper Images tagged with `*-mcu`.  
See https://github.com/mkuf/prind#building-mcu-code for further Info. 

-Markus

closed 2022-06-04 by mkuf

ciekawy, 2022-06-04:
the `building-mcu-code` section was exactly what I was missing. That absolutely makes sense. Probably too much information at once and I just overlooked that part.