mkuf/prind #26: Port is invalid

truong0vanchien, 2022-06-13:
Hi Mkuf, I got a issue after deploying octoprint on docker.
There is no /tmp/printer on Serial port when creating a new additional port.
Am I missing something?

truong0vanchien, 2022-06-13:
Hi, do I need to manually create a home directory and tmp directory in docker volume? Since Octoprint in docker cannot see printer.cfg and log when starting.

truong0vanchien, 2022-06-14:
> From [a1a7f84#commitcomment-76006795](https://github.com/mkuf/prind/commit/a1a7f84#commitcomment-76006795)
> 
> > Hi Mkuf, thanks for your effort. But how can I deal about printer.cfg and klippy.log? Yesterday, I had to create a printer.cfg in Octoprint at first, then I looked up this file in the docker directory. After locating the root folder, I created a klippy.log in tmp folder. Is there any ways to work on that? Thanks in advance.
> 
> @truong0vanchien Your config is stored within the `config` Directory of this repository and is then mounted into the Containers at runtime. So your personal config goes into these files, as stated in [Configuring Klipper/Moonraker](https://github.com/mkuf/prind#configuring-klippermoonraker)
> 
> There is no need for you to copy any files into containers. klippy.log is accessable on your host via the `prind_log` docker volume. I don't think I can follow you on why you would need a printer.cfg within the octoprint container, could you elaborate on this?
> 
> -Markus

Hi Mkuf, github took me a while to find your comment via email.

As you were saying that the config in this repo will automatically be mounted to config in the Octoprint, weren't you?

But as I remember when deploying Octoprint, it kept saying "Klippy.log" not found, "printer.cfg" not found, "/tmp/printer" not found. So I had to manually create a printer.cfg via Octoprint in order to locate the root folder in Docker directory, from that point I created /tmp/Klipper.log. Then there was no alert about "Klippy.log" and "printer.cfg" but still got no "/tmp/printer".

If you are right, I think it should be no alert about "printer.cfg" at the beginning. Anyways, I think I will test this update later on.

Thanks for your support.

mkuf, 2022-06-14:
> As you were saying that the config in this repo will automatically be mounted to config in the Octoprint, weren't you?

Yes, this is what prind is all about. A turnkey solution for your 3d printers software.

 
> But as I remember when deploying Octoprint, it kept saying "Klippy.log" not found, "printer.cfg" not found, "/tmp/printer" not found. So I had to manually create a printer.cfg via Octoprint in order [...]

Maybe you installed a plugin that requires these files? Octoprint itself does not know or require any of them. Please open a new issue if you encounter this behaviour again and attach a logfile. 

-Markus

truong0vanchien, 2022-06-14:
> Yes, this is what prind is all about. A turnkey solution for your 3d printers software.
>
>
> Maybe you installed a plugin that requires these files? Octoprint itself does not know or require any of [...]

Hi Markus, sure, I will issue the new ticket when meeting this again.
Since now I am fine with your Fluidd, I will try Octoprint later in the near future.
Thanks and regards.

commit a1a7f84c (2022-06-13): octoprint: add basic config
  * adds config/octoprint.yaml
  * sets the serial device for octoprint
  * sets the stream and snapshot urls for octoprint

resolves #26
commit dc5c059a (2022-06-13): octoprint: add basic config
  * adds config/octoprint.yaml
  * sets the serial device for octoprint
  * sets the stream and snapshot urls for octoprint

resolves #26