mkuf/prind #32: Where is /tmp/printer?

kerta1n, 2022-07-27:
I'm running Octoprint in a separate container, on a separate port so that there is no interference, but when I attach to the klipper container, there is nothing in /tmp (using Mainsail profile currently). Do I have to actually switch my profile to Octoprint? The only reason I need to have both running is because it seems to be easier to run Obico (TSD) in Octoprint rather than Mainsail/Fluidd.

mkuf, 2022-07-27:
Hi there, 

You're looking for `/opt/run/klipper.tty`. Be sure to start the Klipper and Octoprint Container in privileged mode and mount `/dev` in both of them. Have a look at the Docs for the Klipper Image: https://github.com/mkuf/prind/blob/main/docker/klipper/README.md  

You can also refer to the docker-compose file of this repo on how to set up klipper and octoprint to work together.
Klipper: https://github.com/mkuf/prind/blob/main/docker-compose.yaml#L29-L39  
Octoprint: https://github.com/mkuf/prind/blob/main/docker-compose.yaml#L77-L90   
Octoprint Config: https://github.com/mkuf/prind/blob/main/config/octoprint.yaml#L1-L3

-Markus

closed 2022-07-27 by mkuf

kerta1n, 2022-07-27:
If I attempt to run both Mainsail and Octoprint using prind's docker compose file, container "klipper" crashes as "klipper-priv" has primary access to the dev folder (I need to be able to run both). I tried mounting all of the /opt/run directories for all containers that used it in the /tmp directory on host, but Klipper again crashed and for reasons I don't know why (configured logging driver does not support reading). Is there another way to pass klipper.tty to Octoprint? MY Octoprint's compose file is under ~/docker/octoprint, and the compose file for prind is in ~/docker/prind.

mkuf, 2022-07-28:
I don't think it was clear that I was referring to your setup where octoprint is run outside of prind and not running both Frontends with prind. As you discovered, this does not work because klipper is set up differently for each frontend.  
  
### Scenario 1
You'd configure the `klipper` service in your `docker-compose.override.yaml` to run in privileged mode and also mount `/dev` as well as all the other volumes and start the stack with either the `mainsail` or `fluidd` profile.
Then you set up your external compose file for octoprint and refer the volumes of prind.  

```yaml
[...]

## Add your personal config here
services:
  klipper:
    privileged: true
    volumes:
      - /dev:/dev
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
[...]
```
### Scenario 2
It would problably be the cleanest setup if you add an ocotprint service to your override file, instead of running octoprint via a second docker compose stack. Something like this:

```yaml
[...]

## Add your personal config here
services:
  klipper:
    privileged: true
    volumes:
      - /dev:/dev
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log

  octoprint-kerta1n:
    image: octoprint/octoprint:minimal
    restart: unless-stopped
    depends_on:
      klipper:
        condition: service_started
    privileged: true
    ports:
      - 3000:3000
    volumes:
      - /dev:/dev
      - run:/opt/run
      - octoprint:/octoprint
      - ./config/octoprint.yaml:/octoprint/octoprint/config.yaml
[...]
```

kerta1n, 2022-08-13:
After trying this, it appears that klipper.tty is attached to my SSH terminal: for example, telling Octoprint to connect to /opt/run/klipper.tty sends the GCode `N0 M110 N0*125` into my SSH session. Do I need to run this stack as another user? Or did I do something wrong (Octoprint serves its webpage on port 5000 not 3000 so I disabled that)? I used scenario 2 as a reference.

kerta1n, 2022-08-13:
Override file for reference:
```## Ustreamer base Service
x-ustreamer-svc: &ustreamer-svc
  image: mkuf/ustreamer:nightly
  restart: unless-stopped
  command:
    - "--host=0.0.0.0"
    - "--port=8080"
    - "--slowdown"
    - "--device=/dev/webcam"
    - "--resolution=1920x1080"
    - "--format=MJPEG"
    - "--desired-fps=30"

## Add your personal config here
services:
  klipper:
    privileged: true
    volumes:
      - /dev:/dev
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
    devices:
      #- /dev/ttyUSB0:/dev/ttyUSB0
      - /dev/serial/by-id/usb-Klipper_stm32g0b1xx_090025000350415339373620-if00:/dev/serial/by-id/usb-Klipper_stm32g0b1xx_090025000350415339373620-if00

  webcam:
    <<: *ustreamer-svc
    container_name: webcam
    devices:
      - /dev/v4l/by-id/usb-GENERAL_2K_HD_Camera-video-index0:/dev/webcam
    labels:
      - "traefik.enable=true"
      - "traefik.http.services.webcam.loadbalancer.server.port=8080"
      - "traefik.http.routers.webcam.rule=PathPrefix(`/webcam`)"
      - "traefik.http.routers.webcam.entrypoints=web"
      - "traefik.http.middlewares.webcam.stripprefix.prefixes=/webcam"
      - "traefik.http.routers.webcam.middlewares=webcam"
  mainsail:
    volumes:
      - ./gcode:/opt/gcode

  octoprint-kerta1n:
    container_name: octoprint
    image: octoprint/octoprint:minimal
    restart: unless-stopped
    depends_on:
      klipper:
        condition: service_started
    privileged: true
    ports:
      - 3000:3000
      - 5000:5000
    volumes:
      - /dev:/dev
      - run:/opt/run
      - octoprint:/octoprint
      - ./config/octoprint.yaml:/octoprint/octoprint/config.yaml
```

mkuf, 2022-08-26:
> After trying this, it appears that klipper.tty is attached to my SSH terminal: for example, telling Octoprint to connect to /opt/run/klipper.tty sends the GCode `N0 M110 N0*125` into my SSH session. Do I need to run this stack as another user? Or did I do something wrong (Octoprint serves its webpage on port 5000 not 3000 so I disabled that)? I used scenario 2 as a reference. Edit: I entered the klipper container and found nothing in the /tmp directory. I didn't find anything about changing /tmp/printer, does the Dockerfile make a symlink between /tmp/printer and /opt/run/klipper.tty?

Running the stack as another user won't make any difference, as users are handled within the containers.  
Using port 5000 is correct, I had the wrong Port in mind when writing the example.  
`/tmp/printer` does not exist, as klipper is started with `-I run/klipper.tty`, as described in the [Docs](https://github.com/mkuf/prind/tree/main/docker/klipper#defaults)

I have no Idea what is happening with the Output appearing on your Terminal.  
`klipper.tty` is a symlink setup by klippy that points to a device within `/dev`. That is why the containers need to run in privileged mode and mount the hosts `/dev`. You could check klippy logs if there is anything about symlink creation and also check the actual device file.   
Also, I'm not sure if the symlink even works when moonraker is attached to the socket.  
You may want to check if the setup works with either of the applications and then go from there.