mkuf/prind #33: Klipppy logs state that serial device can't be found

adrianocr, 2022-08-31:
Here is the klippy.log:

```console
Starting Klippy...
Args: ['klipper/klippy/klippy.py', '-I', 'run/klipper.tty', '-a', 'run/klipper.sock', 'cfg/printer.cfg', '-l', 'log/klippy.log']
Git version: '?'
CPU: 4 core Intel(R) Core(TM) i5-4590 CPU @ 3.30GHz
Python: '2.7.18 (default, Apr 20 2020, 19:34:11) \n[GCC 8.3.0]'
Start printer at Wed Aug 31 01:51:26 2022 (1661910686.5 8174.5)
===== Config file =====
[mcu]
serial = /dev/ttymxc3

[printer]
kinematics = cartesian
max_velocity = 500
max_accel = 3000
max_z_velocity = 25
max_z_accel = 30

[virtual_sdcard]
path = /opt/gcode

[display_status]

[pause_resume]

[gcode_macro PAUSE]
description = Pause the actual running print
rename_existing = PAUSE_BASE
variable_extrude = 1.0
gcode = 
	{% set E = printer["gcode_macro PAUSE"].extrude|float %}
	{% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
	{% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
	{% set max_z = printer.toolhead.axis_maximum.z|float %}
	{% set act_z = printer.toolhead.position.z|float %}
	{% if act_z < (max_z - 2.0) %}
	{% set z_safe = 2.0 %}
	{% else %}
	{% set z_safe = max_z - act_z %}
	{% endif %}
	PAUSE_BASE
	G91
	{% if printer.extruder.can_extrude|lower == 'true' %}
	G1 E-{E} F2100
	{% else %}
	{action_respond_info("Extruder not hot enough")}
	{% endif %}
	{% if "xyz" in printer.toolhead.homed_axes %}
	G1 Z{z_safe} F900
	G90
	G1 X{x_park} Y{y_park} F6000
	{% else %}
	{action_respond_info("Printer not homed")}
	{% endif %}

[gcode_macro RESUME]
description = Resume the actual running print
rename_existing = RESUME_BASE
gcode = 
	{% set E = printer["gcode_macro PAUSE"].extrude|float %}
	{% if 'VELOCITY' in params|upper %}
	{% set get_params = ('VELOCITY=' + params.VELOCITY)  %}
	{%else %}
	{% set get_params = "" %}
	{% endif %}
	{% if printer.extruder.can_extrude|lower == 'true' %}
	G91
	G1 E{E} F2100
	{% else %}
	{action_respond_info("Extruder not hot enough")}
	{% endif %}
	RESUME_BASE {get_params}

[gcode_macro CANCEL_PRINT]
description = Cancel the actual running print
rename_existing = CANCEL_PRINT_BASE
gcode = 
	TURN_OFF_HEATERS
	CANCEL_PRINT_BASE
=======================
Config error
Traceback (most recent call last):
  File "klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "klipper/klippy/klippy.py", line 143, in _read_config
    m.add_printer_objects(config)
  File "/opt/klipper/klippy/toolhead.py", line 599, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/opt/klipper/klippy/toolhead.py", line 256, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/opt/klipper/klippy/kinematics/cartesian.py", line 142, in load_kinematics
    return CartKinematics(toolhead, config)
  File "/opt/klipper/klippy/kinematics/cartesian.py", line 16, in __init__
    for n in 'xyz']
  File "/opt/klipper/klippy/stepper.py", line 422, in LookupMultiRail
    default_position_endstop, units_in_radians)
  File "/opt/klipper/klippy/stepper.py", line 302, in __init__
    self.add_extra_stepper(config)
  File "/opt/klipper/klippy/stepper.py", line 372, in add_extra_stepper
    stepper = PrinterStepper(config, self.stepper_units_in_radians)
  File "/opt/klipper/klippy/stepper.py", line 238, in PrinterStepper
    step_pin = config.get('step_pin')
  File "/opt/klipper/klippy/configfile.py", line 58, in get
    note_valid=note_valid)
  File "/opt/klipper/klippy/configfile.py", line 33, in _get_wrapper
    % (option, self.section))
Error: Option 'step_pin' in section 'stepper_x' must be specified
webhooks client 139881473585632: New connection
webhooks client 139881473585632: Client info {'program': 'Moonraker', 'version': '?'}
webhooks client 139881473585632: Disconnected
Restarting printer
Start printer at Wed Aug 31 01:51:56 2022 (1661910716.6 8204.6)
===== Config file =====
[stepper_x]
step_pin = PC2
dir_pin = PB9
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA5
position_endstop = 0
position_max = 235
homing_speed = 50

[stepper_y]
step_pin = PB8
dir_pin = PB7
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA6
position_endstop = 0
position_max = 235
homing_speed = 50

[stepper_z]
step_pin = PB6
dir_pin = !PB5
enable_pin = !PC3
microsteps = 16
rotation_distance = 8
endstop_pin = probe:z_virtual_endstop
position_max = 250

[extruder]
max_extrude_only_distance = 100.0
step_pin = PB4
dir_pin = PB3
enable_pin = !PC3
microsteps = 16
rotation_distance = 32.8542094456
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PA1
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC5
control = pid
pid_kp = 21.527
pid_ki = 1.063
pid_kd = 108.982
min_temp = 0
max_temp = 250

[heater_bed]
heater_pin = PA2
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC4
control = pid
pid_kp = 54.027
pid_ki = 0.770
pid_kd = 948.182
min_temp = 0
max_temp = 130

[fan]
pin = PA0

[mcu]
serial = /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method = command

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[respond]
default_type = command

[pause_resume]
recover_velocity = 50

[bltouch]
sensor_pin = ^PB1
control_pin = PB0
x_offset = -40.5
y_offset = -13.7
z_offset = 1.59

[safe_z_home]
home_xy_position = 117.5,117.5
z_hop = 10

[bed_mesh]
speed = 120
horizontal_move_z = 5
mesh_min = 15, 15
mesh_max = 205, 205
probe_count = 5,3
algorithm = bicubic
fade_start = 1
fade_end = 10
fade_target = 0

[screws_tilt_adjust]
screw1 = 70.5,37.5
screw1_name = front left screw
screw2 = 240,37.5
screw2_name = front right screw
screw3 = 240,207.5
screw3_name = rear right screw
screw4 = 70.5,207.5
screw4_name = rear left screw
horizontal_move_z = 10
speed = 50
screw_thread = CW-M4
=======================
bed_mesh: bicubic interpolation with a probe_count of less than 4 points detected.  Forcing lagrange interpolation. Configured Probe Count: 5, 3
Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'
webhooks client 139881472119984: New connection
webhooks client 139881472119984: Client info {'program': 'Moonraker', 'version': '?'}
mcu 'mcu': Unable to open serial port: [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'

```

As you can see it is referencing a `/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0` device. In the overrides file I specified `- "/dev/ttyUSB0:/dev/ttymxc3"` so I'm not even sure where the firmware got that value from. Read the host's serial connections in /dev/ maybe?

Anyway I was also able to confirm that the device it says it cannot find does actually exist in /dev/serial/by-id/ so I'm not sure why it can't seem to find it. 

Suggestions?

adrianocr, 2022-08-31:
OK so go ahead and disregard the above, I hadn't realized I copied over the [mcu] section in the printer.cfg file and set a bad path to the serial device. 

HOWEVER now that I figured that out and changed it back to ttymxc3 it also didn't work, the logs said there's a permission error. So I tried swapping it out to test with ttyUSB0 and I got the same result. I attached into the container and checked /dev/ and was able to confirm that the device is in there. Here is the bottom of the log as proof:

```log
mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
webhooks client 139665361786272: New connection
webhooks client 139665361786272: Client info {'program': 'Moonraker', 'version': '?'}
mcu 'mcu': Unable to open serial port: [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'

```

ximikator, 2022-09-03:
Had the same problem on arch linux. Try to add permission to the serial port on a host machine: chmod 666 /dev/ttyUSB0 (or ttyACM0 in my case).
Not sure if this is the good solution but it works for me

mkuf, 2022-09-05:
My first reaction would be to set udev rules as @ximikator suggested. 
Check the file permissions on your printers serial port. 
```
ls -al /dev
docker compose exec klipper ls -al /dev
```

Also, run the get-info Script and attach the generated archive to this issue. 
```
./scripts/get-info.sh
```

-Markus

adrianocr, 2022-09-07:
udev rules worked fine

closed 2022-09-07 by adrianocr