mkuf/prind #34: Serial device not found on restarting printer in some time

shogowarrior, 2022-09-03:
Thank you for setting this up. I had been running my own docker with kiauh before but this is much better and maintained 😁 .

My printer is connected to workstation running Klipper. Whenever I turn on my printer, I have to restart the Klipper container to avoid this error. `/dev/ttyACM0` does exist. I have tried restarting from Mainsail but don't see moonraker log getting updated. 

```shell
crw-rw---- 1 root    dialout 166, 1 Sep  3 01:01 ttyACM0

klipper@b22f08cdf8ee:/opt$ tail -f log/klippy.log
=======================
bed_mesh: bicubic interpolation with a probe_count of less than 4 points detected.  Forcing lagrange interpolation. Configured Probe Count: 3, 3
Extruder max_extrude_ratio=0.266081
mcu 'mcu': Starting serial connect
mcu 'mcu': Unable to open serial port: [Errno 6] could not open port /dev/ttyACM0: [Errno 6] No such device or address: '/dev/ttyACM0'
webhooks client 139812987472144: New connection
webhooks client 139812987472144: Client info {'program': 'Moonraker', 'version': '?'}
mcu 'mcu': Unable to open serial port: [Errno 6] could not open port /dev/ttyACM0: [Errno 6] No such device or address: '/dev/ttyACM0'
```

mkuf, 2022-09-05:
Hey there, 

your printer needs to be connected before docker is starting the container, as docker is mounting the serial device into the container. If your printer is connected after the stack is brought up, the klipper container won't start as the device is missing, this is also why you don't see any logs created by klippy.

You might try to circumvent this by configuring the klipper container to start in privileged mode, mount `/dev` from your host and omit the `device` directive in docker-compose.override.yaml. I haven't tested this, but it might be worth a try.

```yaml
services:
  klipper:
    privileged: true
    volumes:
      - /dev:/dev
      - ./config:/opt/cfg
      - run:/opt/run
      - gcode:/opt/gcode
      - log:/opt/log
```

Let me know if this works for you. 

-Markus

shogowarrior, 2022-09-05:
That actually worked! I didn't know I could share  `dev` as a volume itself.  thank you so much!
```yaml
services:
  klipper:
    privileged: true
    volumes:
      - /dev:/dev
```

I needed help on one more thing.  I am not knowledgable at all about the reverse proxy with `nginx`  & `traefic`.
In my case, my workstation is hosted using `OMV` and has heimdall on `omv.local:80`. How can I configure `traefic` to use another port and a virtual hostname in this case say `printer1.local` will route to some other port `omv.local:8888`.

Thanks,
Abhinab

mkuf, 2022-09-07:
Have a look on how the traefik container is configured in `docker-compose.yaml`.  
https://github.com/mkuf/prind/blob/3f5b5ecf4de600e60501951a22422f768c39e64a/docker-compose.yaml#L141-L153  

To change the port to which traefik is bound, add the following to your override file.  
You can then access your printer via you existing hostname (`omv.local`) on Port 81.
```yaml
services:
  traefik:
    command: 
      - "--accesslog" 
      - "--providers.docker=true" 
      - "--providers.docker.exposedbydefault=false" 
      - "--entrypoints.web.address=:81" 
    ports: 
      - "81:81" 
```

- Markus

closed 2022-09-07 by mkuf