mkuf/prind #36: klipperscreen on android device

adrianocr, 2022-09-09:
It is possible to use klipperscreen on an android device: https://klipperscreen.readthedocs.io/en/latest/Android/

I'm at a loss on how to connect it with the container in this repo though. The tutorial seems to rely on having access to certain scripts and a systemd service. 

Is there a way to get the container to forward X11 to the android device?

adrianocr, 2022-09-10:
Figured this out on my own. 

First of all the android device needs to be passed to the container. Since it doesn't show up as a serial device under /dev/serial or something like /dev/ttyUSB0 it is necessary to pass the USB bus to the container. This can be done in your docker compose file like:

```
volumes:
    - /dev/bus/usb/002/006:/dev/bus/usb/002/006 #bus 002 device 006 -- this can be found out with `lsusb`
```

Then I had to manually modify the Dockerfile to allow me to run a couple of `adb` commands. This could likely be done with a shell script, something like (random example) `start_klipperscreen_container.sh` but I personally handled it via supervisord. Here is the relevant part of the dockerfile I modified:

```
...
RUN apt update \
 && apt install -y \
      git \
      xdotool \
      x11-xserver-utils \
      libglib2.0-0 \
      libgirepository-1.0-1 \
      gir1.2-gtk-3.0 \
      libopenjp2-7 \
      fonts-freefont-ttf \
      libcairo2 \
      libatlas3-base \
      adb \
      supervisor \
 && apt clean

WORKDIR /opt
COPY --from=build /opt/klipperscreen ./klipperscreen
COPY --from=build /opt/venv ./venv
ADD supervisord.conf /etc/supervisor/conf.d/supervisord.conf

## Start klipperscreen
ENV XAUTHORITY=/tmp/.Xauthority
ENV DISPLAY=:100
VOLUME ["/opt/cfg"]
ENTRYPOINT ["/usr/bin/supervisord"]
```

(notice I changed the display number to :100 -- for some reason it wasn't working with :0)

and my supervisord.conf file looks like this:

```
[supervisord]
nodaemon=true

[program:adb]
command=/usr/bin/adb devices

[program:androidX11]
command=/bin/bash /opt/klipperscreen/scripts/android-adb.sh

[program:klipperscreen]
command=/opt/venv/bin/python /opt/klipperscreen/screen.py -c /opt/cfg/klipperscreen.conf
```

Since this doesn't require an x11 server on the host machine you can comment out the `/tmp/.X11-unix:/tmp/.X11-unix` line in the klipperscreen service. You also need to change the image to whatever you called your local build of the klipperscreen image. Personally mine is just called `klipperscreen` (after running `docker build -t klipperscreen docker/klipperscreen`).

Make sure your device is plugged in, run `docker compose --profile fluidd --profile klipperscreen up -d`, and viola. 

@mkuf any interest in including the above as an option of some sort? Perhaps as an additional profile or an environment option or something?

mkuf, 2022-09-11:
Hey there, 
I just looked into the klipperscreen docs and can't quite make out, what the script is used for and why you should have to tinker around with adb.  

The [XServer XSDL App](https://play.google.com/store/apps/details?id=x.org.server) mentioned in the Docs is a fully featured XServer, that the klipperscreen container can conntect to without any modifications. 
When starting the App, it displays you the IP/Port under wich the XServer is available.   
Just add this as an env to the klipperscreen service in your override file like so.

```yaml
services:
  klipperscreen:
    environment:
      DISPLAY: 10.0.0.167:0
```

Afterwards start the Stack and the klipperscreen container is connecting to your phone. The only requirement for this to work is that your phone is on the same network as your printer. 

-Markus

adrianocr, 2022-09-11:
@mkuf gonna level with you here.... that didn't even cross my mind. I'll try it when I'm back home. 

However, I'm also assuming that the connection might not be super stable or as fast as it could be over the network. Which is partially why I went the USB route.

I'll do some testing and report back.

mkuf, 2022-09-13:
That makes sense.  
Getting this to work via a wired connection might also be possible without modifying the KlipperScreen image at all.  

The klipperscreen container is using the hosts network stack by https://github.com/mkuf/prind/blob/3f5b5ecf4de600e60501951a22422f768c39e64a/docker-compose.yaml#L122

So you _should_ be able to do the adb tunneling on the host itself and just let the container use the already established connection.  
I'd set up a systemd service on the host that utilizes the [android-adb.sh](https://github.com/jordanruthe/KlipperScreen/blob/master/scripts/android-adb.sh) script to setup the phone and then set the display variable for the klipperscreen container accordingly.

-Markus

mkuf, 2022-10-17:
@adrianocr did you try any of the suggestions or did you stick with your initial setup?
If not, I'd close this issue as resolved.

adrianocr, 2022-10-17:
@mkuf I not only stopped using this project (I spun up a VM as a fake "rpi" of sorts in proxmox) and installed klipper via kiauh, but I also stopped using klipperscreen because I realized I don't actually need it.

closed 2022-10-17 by adrianocr