mkuf/prind #37: Moonraker warnings found. Supplied path (/home/moonraker/gcode_files) for (gcodes) is invalid. Make sure that the path exists and is not the file system root.

Sail2019, 2022-09-13:


mkuf, 2022-09-13:
Please provide a Description to your issue.  
Describe your exact setup and what you attempted to achieve.  
Also run `scripts/get-infos.sh` and attach the generated archive to your response. 

-Markus

mkuf, 2022-09-15:
> [virtual_sdcard]
> path: ./gcode

Thats not wrong, but it also does not really make sense.  
The workingdir for the klipper image is `/opt`, where the gcode volume is mounted in `/opt/gcode`.  
So it happens to work that you can specify the relative path `./gcode` without getting an error. 
The existing config sets the absolute path for the gcode directory. So in the end, the location is the same.
https://github.com/mkuf/prind/blob/3f5b5ecf4de600e60501951a22422f768c39e64a/config/printer.cfg#L14-L15  

I verified this in a local debugging stack and I do not get any Warnings in Moonraker or (in my case) Mainsail.  
I'd be happy to get to the core of this issue, but I need more infos about your setup and the support archive that is generated by the above mentioned script. 

-Markus

mkuf, 2022-09-21:
Any Updates on this? @Sail2019

mkuf, 2022-09-27:
Closing, as this is missing information for debugging.

closed 2022-09-27 by mkuf