mkuf/prind #38: Component 'dbus_manager' failed to load

pelaxa, 2022-09-20:
Hi,

Just getting started on all this.

1. I built the firmware and flashed the printer.  The screen is frozen on Crealty so not sure if there is an issue there or not.
2. updated all the configs per the readme here.  I do not have a webcam.
3. started the containers with the Fluidd profile.

Now I get the following error:
```
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.

dbus_manager

Moonraker warnings found.

Component 'dbus_manager' failed to load with error:
```

Not sure if I have not mapped the USB device properly or there is another issue.
[prind-info-20092022-142440.tar.gz](https://github.com/mkuf/prind/files/9608601/prind-info-20092022-142440.tar.gz)

```
ubuntu@ender3v2:~/prind$ cat docker-compose.override.yaml
## Ustreamer base Service
x-ustreamer-svc: &ustreamer-svc
  image: mkuf/ustreamer:latest
  restart: unless-stopped
  command: --host=0.0.0.0 --port=8080 --slowdown --device=/dev/webcam --resolution=1280x960 --format=MJPEG --desired-fps=30

## Add your personal config here
services:
  klipper:
    devices:
      - /dev/ttyUSB1:/dev/ttymxc3

        #  webcam:
        #    <<: *ustreamer-svc
        #    devices:
        #      - /dev/video0:/dev/webcam
        #    labels:
        #      - "traefik.enable=true"
        #      - "traefik.http.services.webcam.loadbalancer.server.port=8080"
        #      - "traefik.http.routers.webcam.rule=PathPrefix(`/webcam`)"
        #      - "traefik.http.routers.webcam.entrypoints=web"
        #      - "traefik.http.middlewares.webcam.stripprefix.prefixes=/webcam"
        #      - "traefik.http.routers.webcam.middlewares=webcam"
```

mkuf, 2022-09-20:
Hi there, 

please execute the get-info script from the root of the repository and upload it again.  
It seems your pwd was `/scripts` when executing the script. This causes it to miss some files for the archive.

```
cd prind
./scripts/get-info.sh
```
-Markus

pelaxa, 2022-09-20:
apologies.  here it is again
[prind-info-20092022-164437.tar.gz](https://github.com/mkuf/prind/files/9609727/prind-info-20092022-164437.tar.gz)

pelaxa, 2022-09-21:
any help here?

mkuf, 2022-09-21:
Please be patient. I also have other obligations and hadn't found any Time to dive into your issue. 

The runtime-info from the archive shows only two containers (init & traefik).  
You mentioned that you use the fluidd profile. There should also be a fluidd container up and running, no matter if there are problems with klipper or moonraker.

I assume the dbus_manager error is not preventing moonraker from working properly. 
The logs also state an error while connecting to Klippy. 
```
2022-09-20 13:54:41,950 [klippy_connection.py:_get_peer_credentials()] - Failed to get Klippy Peer Credentials, raw: 0xf8250000e8030000e8030000
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/klippy_connection.py", line 244, in _get_peer_credentials
    pid, uid, gid = struct.unpack("@LLL", data)
struct.error: unpack requires a buffer of 24 bytes
```

I just pushed an update to the get-info Script. Please pull these changes to your host, re-run the script and upload the archive again. 

-Markus

pelaxa, 2022-09-21:
I noticed some errors in the log with access denied. see below.

```
2022-09-21 17:27:32,710 [shell_command.py:run_with_response()] - Shell command 'systemctl status 36595' output:
Failed to get unit for PID 36595: Access denied

2022-09-21 17:27:33,211 [machine.py:extract_service_info()] - Error extracting service info
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/components/machine.py", line 724, in extract_service_info
    resp: str = await self.shell_cmd.exec_cmd(
  File "/opt/moonraker/moonraker/components/shell_command.py", line 267, in run_with_response
    raise ShellCommandError(
components.shell_command.ShellCommandError: Error running shell command: '['systemctl', 'status', '36595']'
2022-09-21 17:27:33,211 [moonraker.py:_initialize_component()] - Performing Component Post Init: [proc_stats]
2022-09-21 17:27:33,212 [moonraker.py:_initialize_component()] - Performing Component Post Init: [announcements]
2022-09-21 17:27:33,213 [moonraker.py:_initialize_component()] - Performing Component Post Init: [webcam]
2022-09-21 17:27:33,214 [webcam.py:_set_default_host_ip()] - Default public webcam address set: http://172.26.0.5
2022-09-21 17:27:33,214 [moonraker.py:_initialize_component()] - Performing Component Post Init: [authorization]
2022-09-21 17:27:33,219 [moonraker.py:start_server()] - Starting Moonraker on (0.0.0.0, 7125), Hostname: 4409522d226c
2022-09-21 17:27:33,222 [app.py:listen()] - SSL Certificate/Key not configured, aborting HTTPS Server startup
2022-09-21 17:27:33,235 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 10.13.1.32
2022-09-21 17:27:33,235 [app.py:_log_debug()] - HTTP Request::GET /server/database/item::{'namespace': 'webcams'}
2022-09-21 17:27:33,239 [web.py:log_exception()] - 404 GET /server/database/item?namespace=webcams (10.13.1.32): Namespace 'webcams' not found
2022-09-21 17:27:33,250 [app.py:log_request()] - 404 GET /server/database/item?namespace=webcams (10.13.1.32) [_TRUSTED_USER_] 17.82ms
2022-09-21 17:27:33,324 [announcements.py:_fetch_moonlight()] - Content at moonraker.xml not modified
2022-09-21 17:27:33,379 [announcements.py:_fetch_moonlight()] - Content at klipper.xml not modified
2022-09-21 17:27:33,461 [app.py:_log_debug()] - HTTP Request::GET /access/oneshot_token::{}
2022-09-21 17:27:33,462 [app.py:_log_debug()] - HTTP Response::GET /access/oneshot_token::{'result': '<sanitized>'}
2022-09-21 17:27:33,466 [app.py:log_request()] - 200 GET /access/oneshot_token (10.13.1.32) [_TRUSTED_USER_] 6.37ms
2022-09-21 17:27:33,477 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2022-09-21 17:27:33,477 [klippy_connection.py:_get_peer_credentials()] - Failed to get Klippy Peer Credentials, raw: 0x968e0000e8030000e8030000
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/klippy_connection.py", line 244, in _get_peer_credentials
    pid, uid, gid = struct.unpack("@LLL", data)
struct.error: unpack requires a buffer of 24 bytes
2022-09-21 17:27:33,705 [app.py:log_request()] - 101 GET /websocket?token=KAKF4346PM5T7ZKKF2JM65GNMU2ZL54E (10.13.1.32) [_TRUSTED_USER_] 5.66ms
```

then I updated the Moonraker docker config to add `privileged: true` and it seems to be working now.

is this something that is missing from the repo or is my setup somehow underprivileged?

mkuf, 2022-09-21:
The ShellCommandError is expected, as the process is running isolated in docker.  
With adding the privileged flag you cancel this isolation and grant the container access to your host. This is usually not necessary. 

I just checked on my machine and could not replicate the behaviour with the latest moonraker image.  

-Markus

pelaxa, 2022-09-21:
ok.  here is the  get-info details from before my change
[prind-info-21092022-172030.tar.gz](https://github.com/mkuf/prind/files/9619758/prind-info-21092022-172030.tar.gz)

jmis, 2022-09-23:
I’m seeing the same issue as well.

pelaxa, 2022-09-26:
@jmis are you also running on ubuntu?  I thought maybe my issue was related to running on ubuntu.  but in any case I kept [my change](https://github.com/mkuf/prind/issues/38#issuecomment-1254054965) and so far things are working pretty good.

mkuf, 2022-09-26:
@jmis please also execute the latest version of the get-info script as mentioned in https://github.com/mkuf/prind/issues/38#issuecomment-1252584536 and upload the archive here.

mkuf, 2022-09-26:
@pelaxa 

> ok. here is the get-info details from before my change [prind-info-21092022-172030.tar.gz](https://github.com/mkuf/prind/files/9619758/prind-info-21092022-172030.tar.gz)

This archive was generated using the outdated get-info script.
Please update the Prind repository on your host to get the most recent version of the script.

Then please remove the privileged flag from moonraker and start the stack again. 
Only after that, execute the the updated get-info script, otherwise I can not get any Insight, as to why this might be happening. 

-Markus

jmis, 2022-09-30:
[prind-info-30092022-142220.tar.gz](https://github.com/mkuf/prind/files/9686851/prind-info-30092022-142220.tar.gz)

@pelaxa  Yeah, I'm running Ubuntu.  I'll try adding privileged as a temporary workaround.

jmis, 2022-09-30:
Adding privileged also worked for me.

closed 2022-10-27 by mkuf

champloo, 2023-01-08:
I am getting the same issue. It isn't a showstopper since the printer prints just fine. Here are the klipper and moonraker logs
[klippy.log](https://github.com/mkuf/prind/files/10367483/klippy.log)
[moonraker.log](https://github.com/mkuf/prind/files/10367484/moonraker.log)

acidcoolxxx, 2023-05-10:
@champloo did you find any solution? I have the same problem with the DBUS not a problem but anoying.