mkuf/prind #39: Setting up project in portainer

HitLuca, 2022-10-15:
I am currently using the [klipper-web-control-docker](https://github.com/dimalo/klipper-web-control-docker) project for controlling my printer from my nas, but I discovered this project and I am interested in trying it out. Is it possible to set it up in Portainer?

mkuf, 2022-10-17:
There is currently no documentation on how to set this up in portainer, as I am unfamiliar with it.   
From a short glance at the portainer docs, there is no way to provide an override file for compose, I'm also unsure about if portainer actually uses compose under the hood or is just able to interpret the compose spec.  
  
You could try to add your personal settings directly to the main docker-compose file and then uploading it to portainer.  
The issue I see here is, that the config directory is used as a docker volume and I'm not sure if you can upload the directory structure with the prefilled config files into portainer.   

You may want to create a simple docker-compose file that only consists of the services you'd like to run and also holds the config files in an actual docker volume instead of a bind mounted directory from within this repository. 
  
I'd be happy to add a portainer guide if you get it to work. 

-Markus

gminadak, 2022-10-20:
@HitLuca Hello
I'm by no means an expert on docker or Portainer.

With some help from a friend and a lot of reading online I'm able to run prind from 
Portainer (Community Edition 2.15.1) with the following configuration.

I have commend out the stuff i don't need and i can add what i want for example ustreamer inside the main config.
Obviously you have to change folder paths etc to your need's and don't forget to change the "devices:" port under "klipper:".

I hope it helps.

[image]

[## Common Templates NEW v1.1.1.txt](https://github.com/mkuf/prind/files/9833513/Common.Templates.NEW.v1.1.1.txt)

mkuf, 2022-10-23:
Thanks @gminadak, 

your config seems to be in line with my suggestion. 

* It removes the profile functionality and sets up services in a static way
* it is using `/data/prind` as a directory for all volume mounts that are usually created as docker volumes

The comment you added at the Top of the file suggests to set permissions on that directory as traefik would then not find the moonraker container. Could you elaborate on which permissions should be set?  
This sounds like what the `init` Service is doing once before moonraker/klipper start. Traefik probably has no issues in finding the moonraker container, but the container is stuck in a restart loop as the permissions on the files are set wrong.

-Markus

mkuf, 2022-10-23:
Based on the file provided by @gminadak and some of my own assumptions I created a file that **might** be used with portainer.  
It currently lives in a separate branch but I'd like to merge it into main if I receive positive Feedback.  

I'd be happy if one of you could give it a try and report back. Please pay attention to the setup notes at the Top of the file.
 https://github.com/mkuf/prind/blob/custom-templates/custom/docker-compose.custom.portainer.yaml

-Markus

HitLuca, 2022-10-25:
Testing this approach now, as first feedback I can say that the yaml provided by @mkuf has a

```yaml
profiles:
  - fluidd
```

section that should be removed, or the fluidd container wouldn't launch. I had to remove the `init` container as I couldn't get it to work, but I seem to be able to use the printer without issues.

I am quite in a hurry these days so I cannot finish testing everything, but will add updates as I make progress

gminadak, 2022-10-25:
@mkuf Sorry my bad you need to remove lines 84, 85 from [your config](https://github.com/mkuf/prind/blob/custom-templates/custom/docker-compose.custom.portainer.yaml) for fluid to start

mkuf, 2022-10-25:
Thanks for your Feedback @HitLuca and @gminadak  
The profile has been removed from the fluidd service in https://github.com/mkuf/prind/commit/4fce13be42703abf74cecb66c6d34de0e4daeb42

Did the init conainer work for you @gminadak?

-Markus

gminadak, 2022-10-25:
@mkuf 
> 1. The comment you added at the Top of the file suggests to set permissions on that directory as traefik would then not find the moonraker container. Could you elaborate on which permissions should be set?
>    This sounds like what the `init` Service is doing once before moonraker/klipper start. Traefik probably has no issues in finding the moonraker container, but the container is stuck in a restart loop as the permissions on the files are set wrong.

As I test the configuration and starting to remember thinks....the second reason is that Portainer will make all folders on the persistent volume as `root` as you mentioned
[image]
 Strange think is it's the same as with non persistent volumes and works fine
[image]
and Traefik will start to throw errors.
[image]
As soon as I change the permissions to user everything work's fine
[image]

>Did the init conainer work for you @gminadak?

From what I can see it changed the user:group on /config fine
[image]
but I had to create/copy the folder config by hand for the container to start
[image]

I had to restore my old VM-docker-portainer backup to test all this as I have now move the setup to VM without docker-portainer just plain KIAUH installation!

mkuf, 2022-10-27:
Implemented with https://github.com/mkuf/prind/commit/ff4ca0222645677ef88b765666e7a54d67d53fc1.  
Feel free to create pull requests if you encounter issues with the portainer template.

closed 2022-10-27 by mkuf

HitLuca, 2023-01-12:
May as well also add my configuration for klipperscreen:

In portainer

```
  klipperscreen:
    image: mkuf/klipperscreen:latest
    restart: unless-stopped
    network_mode: host
    environment:
      DISPLAY: ${KLIPPERSCREEN_IP}
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /tmp/.X11-unix:/tmp/.X11-unix
      - ${CONFIG_FOLDER}/klipperscreen.conf:/opt/cfg/klipperscreen.conf
    labels:
      org.prind.service: klipperscreen
```

the env var `KLIPPERSCREEN_IP` is the ip of my phone as shown in the xserver app, like `192.168.1.2:0`

I moved the `klipperscreen.conf` file in the root of my prind config folder, and the contents for two printers are

```
[printer eryone-thinker-se]
moonraker_port: 8013

[printer voron-0]
moonraker_port: 8014
#~# --- Do not edit below this line. This section is auto generated --- #~#
#~#
#~# [main]
#~# use_dpms = False
#~#
```