mkuf/prind #40: Please check option "config_path" in the [file_manager] section of the Moonraker configuration.

tinuva, 2022-10-18:
Installed the prind stack with mainsail.

In the UI I have the following error:

[image]

I couldn't understand this error because there is the following in the moonraker config:
```
[file_manager]
config_path: /opt/cfg
```

When reading the docs: https://moonraker.readthedocs.io/en/latest/configuration/
There is the following quote:
> !!! Note: Previously the [file_manager] section contained config_path and log_path options. These options are now deprecated, as both locations are determined by the data path configured on the command line.

The command line usage now shows: (from https://moonraker.readthedocs.io/en/latest/installation/ )
> usage: moonraker.py [-h]p [-d <data path>] [-c <configfile>] [-l <logfile>] [-n]
>
>Moonraker - Klipper API Server
>
>options:
>  -h, --help            show this help message and exit
>  -d <data path>, --datapath <data path>
>                        Location of Moonraker Data File Path
>  -c <configfile>, --configfile <configfile>
>                        Location of moonraker configuration file

It looks like we going to need a few changes.

ps. I am still brand new to klipper and prind, this just looked liked the most appropriate setup method for what I have available (no raspberry pi available to me)

tinuva, 2022-10-18:
ok I tried to work around most of the issues, but now I still can't upload gcode files.
```
2022-10-18 14:23:03,438 [web.py:log_exception()] - Uncaught exception POST /server/files/upload (10.0.0.40)
HTTPServerRequest(protocol='http', host='octo.heaven.za.net', method='POST', uri='/server/files/upload', version='HTTP/1.1', remote_ip='10.0.0.40')
Traceback (most recent call last):
  File "/opt/venv/lib/python3.10/site-packages/tornado/web.py", line 1681, in _execute
    result = self.prepare()
  File "/opt/moonraker/moonraker/app.py", line 882, in prepare
    fm.check_write_enabled()
  File "/opt/moonraker/moonraker/components/file_manager/file_manager.py", line 201, in check_write_enabled
    raise self.server.error(
utils.ServerError: Write access is currently disabled.  Check notifications for warnings.
2022-10-18 14:23:03,451 [app.py:log_request()] - 500 POST /server/files/upload (10.0.0.40) [_TRUSTED_USER_] 15.49ms
```
I suspect this is because of https://github.com/Arksine/moonraker/issues/516

Maybe I should look for an older moonraker for now until prind is updated to work with the new change

tinuva, 2022-10-18:
Can confirm, on `mkuf/moonraker:60b871a` (20 days old at this time) all the config issues are resolved.

mkuf, 2022-10-18:
Thanks for bringing this to my attention.  
This required some rework on the directory structure of the klipper and moonraker images, which is now merged into `main`.  
  
This change also requires a rebuild of the current latest images.
As soon as this is done, I'll add a new Release that can be safely checked out. 

-Markus

mkuf, 2022-10-18:
Images are being rebuilt: 

* klipper: https://github.com/mkuf/prind/actions/runs/3276148154
* moonraker: https://github.com/mkuf/prind/actions/runs/3276148860

mkuf, 2022-10-18:
Fixed in Release [v1.1.0](https://github.com/mkuf/prind/releases/tag/v1.1.0).

commit 2590a3ea (2022-10-18): Add changes introduced in https://github.com/Arksine/moonraker/pull/491 to moonraker and klipper images.

fixes #40
closed 2022-10-18 by mkuf