mkuf/prind #41: Need extra permissions set in init

tinuva, 2022-10-19:
Thank you for the fix for #40 but I seem to still run into an issue with moonraker.

```
root@trex:~/prind# docker compose run moonraker
[+] Running 4/4
 ⠿ Network prind_default        Created                                                                                                                                                                                                0.3s
 ⠿ Volume "prind_moonraker-db"  Created                                                                                                                                                                                                0.0s
 ⠿ Container prind-init-1       Created                                                                                                                                                                                                0.2s
 ⠿ Container prind-klipper-1    Created                                                                                                                                                                                                0.2s
[+] Running 2/2
 ⠿ Container prind-init-1     Exited                                                                                                                                                                                                   2.0s
 ⠿ Container prind-klipper-1  Started                                                                                                                                                                                                  1.5s
[utils.py:setup_logging()] - data_path: /opt/printer_data
[utils.py:setup_logging()] - is_default_data_path: False
[utils.py:setup_logging()] - config_file: /opt/printer_data/config/moonraker.conf
[utils.py:setup_logging()] - startup_warnings: []
[utils.py:setup_logging()] - verbose: False
[utils.py:setup_logging()] - debug: False
[utils.py:setup_logging()] - asyncio_debug: False
[utils.py:setup_logging()] - log_file: /opt/printer_data/logs/moonraker.log
[utils.py:setup_logging()] - software_version: ?
[utils.py:setup_logging()] - python_version: 3.10.8 (main, Oct 13 2022, 22:47:45) [GCC 10.2.1 20210110]
[confighelper.py:read_file()] - Configuration File '/opt/printer_data/config/moonraker.conf' parsed, total size: 240 B
[moonraker.py:add_log_rollover_item()] -
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /opt/printer_data/run/klipper.sock

[machine]
provider = systemd_cli
validate_service = False

[authorization]
trusted_clients =
	0.0.0.0/0
cors_domains =
	*

[octoprint_compat]

[history]

#################################################################
All Configuration Files:
/opt/printer_data/config/moonraker.conf
#################################################################
[app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/info
[websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.info
[app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /server/config
[websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.config
[app.py:register_local_handler()] - Registering HTTP Endpoint: (POST) /server/restart
[websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: server.restart
[moonraker.py:load_component()] - Component (dbus_manager) loaded
[moonraker.py:load_component()] - Unable to load component: (database)
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/moonraker.py", line 250, in load_component
    component = load_func(config)
  File "/opt/moonraker/moonraker/components/database.py", line 944, in load_component
    return MoonrakerDatabase(config)
  File "/opt/moonraker/moonraker/components/database.py", line 103, in __init__
    self.lmdb_env = lmdb.open(self.database_path, map_size=MAX_DB_SIZE,
lmdb.ReadonlyError: /opt/printer_data/database: Permission denied
[moonraker.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/moonraker.py", line 250, in load_component
    component = load_func(config)
  File "/opt/moonraker/moonraker/components/database.py", line 944, in load_component
    return MoonrakerDatabase(config)
  File "/opt/moonraker/moonraker/components/database.py", line 103, in __init__
    self.lmdb_env = lmdb.open(self.database_path, map_size=MAX_DB_SIZE,
lmdb.ReadonlyError: /opt/printer_data/database: Permission denied

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/moonraker/moonraker/moonraker.py", line 497, in main
    server.load_components()
  File "/opt/moonraker/moonraker/moonraker.py", line 227, in load_components
    self.load_component(config, component)
  File "/opt/moonraker/moonraker/moonraker.py", line 257, in load_component
    raise ServerError(msg)
utils.ServerError: Unable to load component: (database)
[moonraker.py:main()] - Server Shutdown
```

Ok so lets override the entry point to see what is going on.

```
~/prind# docker compose run --entrypoint="bash" moonraker
moonraker@f52f5db8f1e2:/opt$ ls -lsah printer_data/
total 20K
4.0K drwxr-xr-x 7 moonraker moonraker 4.0K Oct 19 09:33 .
4.0K drwxr-xr-x 1 root      root      4.0K Oct 19 09:33 ..
4.0K drwxr-xr-x 2 moonraker moonraker 4.0K Oct 19 05:53 config
4.0K drwxr-xr-x 2 root      root      4.0K Oct 19 09:25 database
4.0K drwxr-xr-x 2 root      root      4.0K Oct 19 09:25 gcodes
   0 drwxrwxrwt 2 root      root        80 Oct 19 09:26 logs
   0 drwxrwxrwt 2 root      root        80 Oct 19 09:25 run
```

Ok so you already have the init container/service which I believe can fix this.

So after changing it to:
```
  init:
    image: busybox:latest
    command: chown -R 1000:1000 /prind/config /opt/run /opt/gcodes /opt/logs /opt/database
    volumes:
      - .:/prind
      - run:/opt/run
      - gcode:/opt/gcodes
      - log:/opt/logs
      - moonraker-db:/opt/database
```

I was able to get moonraker finally to run.

mkuf, 2022-10-19:
Workflows are currently running again:

* klipper: https://github.com/mkuf/prind/actions/runs/3283110130
* moonraker: https://github.com/mkuf/prind/actions/runs/3283110798

mkuf, 2022-10-19:
Fixed with Release [v1.1.1](https://github.com/mkuf/prind/releases/tag/v1.1.1).  
Be sure to re-pull the Moonraker and Klipper image, as their digests have changed.

-Markus

tinuva, 2022-10-20:
Thank you very much. Really awesome how quick you are on the fixes.