mkuf/prind #45: Enable remote access with moonraker-telegram-bot

moookino, 2022-12-05:
Hi,

Do you think it's possible to add telegram bot? I think it's good way to keep an eye on prints
Existing projects:
https://github.com/nlef/moonraker-telegram-bot
https://github.com/Raabi91/moonraker-telegram

Thanks for your work.

mkuf, 2022-12-08:
Hey there, 

this seems doable. 
I'd go with the moonraker-telegram-bot by @nlef as this is also supported by kiauh and more users should be familliar with it. 

To add this to prind, we need a new subdirectory under `docker/` and a Dockerfile.  
The Image generated by this can then be added to docker-compose.yaml linked to a new profile, in the same way as klipperscreen.

I personally do not see a value in this, as I'm not leaving my printer alone. Also, I don't use Telegram 😅 

If you or someone else want to give it a try, I'm happy to accept pull requests. 

-Markus

nlef, 2022-12-08:
Greeting's.
If you need docker-images with my bot, please, create an issue with feature request. I'll work on it)

moookino, 2022-12-08:
@nlef please see https://github.com/nlef/moonraker-telegram-bot/issues/227, thank you

nlef, 2022-12-11:
@mkuf https://github.com/nlef/moonraker-telegram-bot/pull/228 is ready to be reviewed)

mkuf, 2022-12-19:
Further progress will be tracked in the pr https://github.com/mkuf/prind/pull/48

closed 2022-12-19 by mkuf