mkuf/prind #46: timelapse support out of the box

ajahongir, 2022-12-06:
hey guys is there any was to configuring timelapse support out of box?

rep: https://github.com/mainsail-crew/moonraker-timelapse
instruction: https://www.youtube.com/watch?v=Z2ut7lGHWuQ

mkuf, 2022-12-08:
Hey there, 

I'd rather not have this added to the moonraker base image. 
Installing ffmpeg adds about 500M to the Image size. That is too much for a functionality not all users of prind might need.

You can easily create a new Dockerfile that references mkuf/moonraker:latest and adds the necessary requirements.  
From glancing over the docs provided, I'd do something like this:

1. Create `docker/moonraker/Dockerfile.ffmpeg`
```Dockerfile
FROM mkuf/moonraker:latest

USER root
RUN apt update \
 && apt install -y ffmpeg \
 && apt clean

USER moonraker
```
2. check out https://github.com/mainsail-crew/moonraker-timelapse to a separate directory
```
git clone https://github.com/mainsail-crew/moonraker-timelapse /opt/moonraker-timelapse
```
3. add the module as a volume to the moonraker service in docker-compose.override.yaml
4. add a build instruction to docker-compose.override.yaml for the moonraker service
```yaml
[...]

services:
  moonraker:
    image: moonraker:ffmpeg
    build:
      context: docker/moonraker
      dockerfile: Dockerfile.ffmpeg
    volumes:
      - /opt/moonraker-timelapse/component/timelapse.py:/opt/moonraker/moonraker/components/timelapse.py
      - /dev/null:/opt/klipper/config/null
      - /dev/null:/opt/klipper/docs/null
      - /run/dbus:/run/dbus
      - /run/systemd:/run/systemd
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - moonraker-db:/opt/printer_data/database
      - ./config:/opt/printer_data/config

[...]
```

You can then go ahead and add the config macros to klipper and might also want to add additional Volumes to moonraker to extract the generated Timelapse files from the container. 

Let me know if this works for you.

-Markus

mkuf, 2022-12-16:
I'm closing this for now, we can reopen this if you have any further questions.  
Feel free to post your setup for others, if you're successful in using the timelapse plugin.  

-Markus

closed 2022-12-16 by mkuf

mkuf, 2023-04-07:
This setup is now also part of the `custom` directory within the repository.  
See https://github.com/mkuf/prind/tree/main/custom/moonraker-timelapse

domi-a, 2023-09-11:
> This setup is now also part of the `custom` directory within the repository. See https://github.com/mkuf/prind/tree/main/custom/moonraker-timelapse

AWESOME work! @mkuf - got it working nerly first try!

except that `wget` seems to be missing in moonraker:ffmpeg.
for others passing by: i had to add `&& apt install -y wget \` to the dockerfile_inline of [docker-compose.custom.moonraker-timelapse.override.yaml](https://github.com/mkuf/prind/blob/main/custom/docker-compose.custom.moonraker-timelapse.override.yaml)

Thank You So Much for this docker configuration of klipper & co - very advanced implementation!
i was tinkering around for hours with other installation scripts till i found this one!

mkuf, 2023-09-13:
Thanks for the feedback @domi-a, I added this in https://github.com/mkuf/prind/commit/5c619921edd5dc3274e0aed85342eb73d514b732