mkuf/prind #47: Connection refused & Failed to initialize Gtk

mela93, 2022-12-07:
[image]

```
prind-klipperscreen-1  | Unable to init server: Could not connect: Connection refused
prind-klipperscreen-1  | Unable to init server: Could not connect: Connection refused
prind-klipperscreen-1  | 2022-12-07 21:30:58,577 [screen.py:main()] - KlipperScreen version: v0.2.9-104-g36faaa1
prind-klipperscreen-1  | Unable to init server: Could not connect: Connection refused
prind-klipperscreen-1  | 2022-12-07 21:30:58,579 [screen.py:main()] - Failed to initialize Gtk
prind-klipperscreen-1 exited with code 1
```

[image]

```
## System Info 
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Docker Buildx (Docker Inc., v0.9.1-docker)
  compose: Docker Compose (Docker Inc., v2.12.2)

Server:
 Containers: 7
  Running: 4
  Paused: 0
  Stopped: 3
 Images: 7
 Server Version: 20.10.21
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 770bd0108c32f3fb5c73ae1264f7e503fe7b2661
 runc version: v1.1.4-0-g5fd4c4d
 init version: de40ad0
 Security Options:
  seccomp
   Profile: default
  cgroupns
 Kernel Version: 5.15.80-flippy-79+o
 Operating System: Armbian 23.02.0-trunk Bullseye
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 1.908GiB
 Name: armbian
 ID: N6NJ:FV4T:DDRX:P5ZQ:EIRQ:VCVA:MYE2:M4HO:2OCH:PTK3:NBLG:F4HC
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Registry Mirrors:
  https://mirror.baidubce.com/
[... 276 lines cut ...]
/tmp/tmp.pRfsTGzxGk-prind/prind/custom/docker-compose.custom.portainer.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/custom/docker-compose.custom.multiple-printers.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/custom/README.md
/tmp/tmp.pRfsTGzxGk-prind/prind/config/
/tmp/tmp.pRfsTGzxGk-prind/prind/config/mks_mini_12864_v3.cfg
/tmp/tmp.pRfsTGzxGk-prind/prind/config/printer.cfg
/tmp/tmp.pRfsTGzxGk-prind/prind/config/.moonraker.conf.bkp
/tmp/tmp.pRfsTGzxGk-prind/prind/config/printer.20221206.cfg
/tmp/tmp.pRfsTGzxGk-prind/prind/config/printer.cfg.backup
/tmp/tmp.pRfsTGzxGk-prind/prind/config/printer-simulavr.cfg
/tmp/tmp.pRfsTGzxGk-prind/prind/config/octoprint.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/config/moonraker.conf
/tmp/tmp.pRfsTGzxGk-prind/prind/config/klipperscreen.conf
/tmp/tmp.pRfsTGzxGk-prind/prind/config/build.config
/tmp/tmp.pRfsTGzxGk-prind/prind/README.md
/tmp/tmp.pRfsTGzxGk-prind/prind/LICENSE
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/ustreamer.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/moonraker.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/klipperscreen.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/klipper.yaml
/tmp/tmp.pRfsTGzxGk-prind/prind/.github/workflows/image-workflow.yaml
/tmp/tmp.pRfsTGzxGk-prind/runtime-info.txt                     
```

mkuf, 2022-12-08:
Hey there, 

* please update to **v1.3.1** and re-run `scripts/get-info.sh`. 
This generates an **archive** which is needed for troubleshooting. The log you posted, alone is not enough for me to understand your Setup. 
* Please also provide the **commands** you ran when starting the stack. 
* From the Outputs you posted, neither klipper nor moonraker is up and running, which explaines why klipperscreen is logging about connection failures. 
* Is the xinit service running? What do you see on the connected screen?

-Markus

mkuf, 2022-12-16:
Any update on this?

mela93, 2022-12-18:
The archive here:
[prind-info-18122022-193457.tar.gz](https://github.com/mkuf/prind/files/10253729/prind-info-18122022-193457.tar.gz)

My command: `docker compose --profile fluidd --profile klipperscreen up -d`

Thanks for last suggestion, I ran the `startx` and it working now.

closed 2022-12-18 by mela93