mkuf/prind #51: Reconnect to printer if it has been power cycled

HitLuca, 2023-01-05:
Not sure how feasible this is or if there is already a way of doing it, but I always had this issue with "klipper on docker" projects: if I start the klipper container while the printer is on I don't have issues. If I power cycle the printer (turn it off and then on again, the docker containers are hosted on my nas) then I cannot reconnect to it, until I restart the klipper container. Is there a way of making sure that the connection is recovered or the container is restarted automatically? Failure to connect to the printer doesn't exit the klipper process so even adding a `restart: always` directive to the container doesn't work

mkuf, 2023-01-05:
As klipper is not exiting when the device is disconnected, I don't think theres a docker-way of handling this.  
I'd create a udev rule on the docker host, which starts a script when the device is connected.   
See this stackexchange thread for more info: https://unix.stackexchange.com/questions/28548/how-to-run-custom-scripts-upon-usb-device-plug-in  

The script could be as simple as this:
```bash
#!/bin/bash

klipper_container=$(docker ps -aqf "label=org.prind.service=klipper")
docker restart ${klipper_container}
```

Let me know if this works for you.

-Markus

HitLuca, 2023-01-09:
Worked as a charm! For anyone else that may find this useful, here a test udev rule that is working for me (detecting when a Fysetc Cheetah 2.0 board gets reconnected)

```
SUBSYSTEM=="tty", ATTRS{idVendor}=="1d50", ATTRS{product}=="stm32f401xc", ATTRS{serial}=="160035000751383531393138", RUN+="/home/luca/Desktop/restart_klipper.sh"
```

closed 2023-01-09 by HitLuca