mkuf/prind #52: I want use CAN bus as device connect how to update yaml config?

mela93, 2023-01-07:
this is Serial Port config at docker-compose.override.yaml
```
klipper:
  devices:
    - /dev/serial/by-id/xxx-example:/dev/serial/by-id/xxx-example
```

when i use can bus,
the config.cfg will change to this, and how to update docker-compose.override.yaml?
```
[mcu]
#serial: /dev/serial/by-id/xxx-example
canbus_uuid: 9999c29dd86f
canbus_interface: can0
```

mkuf, 2023-01-07:
Hi there, 

I have no experience using CAN bus. Neither with klipper nor with docker.
A quick google search pointed me to this stackoverflow thread: https://stackoverflow.com/questions/69078501/how-to-make-can-bus-operate-from-within-a-docker-container

Following this example, you can remove the device section from the klipper service and add the host network like so:

```yaml
services:
  klipper:
    network_mode: host
```

Let me know if this works for you. 

-Markus

mela93, 2023-01-08:
Yes, it works.
Thank you so much.

closed 2023-01-08 by mela93