mkuf/prind #53: Can host reboot/shutdown from mainsail be enabled?

HitLuca, 2023-02-02:
I was wondering, since I've deployed this project to a raspberry pi, would it be possible to have the reboot/shutdown commands working from moonraker? I get an error when I try to do it:
```
Traceback (most recent call last):

  File "/opt/moonraker/moonraker/websockets.py", line 299, in execute_method
    result = await callback(params)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/websockets.py", line 412, in func
    result = await callback(
             ^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/machine.py", line 277, in _handle_machine_request
    raise self.server.error(
utils.ServerError: Cannot reboot from within a docker container
```

I saw that a method to do this is to mount `/etc/systemd` in the container, but it's already done

mkuf, 2023-02-02:
This is certainly possible and should work out of the box, as `/run/systemd` and `/run/dbus` are mounted within the container.
I just tested this with the most recent moonraker image (`61d3182`) and could not reproduce the described behaviour.

Please run the get-info script, upload the generated archive and provide the **command you used to start the stack**.
```bash
cd prind
bash ./scripts/get-info.sh
```

-Markus

HitLuca, 2023-02-02:
Very strange, it may be related to my setup. I'm using portainer/ubuntu server/raspberry pi 3b+

HitLuca, 2023-02-02:
@mkuf here is my portainer compose, as well as the output of `get-info.sh` ran from my prind config folder

```
services:
  klipper:
    image: mkuf/klipper:latest
    restart: unless-stopped
    command: -I printer_data/run/klipper.tty -a printer_data/run/klipper.sock printer_data/config/printer.cfg -l printer_data/logs/klippy.log
    devices:
      - /dev/serial/by-id/usb-Klipper_stm32f401xc_160035000751383531393138-if00:/dev/serial/by-id/usb-Klipper_stm32f401xc_160035000751383531393138-if00
    volumes:
      - /home/pi/docker/prind/config:/opt/printer_data/config
      - /home/pi/docker/prind/run:/opt/printer_data/run
      - /home/pi/docker/prind/gcode:/opt/printer_data/gcodes
      - /home/pi/docker/prind/log:/opt/printer_data/logs
    labels:
      org.prind.service: klipper
      org.prind.printer: voron-0
      com.centurylinklabs.watchtower.enable: false
      
  moonraker:
    image: mkuf/moonraker:latest
    restart: unless-stopped
    privileged: true
    network_mode: host
    volumes:
      - /dev/null:/opt/klipper/config/null
      - /dev/null:/opt/klipper/docs/null
      - /run/dbus:/run/dbus
      - /run/systemd:/run/systemd
      - /home/pi/docker/prind/run:/opt/printer_data/run
      - /home/pi/docker/prind/gcode:/opt/printer_data/gcodes
      - /home/pi/docker/prind/log:/opt/printer_data/logs
      - /home/pi/docker/prind/moonraker-db:/opt/printer_data/database
      - /home/pi/docker/prind/config:/opt/printer_data/config
    labels:
      org.prind.service: moonraker
      org.prind.printer: voron-0
      com.centurylinklabs.watchtower.enable: false
      
  mainsail:
    image: ghcr.io/mainsail-crew/mainsail:edge
    restart: unless-stopped
    network_mode: host
    volumes:
      - /home/pi/docker/prind/mainsail-config.json:/usr/share/nginx/html/config.json
    labels:
      org.prind.service: mainsail
```
[prind-info-02022023-192510.tar.gz](https://github.com/mkuf/prind/files/10572193/prind-info-02022023-192510.tar.gz)

mkuf, 2023-02-02:
Have a look at how moonraker is configured in docker-compose.yaml.
In your custom configuration it seems to be missing the `pid` parameter. This is required for systemd to be called from within the container. 
https://github.com/mkuf/prind/blob/6c40d55c7c2971541ab0877a7b8c41c6d7d68e76/docker-compose.yaml#L49

-Markus

HitLuca, 2023-02-03:
Added `pid: host` on my moonraker container, and now the issue has shifted to this error
`dbus_next.errors.DBusError: Interactive authentication required`

mkuf, 2023-02-03:
Your moonraker.conf is missing the machine configuration.
https://github.com/mkuf/prind/blob/18435827c499f1d3ee0abab583d7660481171987/config/moonraker.conf#L6-L8

I'd suggest to keep your config files and service configurations as close as possible to the defaults provided in this repository. 

-Markus

HitLuca, 2023-02-03:
Thank you! Sadly my current config was the result of a bunch of tries to fix an issue I was having, and I accidentally left out stuff. Will be more careful next time!

mkuf, 2023-02-03:
No worries, feel free to reach out any time if you happen to have any issues with the packages provided here. :)

closed 2023-02-03 by mkuf

HitLuca, 2023-02-06:
Have to reopen this issue, even with `pid: host` and the proper `[machine]` configuration, the container cannot shutdown the host. This time the moonraker logs don't show any warning or error.

mkuf, 2023-02-06:
What do the logs show, if no warning or error?  
Did you check your hosts log for errors?

Does the shutdown work if you run this command on your host?
```
curl -X POST http://localhost/machine/shutdown \
   -H 'Content-Type: application/json' \
   -d '{"jsonrpc": "2.0", "method": "machine.shutdown", "id": 4665}'
```

HitLuca, 2023-02-06:
Regarding your command, I get a `Permanent Redirect`. I don't seem to have found any weird logs on my host system when checking `journalctl`. I also tried to exec into my moonraker container and send a `shutdown`command, but I get this

```
my_host:~$ docker exec -it  prind-eryone-thinker-se-moonraker-1 bash
moonraker@637205fe3cec:/opt$ shutdown now
Failed to set wall message, ignoring: Interactive authentication required.
Failed to power off system via logind: Interactive authentication required.
Failed to talk to init daemon.
```

HitLuca, 2023-02-06:
After further testing I got to the point where I could trigger a host reboot from a docker container (not moonraker) by using an image that matches my host os and version. Could it be that the issue is related to this? Keep in mind that the command only worked when I added the last volume mapping `/usr/bin/systemctl:/usr/bin/systemct`, as before it was giving me errors

```
docker run --privileged -v /run/systemd/system:/run/systemd/system -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket -v /usr/bin/systemctl:/usr/bin/systemctl -it ubuntu:22.04 systemctl reboot
```

Maybe this means nothing, but I'll try to do further research to see if we can figure this out together, as it may affect other people in the future

mkuf, 2023-02-07:
Please generate a new archive of your current setup using `get-info.sh` and share it here.

---

> If I send the jsonrpc request to `localhost:7125` which is the port used by my moonraker instance I get a `{"result": "ok"}`, but still no shutdown or anything
> 
> Sending a shutdown command from the moonraker container results in this
> 
> ```
> my_host:~$ docker exec -it  prind-eryone-thinker-se-moonraker-1 bash
> moonraker@637205fe3cec:/opt$ shutdown now
> Failed to set wall message, ignoring: Interactive authentication required.
> Failed to power off system via logind: Interactive authentication required.
> ```
> 
> This leads me to believe the container still doesn't have the proper authorizations to send a shutdown command, but can't really tell what's missing

Does moonraker log the shutdown request?  
The user is not authorized to run systemctl, you'll have to use `sudo` to call `systemctl halt`, as the moonraker code suggests.   

https://github.com/Arksine/moonraker/blob/f9086181b89dcc6ea71a8606aeb484ff34b2f31e/moonraker/components/machine.py#L783-L787

```
root@sparkcube:~# docker exec -it prind-moonraker-1 bash
moonraker@e1f124943310:/opt$ which systemctl
/bin/systemctl
moonraker@e1f124943310:/opt$ sudo systemctl halt
Connection to sparkcube closed by remote host.
Connection to sparkcube closed.
```

---
> After further testing I got to the point where I could trigger a host reboot from a docker container (not moonraker) by using an image that matches my host os and version. Could it be that the issue [...]

The reason this is working is because the default user for the ubuntu image is root, which is authorized to run the shutdown command.  
Mounting the systemctl binary into the container is not necessary for moonraker as it is installed at container build time. 

https://github.com/mkuf/prind/blob/18435827c499f1d3ee0abab583d7660481171987/docker/moonraker/Dockerfile#L20-L36

HitLuca, 2023-02-07:
Update: I spun up an ubuntu 22.04 vm with only docker, portainer, and prind. 

If I set `pid: host` for the moonraker service and shutdown from mainsail nothing happens, the moonraker logs don't show any request sent and i cannot shutdown the host. If I exec into the container and send a `shutdown` command I get an `Interactive authentication required`. If I send a `sudo shutdown` I get a `Running in chroot, ignoring request.` error.

If I don't set `pid:host` and as before send a `shutdown` command I still get the authentication required error, but if I `sudo` it I can shutdown the host, without being prompted for a password. Looks like in my case I need to avoid using `pid:host` and have moonraker use `sudo` to shutdown/restart the system.

Does it look normal to you and I am just missing a moonraker option to call the commands as sudo or is it weird behavior? If you want I could send you my virtualbox disk, it's 7gb and will take some time to upload somewhere

mkuf, 2023-02-10:
I think this is related to https://github.com/portainer/portainer/issues/2431, where portainer is interfering with the containers configuration because it deems it "unsafe".  
You may want start the stack as described in the readme to avoid issues and use portainer as a frontend to manage the state of containers but not to deploy the stack. Or maybe not use portainer at all. :sweat_smile: 

-Markus

HitLuca, 2023-02-10:
I can totally spin up the stack from outside portainer, I am just using it for everything else, but if nothing else can be done it won't be the biggest issue for me!
Thank you again for all the help, I really appreciate it :)