mkuf/prind #54: cant flash controller

trackhacs, 2023-02-03:
I could successfully compile the firmware within the container. however - if I try to flash it to I get the error:

```
WARN[0000] Found orphan containers ([prind-fluidd-1 prind-moonraker-1 prind-klipper-1 prind-octoprint-1 prind-klipper-priv-1 prind-mainsail-1]) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up. 
  Building hid-flash
/bin/sh: 1: pkg-config: not found
    hid-flash requires libusb-1.0, please install with:
    sudo apt-get install libusb-1.0
  Flashing out/klipper.bin to /dev/ttyACM1
Entering bootloader on /dev/ttyACM1
Device reconnect on /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0
sudo dfu-util -p 1-1.5 -R -a 0 -s 0x8008000:leave -D out/klipper.bin

Traceback (most recent call last):
  File "./scripts/flash_usb.py", line 386, in <module>
    main()
  File "./scripts/flash_usb.py", line 383, in main
    flash_func(options, args[0])
  File "./scripts/flash_usb.py", line 313, in flash_stm32f4
    ["-R", "-a", "0", "-s", start], options.sudo)
  File "./scripts/flash_usb.py", line 156, in flash_dfuutil
    call_dfuutil(["-p", buspath] + extra_flags, binfile, sudo)
  File "./scripts/flash_usb.py", line 139, in call_dfuutil
    res = subprocess.call(args)
  File "/usr/lib/python3.6/subprocess.py", line 287, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'sudo': 'sudo'
src/stm32/Makefile:97: recipe for target 'flash' failed
make: *** [flash] Error 1
```

mkuf, 2023-02-03:
It seems like the mcu image is missing the `sudo` command, therefore dfu-util can't be executed.  
This has been added in https://github.com/mkuf/prind/commit/67ffc9c1c50d60ad4a7297914adefc68d0739542

The klipper image is currently being re-built, this will take about an hour.  
You can follow the progress here https://github.com/mkuf/prind/actions/runs/4083441350/jobs/7038967819  
When the Action is completed, delete all klipper container images from your machine and try again. 

-Markus

trackhacs, 2023-02-03:
that's was fast. :)

tried it, but now get the following:

```
WARN[0000] Found orphan containers ([prind-traefik-1 prind-init-1 prind-webcam-1 prind-fluidd-1 prind-moonraker-1 prind-klipper-1 prind-octoprint-1 prind-klipper-priv-1 prind-mainsail-1]) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up. 
Loaded configuration '/opt/klipper/.config'
No change to configuration in '/opt/klipper/.config'
  Building hid-flash
/bin/sh: 1: pkg-config: not found
    hid-flash requires libusb-1.0, please install with:
    sudo apt-get install libusb-1.0
  Flashing out/klipper.bin to /dev/serial/by-id/usb-STMicroelectronics_MARLIN_BIGTREE_OCTOPUS_V1_CDC_in_FS_Mode_205634505931-if00
Entering bootloader on /dev/serial/by-id/usb-STMicroelectronics_MARLIN_BIGTREE_OCTOPUS_V1_CDC_in_FS_Mode_205634505931-if00
Device reconnect on /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.5/1-1.5:1.0
sudo dfu-util -p 1-1.5 -R -a 0 -s 0x8008000:leave -D out/klipper.bin

sudo: dfu-util: command not found

Failed to flash to /dev/serial/by-id/usb-STMicroelectronics_MARLIN_BIGTREE_OCTOPUS_V1_CDC_in_FS_Mode_205634505931-if00: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=/dev/serial/by-id/usb-STMicroelectronics_MARLIN_BIGTREE_OCTOPUS_V1_CDC_in_FS_Mode_205634505931-if00

src/stm32/Makefile:97: recipe for target 'flash' failed
make: *** [flash] Error 255
```

mkuf, 2023-02-03:
While accumulating the packages for the mcu image I referenced klippers `install-ubuntu-18.04.sh` script which is missing this package. It seems that `dfu-util` only gets installed by klippers `install-octopi.sh`.  

So, same spiel again. 
Added the package in https://github.com/mkuf/prind/commit/18435827c499f1d3ee0abab583d7660481171987
Image is being re-built in https://github.com/mkuf/prind/actions/runs/4084121082/jobs/7040447537

-Markus

trackhacs, 2023-02-03:
ah - deutsche zeitzone ;)

seems to work - however - cant get my Octopus V1.1 into DFU mode ;\

mkuf, 2023-02-03:
I'm not familiar with the Octopus board, but usually those boards have a boot button which needs to be pressed at power-on to enter dfu mode.  

If you're sure your board is in dfu mode, you may also try to start a temporary container instead of using the method described in the readme.  
In this container, you'll have klipper and everything you need to build and flash the firmware and you can interact with it in the same way as it would be directly on your host. This is probably easier for troubleshooting.

```
docker run \
  --rm -it --privileged --workdir /opt/klipper \
  -v /dev:/dev -v $(pwd)/config/build.config:/opt/klipper/.config -v $(pwd)/out:/opt/klipper/out \ 
  mkuf/klipper:latest-mcu
```

The output from your earlier logs also suggest that you'll need to specify a different make command, if your board is already in dfu mode.
> If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

mkuf, 2023-02-11:
Any update on this?  
Can you judge if this is an issue with your hardware or if there is still an issue with the Image?

-Markus

closed 2023-02-18 by mkuf

trackhacs, 2023-02-19:
could unfortunately get the octopus not into DFU mode ...