mkuf/prind #57: folder permissions set by the `init` container don't always match the user's gid

HitLuca, 2023-02-06:
Just a heads up, it's not guaranteed that the permissions set by the `init` container to the `config` folder and inner files (`1000:1000`) will always match the user's `pid:gid` combo. An example is a raspberry-pi 3B+ running Ubuntu Server 22.04, which has the following ids

```
uid=1000(pi) gid=1003(pi) groups=1003(pi),4(adm),5(tty),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),108(render),116(netdev),999(docker),1000(gpio),1001(spi),1002(i2c)
```

You can see that the `init` container would set the `pi:gpio` permissions instead of `pi:pi`. It shouldn't be a big issue but it's useful to keep in mind

mkuf, 2023-02-07:
The permissions are not supposed to match the uid/gid of the current user.  
They are supposed to match the users within the containers, so the services can access the files.  
All Containers of this project use 1000:1000 to run the applications. 

-Markus

closed 2023-02-07 by mkuf

HitLuca, 2023-02-07:
Thank you for the explanation, it's just a bit weird to have folders owned by groups like gpio, but overall it makes sense with how the project works