mkuf/prind #59: Where is the camera supposed to appear?

aguinaldoabbj, 2023-02-13:
Hi!

Congrats on this great work. I got a fully working klipper in a few hours. However, somethings aren't working as supposed.

Should the camera widget be manually configured? What's the URL to get the stream? Or does it need any addition configuration in Klipper? How's the telegram.conf supposed to be loaded and used ?

Thanks in advance!

mkuf, 2023-02-14:
Hi there, 

the webcam has to be manually added to whichever frontend you've chosen to use.  
By default, the ustreamer service is available under the `/webcam` path so that the default Urls provided by mainsail and fluidd point to the right destination. 

For Telegram, please refer to the Readme: https://github.com/mkuf/prind#moonraker-telegram-bot

-Markus

closed 2023-02-14 by mkuf

aguinaldoabbj, 2023-02-14:
> Hi there,
>
> the webcam has to be manually added to whichever frontend you've chosen to use. By default, the ustreamer service is available under the `/webcam` path so that the default Urls provided [...]

@mkuf , thanks for your prompt reply.
So, do I have to put just /webcam in the url stream field?
What about the url snapshot field and the service drop-down (for me this menu doesn't show ustreamer).

[image]