mkuf/prind #63: ARM Support

pauhull, 2023-03-05:
When building the klipperscreen image for ARMv7 this error appears: https://paste.pauhull.de/bukohubama.shell

I could not find any info about if ARM is supported by this stack. If not, is support planned?

mkuf, 2023-03-05:
Hi there, 
Images are currently built and pushed to dockerhub for the following platforms.
https://github.com/mkuf/prind/blob/18435827c499f1d3ee0abab583d7660481171987/scripts/build-images.sh#L17 

Are you trying to cross-build Images for arm on amd64?  
If so, have a look at the command on top of the build script on how to set up the environment.
https://github.com/mkuf/prind/blob/18435827c499f1d3ee0abab583d7660481171987/scripts/build-images.sh#L3-L9

If you are trying to build the image on the machine it is supposed to be run on, please share the output of `docker system info`. Did you change any of the files provided by prind or klipperscreen?

-Markus

pauhull, 2023-03-05:
Thanks, running the lines you mentioned worked, it now builds without errors. I also tried building for ``linux/arm/v6`` and it worked too. Is there a reason for it not to be included in the supported platforms?

mkuf, 2023-03-06:
There haven't been any requests for armv6 images yet but I'm open to add this to the build Script.  
Can you name sbcs that use this platform? As this would increase the build times for the daily builds quite a bit, I'd like to make sure there is a userbase which would benefit from these Images. 

-Markus

pauhull, 2023-03-06:
The RPi Zero and Zero W still run on armv6. So far Docker works well on my Zero W with 2 containers, but I'm not so sure how well the performance will be when you run more.

mkuf, 2023-03-06:
I just did a test and ran into an issue with the ustreamer image, where it does seem that the debian base image is not available for armv6. This requires some changes and probably a little research.  
https://github.com/mkuf/prind/actions/runs/4347881240/jobs/7595727259  

If you're willing to build/test/patch the Dockerfiles for ARMv6, I'm happy to accept a PR, otherwise it will probably take some more Weeks until I can have a closer look. 

-Markus

closed 2023-04-07 by mkuf

mkuf, 2023-04-07:
ARMv6 support has been added in https://github.com/mkuf/prind/pull/70  
Baseimages for the klipper mcu image and ustreamer had to be replaced.  

Current Images have been rebuilt to contain support for ARMv6.  
All builds after the following tags will support ARMv6
|Name|Tag|
|---|---|
|mkuf/klipper|[fec7ddd](https://hub.docker.com/layers/mkuf/klipper/fec7ddd/images/sha256-c35ed50bfc707149d41b5922c6def382b58db019ecb912ae14a44b85809233fa)|
|mkuf/moonraker|[31e589a](https://hub.docker.com/layers/mkuf/moonraker/31e589a/images/sha256-cf580305cb50b2506336ff3dc4503b86fe64c256348904a8432844b721b42a92)|
|mkuf/klipperscreen|[a1c602b](https://hub.docker.com/layers/mkuf/klipperscreen/a1c602b/images/sha256-3704bca307df58d7db346970a214e59e8479f8037e8ea108c5aa4260024c5405)|
|mkuf/ustreamer|[28c8599](https://hub.docker.com/layers/mkuf/ustreamer/28c8599/images/sha256-4bdefd6b27bffa44c94a9d005a273cc886e07baa254c04937d66cc6c5ee16f0e)|

-Markus