mkuf/prind #64: could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:

GAZ082, 2023-03-11:
Hey there. What permissions should i have in both, the metal and docker to let Klipper access the device?

Thanks!

GAZ082, 2023-03-11:
I had to add the host user to the uucp group, which is the archbian equivalent to dialout.

GAZ082, 2023-03-11:
Na, that did not help. What im doing now is setting the permission logging as root in the klipper container, but wonder if this is the solution.

GAZ082, 2023-03-11:
Wonder if i need to map uucp as dialout in the docker file...

GAZ082, 2023-03-13:
I found a solution, which is logging in to the Klipper docker as root and manually changing the group of the dev to dialout.

I'm trying to see how to accomplish this in the docker-compose file, but would be nice to:
1. Fix this in the compose files.
2. If there is nothing to fix (I'm no docker expert :) ), just add a troubleshoot section with common problems to the main README.

GAZ082, 2023-03-18:
@Abikebuk awesome I was helpful!  Spent several hours on this :)

Im also using Arch, but in the form of ArchlinuxARM. I made a little script to set the permissions **after** the container is running:

```#!/bin/bash

serialdev="/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"
id=$(docker ps | grep klipper | cut -d' ' -f1)
docker exec -ti -u 0 $id /bin/bash -c "chown root:dialout $serialdev"
```

Pipshag, 2023-03-30:
This might not be the best solution, but I normally add a udev rule on the host, for the USB-device, granting access. At the same time, I add a fixed device path (think /dev/usbRUMBA) to make life easier.

GAZ082, 2023-03-30:
> This might not be the best solution, but I normally add a udev rule on the host, for the USB-device, granting access. At the same time, I add a fixed device path (think /dev/usbRUMBA) to make life easier.

Thanks Pip, thats a good one to try, should work after every reboot. Until this gets fixed in the docker-compose script.

GAZ082, 2023-03-31:
@Pipshag 
Its not working.
Created a rule in the host but when you check the  docker it gets assigned with groupID instead of plain
crw-rw---- 1 root  986 188, 0 Mar 31 01:08 ender6

I have to specifilcally set it to dialout with my script:

```
#!/bin/bash

serialdev="/dev/ender6"
id=$(docker ps | grep klipper | cut -d' ' -f1)
docker exec -ti -u 0 $id /bin/bash -c "chown root:dialout $serialdev"
```

crw-rw---- 1 root dialout 188, 0 Mar 31 01:08 ender6

mkuf, 2023-04-07:
Please run the following commands on your host before any permissions have been modified. 
This gives the named and numeric groupids that may help us in finding a solution for this. 
```
ls /dev
ls -n /dev

docker exec $(docker ps -aqf "label=org.prind.service=klipper") ls /dev
docker exec $(docker ps -aqf "label=org.prind.service=klipper") ls -n /dev
```
---
> @Pipshag Its not working. Created a rule in the host but when you check the docker it gets assigned with groupID instead of plain crw-rw---- 1 root 986 188, 0 Mar 31 01:08 ender6

Without any further info, the solution proposed by @Pipshag makes most sense to me, as long as you use numeric groupids in the udev rule, not named groupids.  
Creating a udev rule that sets the groupid for the device to `20` should have the desired effect.

-Markus

GAZ082, 2023-04-07:
Looks like using numbers instead of text worked:

`ACTION=="add",SUBSYSTEM=="tty",ATTRS{idVendor}=="1a86",ATTRS{idProduct}=="7523",SYMLINK+="ender6",GROUP="20"`

This code is for a CUBOXi4-Pro.

Dont know if you Markus can implement something automated or its not necessary, but looks like at least two already had to do some workarounds with permissions. At the very minimum we need another bullet point in the troubleshooting section of the Readme.

mkuf, 2023-04-07:
Could you share the output of the commands I mentioned?  
Please disable the udev rule beforehand and re-attach your printer, so that permissions for the device are reset.

Without any further info I can not assess if a automated solution can be implemented into the project.

GAZ082, 2023-04-07:
```
[alarm@alarm dev]$ ls /dev
autofs           gpiochip1     loop1      port      stderr  tty19  tty32  tty46  tty6     ttymxc0       v4l-subdev3  vcsa1        video0
block            gpiochip2     loop2      ppp       stdin   tty2   tty33  tty47  tty60    ttymxc3       v4l-subdev4  vcsa2        video1
btrfs-control    gpiochip3     loop3      pps0      stdout  tty20  tty34  tty48  tty61    ubi_ctrl      v4l-subdev5  vcsa3        video2
bus              gpiochip4     loop4      ptmx      tty     tty21  tty35  tty49  tty62    uhid          v4l-subdev6  vcsa4        video3
cec0             gpiochip5     loop5      ptp0      tty0    tty22  tty36  tty5   tty63    uinput        v4l-subdev7  vcsa5        video4
char             gpiochip6     loop6      pts       tty1    tty23  tty37  tty50  tty7     urandom       v4l-subdev8  vcsa6        video5
console          hwrng         loop7      random    tty10   tty24  tty38  tty51  tty8     userfaultfd   v4l-subdev9  vcsu         video6
cpu_dma_latency  i2c-1         mapper     rfkill    tty11   tty25  tty39  tty52  tty9     v4l           vcs          vcsu1        video7
cuse             i2c-2         media0     rtc       tty12   tty26  tty4   tty53  ttyS0    v4l-subdev0   vcs1         vcsu2        video8
disk             input         mem        rtc0      tty13   tty27  tty40  tty54  ttyS1    v4l-subdev1   vcs2         vcsu3        watchdog
dri              kmsg          mmcblk1    rtc1      tty14   tty28  tty41  tty55  ttyS2    v4l-subdev10  vcs3         vcsu4        watchdog0
fd               lirc0         mmcblk1p1  serial    tty15   tty29  tty42  tty56  ttyS3    v4l-subdev11  vcs4         vcsu5        zero
full             log           mqueue     shm       tty16   tty3   tty43  tty57  ttyS4    v4l-subdev12  vcs5         vcsu6
fuse             loop-control  net        snapshot  tty17   tty30  tty44  tty58  ttyS5    v4l-subdev13  vcs6         vga_arbiter
gpiochip0        loop0         null       snd       tty18   tty31  tty45  tty59  ttyUSB0  v4l-subdev2   vcsa         vhci
```
```
[alarm@alarm dev]$ ls -n /dev
total 0
crw-r--r-- 1 0   0  10, 235 Apr  2 15:46 autofs
drwxr-xr-x 2 0   0      240 Mar  9 19:36 block
crw-rw---- 1 0 995  10, 234 Apr  2 15:46 btrfs-control
drwxr-xr-x 3 0   0       60 Jan  1  1970 bus
crw-rw---- 1 0 985 251,   0 Apr  2 15:46 cec0
drwxr-xr-x 2 0   0     3480 Apr  2 15:46 char
crw------- 1 0   0   5,   1 Apr  2 15:47 console
crw------- 1 0   0  10, 125 Apr  2 15:46 cpu_dma_latency
crw------- 1 0   0  10, 203 Mar  9 19:36 cuse
drwxr-xr-x 7 0   0      140 Jan  1  1970 disk
drwxr-xr-x 3 0   0      120 Apr  2 15:46 dri
lrwxrwxrwx 1 0   0       13 Jan  1  1970 fd -> /proc/self/fd
crw-rw-rw- 1 0   0   1,   7 Apr  2 15:46 full
crw-rw-rw- 1 0   0  10, 229 Apr  2 15:46 fuse
crw------- 1 0   0 254,   0 Apr  2 15:46 gpiochip0
crw------- 1 0   0 254,   1 Apr  2 15:46 gpiochip1
crw------- 1 0   0 254,   2 Apr  2 15:46 gpiochip2
crw------- 1 0   0 254,   3 Apr  2 15:46 gpiochip3
crw------- 1 0   0 254,   4 Apr  2 15:46 gpiochip4
crw------- 1 0   0 254,   5 Apr  2 15:46 gpiochip5
crw------- 1 0   0 254,   6 Apr  2 15:46 gpiochip6
crw------- 1 0   0  10, 183 Apr  2 15:46 hwrng
crw------- 1 0   0  89,   1 Apr  2 15:46 i2c-1
crw------- 1 0   0  89,   2 Apr  2 15:46 i2c-2
drwxr-xr-x 3 0   0      100 Apr  2 15:46 input
crw-r--r-- 1 0   0   1,  11 Apr  2 15:46 kmsg
crw------- 1 0   0 237,   0 Apr  2 15:46 lirc0
lrwxrwxrwx 1 0   0       28 Mar  9 19:36 log -> /run/systemd/journal/dev-log
crw-rw---- 1 0 995  10, 237 Apr  2 15:46 loop-control
brw-rw---- 1 0 995   7,   0 Mar  9 19:36 loop0
brw-rw---- 1 0 995   7,   1 Mar  9 19:36 loop1
brw-rw---- 1 0 995   7,   2 Mar  9 19:36 loop2
brw-rw---- 1 0 995   7,   3 Mar  9 19:36 loop3
brw-rw---- 1 0 995   7,   4 Mar  9 19:36 loop4
brw-rw---- 1 0 995   7,   5 Mar  9 19:36 loop5
brw-rw---- 1 0 995   7,   6 Mar  9 19:36 loop6
brw-rw---- 1 0 995   7,   7 Mar  9 19:36 loop7
drwxr-xr-x 2 0   0       60 Mar  9 19:36 mapper
crw-rw---- 1 0 985 252,   0 Apr  2 15:46 media0
crw-r----- 1 0 993   1,   1 Apr  2 15:46 mem
brw-rw---- 1 0 995 179,   0 Mar  9 19:36 mmcblk1
brw-rw---- 1 0 995 179,   1 Mar  9 19:36 mmcblk1p1
drwxrwxrwt 2 0   0       40 Jan  1  1970 mqueue
drwxr-xr-x 2 0   0       60 Mar  9 19:36 net
crw-rw-rw- 1 0   0   1,   3 Apr  2 15:46 null
crw-r----- 1 0 993   1,   4 Apr  2 15:46 port
crw------- 1 0   0 108,   0 Mar  9 19:36 ppp
crw------- 1 0   0 250,   0 Apr  2 15:46 pps0
crw-rw-rw- 1 0   5   5,   2 Apr  2 15:50 ptmx
[... 160 lines cut ...]
tty
urandom
zero

```
```
[alarm@alarm prind]$ docker exec $(docker ps -aqf "label=org.prind.service=klipper") ls -n /dev
total 0
lrwxrwxrwx 1 0 0   13 Apr  2 15:52 fd -> /proc/self/fd
crw-rw-rw- 1 0 0 1, 7 Apr  2 15:52 full
drwxrwxrwt 2 0 0   40 Apr  2 15:52 mqueue
crw-rw-rw- 1 0 0 1, 3 Apr  2 15:52 null
lrwxrwxrwx 1 0 0    8 Apr  2 15:52 ptmx -> pts/ptmx
drwxr-xr-x 2 0 0    0 Apr  2 15:52 pts
crw-rw-rw- 1 0 0 1, 8 Apr  2 15:52 random
drwxr-xr-x 3 0 0   60 Apr  2 15:52 serial
drwxrwxrwt 2 0 0   40 Apr  2 15:52 shm
lrwxrwxrwx 1 0 0   15 Apr  2 15:52 stderr -> /proc/self/fd/2
lrwxrwxrwx 1 0 0   15 Apr  2 15:52 stdin -> /proc/self/fd/0
lrwxrwxrwx 1 0 0   15 Apr  2 15:52 stdout -> /proc/self/fd/1
crw-rw-rw- 1 0 0 5, 0 Apr  2 15:52 tty
crw-rw-rw- 1 0 0 1, 9 Apr  2 15:52 urandom
crw-rw-rw- 1 0 0 1, 5 Apr  2 15:52 zero

```

GAZ082, 2023-04-07:
I think you will also want this:

```
[alarm@alarm prind]$ docker exec $(docker ps -aqf "label=org.prind.service=klipper") ls -n /dev/serial/by-id
total 0
crw-rw---- 1 0 986 188, 0 Apr  2 15:52 usb-1a86_USB_Serial-if00-port0
```

mkuf, 2023-04-07:
Ok, so there is nothing I can do in docker or compose that could fix the mismatching groupids. This would inevitably break if someone uses another distro with completely different groupids again. 

Creating a udev rule on the host is the simplest solution to fix this. 
I added documentation about this to the readme in https://github.com/mkuf/prind/commit/7ca1fc22e512c40366c83a26dff08b24a27d916b

-Markus

closed 2023-04-07 by mkuf