mkuf/prind #67: Mainsail image not using "latest" but "edge" as tag.

GAZ082, 2023-03-13:
This may be a feature, but the documentation of this repo specifically says that all images used in the docker compose file are "latest".

mkuf, 2023-04-07:
This is intentional.  
The readme refers to images built by this project, not external projects. 

https://github.com/mkuf/prind/blob/510c9ee9db6d085670e71b5895fad131688f1c56/README.md?plain=1#L134-L136

-Markus

closed 2023-04-07 by mkuf

GAZ082, 2023-04-07:
Sorry but i don't quiet get it. All the rest of the containers are "latest" but mainsail.

mkuf, 2023-04-07:
Not sure what you mean by *all the rest of the containers*. 😄    
Octoprint uses `minimal`, moonraker-telegram-bot uses `development` and traefik `v2.5`.   

Mainsail using `edge` is so that always the most recent build is used. Have a look at the mainsail repo and how they handle image tags, as this is not under my control.

GAZ082, 2023-04-08:
I may be wrong, I seldom use Docker, but if you check the docker-compose you can see how the containers used are latest:

https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L3
https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L47
https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L82 (minimal here)
https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L103
https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L115
https://github.com/mkuf/prind/blob/7ca1fc22e512c40366c83a26dff08b24a27d916b/docker-compose.yaml#L129

Mainsail has a "latest", why not use it? Again, I may be completely wrong and I should not check this file for that for example.

mkuf, 2023-04-08:
> Mainsail has a "latest", why not use it? Again, I may be completely wrong and I should not check this file for that for example.

I already stated the reason for why `edge` is used in my previous comment.

> Mainsail using `edge` is so that always the most recent build is used. Have a look at the mainsail repo and how they handle image tags, as this is not under my control.

Have a look at the image build action for mainsail: https://github.com/mainsail-crew/mainsail/blob/df83c9d8cc61ecd11c320ec90a9083fdd675d3bf/.github/workflows/publish_docker.yml#L40-L42
You can see that latest is tracking the master branch, while edge is tracking the develop branch.  
To always get the most recent code from develop, the edge image is referenced in the compose file.  

Images built by this repo (klipper, moonraker, klipperscreen and ustreamer) are tagged latest, as they are built on a daily basis and also track the most recent upstream branch.