mkuf/prind #74: Question: how to check version of each component within Mainsail.

GAZ082, 2023-05-29:
Hello! Is there any way? The little icons in Mainsail just show a question mark in the version number.

Thanks!

closed 2023-06-01 by mkuf

mkuf, 2023-06-01:
What you can do however is checking the version of the docker containers.  
The label `org.prind.image.version` contains the upstream sha that represents the commit of the corresponding repository. 

This commands lists the service, image and `org.prind.image.version` for all prind services.
```
  for container in $(docker ps -aqf "label=org.prind.service"); do
    echo "$(docker inspect --format '{{ index .Config.Labels "org.prind.service" }}' ${container}): $(docker inspect --format '{{ index .Config.Image }}' ${container}) $(docker inspect --format '{{ index .Config.Labels "org.prind.image.version"}}' ${container})"
  done
```

Output:
```
moonraker: mkuf/moonraker:latest v0.8.0
klipper: mkuf/klipper:latest 37315bf3365484c3ba6adef2b7da2b4464980e8a
traefik: traefik:v2.5 
klipperscreen: mkuf/klipperscreen:latest 059f1a8026588090d0ea205b02c0d0a9a8e2b80a
mobileraker_companion: ghcr.io/clon1998/mobileraker_companion:latest 
mainsail: ghcr.io/mainsail-crew/mainsail:edge 
init: busybox:latest 
webcam: mkuf/ustreamer:latest dde2190ac9b96795e58908a6b57b06b2842b1010
```

mkuf, 2024-11-13:
Versions are now correctly displayed with https://github.com/mkuf/prind/commit/2472a2815c2e5a9ecf2fb6b42532e00bc0d5ce09 in the latest klipper and moonraker images. 

[image]