mkuf/prind #75: Add docker start timeout.

GAZ082, 2023-06-04:
When klipper cannot detect my USB device of any other reason that may not allow the successful completion of the init process the container just get stuck in "starting".

Would be nice to add a timeout (60s) for these cases, where the start process just stops and allows the user to regain control of the CLI.

Thanks.

mkuf, 2023-06-06:
I'm not sure if I understand you correctly.  
Which container is stuck in starting state?
If the referenced device is unavailable, the klipper container won't start at all, a timeout wouldn't take any affect here.

GAZ082, 2023-06-06:
Well, the issue here is that the Klipper container if all is good takes like 9 seconds to boot in my SBC and if the USB is not available it just keeps going and going...(stopped waiting at about 200 seconds).

mkuf, 2023-06-06:
9 seconds sounds quite excessive for the klipper container alone.  

Nevertheless, this would have to be solved on the docker side, as the code inside the container is not running when the container is in the startup state.   

Docker does not provide any way to set a timeout for a container, wether it is at startup or runtime.  
You may be able to add a healthcheck to the klipper service in your override file, but I'm not sure if the checks are taken into account if the container is still in starting state. 

https://github.com/compose-spec/compose-spec/blob/master/spec.md#healthcheck

closed 2023-06-10 by mkuf