mkuf/prind #76: How to add Python script file to the Klipper container klipper/extras folder?

GAZ082, 2023-06-04:
Hi. I'd like to install the gcode to shell script. How can I add it to the klippy/extras folder inside the container?

Thanks.

mkuf, 2023-06-06:
You can add files to the container via volumes, just like with all the other files that get mounted into the Containers. 

This can be achieved via `docker-compose.override.yaml`, but be sure to include all volumes that are defined for the klipper service in `docker-compose.yaml`. You may then add a new volume at the end of the volumes list as shown below: 

```yaml
services:
  klipper:
    volumes:
      - ./config:/opt/printer_data/config
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - /path/to/my/custom/plugins:/opt/klipper/klippy/extras
```

GAZ082, 2023-06-08:
This is my whole klipper block in the override file:

```
services:
  klipper:
    privileged: true
    volumes:
      - ./config:/opt/printer_data/config
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - /home/alarm/prind/custom/plugins:/opt/klipper/klippy/extras
    devices:
      - dev/serial/by-id/usb-1a86_USB_Serial-if00-port0:dev/ttyUSB0
```
If i add the plugins folder I get the error:

`Unable to load module : stepper_enable`

The plugins folder in my host is empty BTW. Enabled and disabled the line several times to verify this, happened 100% of times.

mkuf, 2023-06-08:
That makes sense, as I did not check if the directory already contains files, when writing the example.  
When mounting anything into a container, you overwrite existing files. In your case, you mount an empty directory and therefore remove all existing files that klippy needs for operation.

You'll have to mount the single file instead of the whole directory to not remove any existing files.  
Replace `myplugin.py` with the actual Filename you want to mount.

```
- /home/alarm/prind/custom/plugins/myplugin.py:/opt/klipper/klippy/extras/myplugin.py
```