mkuf/prind #77: No USB connection after printer restart

boehser-enkel, 2023-06-06:
Hi,

when my printer is turned on again klipper is not able to connect anymore. I have to restart the whole system to be able to connect again.
Also after a restart i have to take down the stack and start it again. The container klipper-klipper-1 is not started and i am not able to connect when i start it manually.

System: Raspbi 4
Connection via USB
Printer: Neptune 3 Pro

Greetings

GAZ082, 2023-06-06:
This is happening to me as well. Just coincidentally I had duct-taped the 5v line of the USB connector and though it was related, but looks like its not.

Perhaps this is a klipper issue and not a prind one.

mkuf, 2023-06-06:
This is usually the case when the serial device changes after a printer reset (e.g. from ttyACM0 to ttyACM1) and can be mitigated by running klipper in privileged mode via your `docker-compose.override.yaml` like so. 

```yaml
services:
  klipper:
    privileged: true
```

If that does not fix your issue, please run `scripts/get-info.sh` when this is occuring again and upload the generated archive here.

boehser-enkel, 2023-06-07:
The device does not seem to change. I don't need to update the entry (/dev/ttyUSB0). Also privileged mode does not help
 

> ## System Info 
Client: Docker Engine - Community
 Version:    24.0.2
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.10.5
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.18.1
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 8
  Running: 6
  Paused: 0
  Stopped: 2
 Images: 21
 Server Version: 24.0.2
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd                                                                                                                                                                                            
 Cgroup Version: 2                                                                                                                                                                                                 
 Plugins:                                                                                                                                                                                                          
  Volume: local                                                                                                                                                                                                    
  Network: bridge host ipvlan macvlan null overlay                                                                                                                                                                 
WARNING: No memory limit support                                                                                                                                                                                   
WARNING: No swap limit support                                                                                                                                                                                     
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog                                                                                                                              
 Swarm: inactive                                                                                                                                                                                                   
 Runtimes: io.containerd.runc.v2 runc                                                                                                                                                                              
 Default Runtime: runc                                                                                                                                                                                             
 Init Binary: docker-init                                                                                                                                                                                          
 containerd version: 3dce8eb055cbb6872793272b4f20ed16117344f8                                                                                                                                                      
 runc version: v1.1.7-0-g860f061                                                                                                                                                                                   
 init version: de40ad0                                                                                                                                                                                             
 Security Options:                                                                                                                                                                                                 
  seccomp                                                                                                                                                                                                          
   Profile: builtin                                                                                                                                                                                                
  cgroupns                                                                                                                                                                                                         
 Kernel Version: 6.1.21-v8+                                                                                                                                                                                        
 Operating System: Debian GNU/Linux 11 (bullseye)                                                                                                                                                                  
 OSType: linux                                                                                                                                                                                                     
[... cut ...]
/tmp/tmp.kFyTVrez0q-prind/klipper/config/printer-20230606_200810.cfg                                                                                                                                               
/tmp/tmp.kFyTVrez0q-prind/logs/                                                                                                                                                                                    
/tmp/tmp.kFyTVrez0q-prind/logs/klippy.log                                                                                                                                                                          
/tmp/tmp.kFyTVrez0q-prind/logs/moonraker.log

boehser-enkel, 2023-06-07:
Okay if i set the device up as ttyUSB* it changes after a reconnect. Also privileged does not change anything.
I also tried /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-port0 but the container does not start with that

boehser-enkel, 2023-06-08:
> ```yaml
> services:
>   klipper:
>     privileged: true
>     volumes:
>       - /dev:/dev
>       - ./config:/opt/printer_data/config
>       - run:/opt/printer_data/run
>       - gcode:/opt/printer_data/gcodes
>       - log:/opt/printer_data/logs
> ```

This - sir - works like a charm :)
Just hit restart in klipper after the printer is turned on and everything is ready to go.
Thanks!

mkuf, 2023-06-08:
With the merge of https://github.com/mkuf/prind/pull/78 this is now in main.  
On the next release you may remove the klipper service completely from your `docker-compose.override.yaml` and still be able to reconnect to your printer after a reset.

GAZ082, 2023-06-08:
@mkuf is it sensible to allow docker run in privilege mode? If i understand correctly, this is basically giving it root access.

mkuf, 2023-06-08:
Running a Container in privileged mode grants access to the hosts devices, **it does not grant root access to the host**.  Klippy is still run by an unprivileged user within the container, just as if it was installed directly on the host.  
  
See the docker docs for reference: https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities

As klipper is hardware (=device) dependent it makes sense to run it in privileged mode.  
Some usecases like the octoprint profile and simulavr stack already run klipper in privileged mode as it is required for operation. The upcoming hostmcu service will also profit from this change.  

From a usability standpoint this is a great improvement, as Users only have to set their serial device in the printer.cfg and are not forced to interact with compose files.

GAZ082, 2023-06-08:
Using this mode will render obsolete the rule for defining the group of the dev for those hosts not based on Debian?

mkuf, 2023-06-08:
As klipper is still run by an unprivileged user within the container, it is still necessary to have the right permissions for the device it is trying to access. So the udev rule for matching the group access to the device is required if groupids do not match between the host an the klipper container, as described in the readme.

https://github.com/mkuf/prind/blob/21f9b51af4b5aa1b94cc906828446b25b5244f06/README.md?plain=1#L144-L156

GAZ082, 2023-06-10:
Im still having the issue. I added volumes in the override config file, do i need to add all the sections of Klipper in that file, thus including the privilege or it picks it up from the original docker compose? @mkuf

mkuf, 2023-06-10:
Please run `scripts/get-info.sh` and upload the generated **archive** here. @GAZ082

GAZ082, 2023-06-11:
@mkuf 
For some reason GitHub rejects my files. Here https://we.tl/t-yeNNclz0sX

GAZ082, 2023-06-11:
BTW, the printer is currently turned off.

mkuf, 2023-06-11:
There is not much sense in trying to troubleshoot the issue if the printer is turned off 😄 

From the klippy log, /dev/ttyUSB0 is missing. That makes sense, when the printer is turned off.  

You'll have to use the symlink provided by klipper under `/dev/serial/by-id`, as ttyUSB0 may change to ttyUSB1 on printer reset, as mentioned in the comment before

> This is usually the case when the serial device changes after a printer reset (e.g. from ttyACM0 to ttyACM1) and can be mitigated by running klipper in privileged mode via your `docker-compose.override.yaml` like so.

mkuf, 2023-06-13:
Any update @GAZ082?

PR #78 (2023-06-08): Support device change
This should fix #77

Changes:
* sets `privileged: true` for klipper service
* mounts `/dev` to klipper container by default
* updates docs accordingly

Removes:
* klipper-priv