mkuf/prind #80: Libcamera (Pi Cam V3) support?

boehser-enkel, 2023-06-08:
Hi again :),

is it possible to support libcamera for e.g. Pi cam V3?

greetings

mkuf, 2023-06-08:
Could you elaborate on what you'd like to be supported?  
I'm not familiar with libcamera not with Pi Cam V3.  

Can't the camera be used with ustreamer or do you expect other functionality?

boehser-enkel, 2023-06-08:
Pi Cam V3 needs libcamera to work (and Bullseye i think).
I looked at ustreamer and there is at least an issue with a feature request for the new cam. 
So we have to wait for that?

mkuf, 2023-06-08:
Yes, we'll have to wait for ustreamer to support this.  
You may link the issue you found here so we can keep track of it and update the dockerfile for ustreamer if necessary.

boehser-enkel, 2023-06-08:
https://github.com/pikvm/ustreamer/issues/212

boehser-enkel, 2023-06-09:
> Yes, we'll have to wait for ustreamer to support this. You may link the issue you found here so we can keep track of it and update the dockerfile for ustreamer if necessary.

> Hi. I've got a camera and experimented with it. Too much work will be needed for native support, but Raspberry offers "libcamerify", which works fine with uStreamer. On Raspberry OS, install libcamera-tools and run as follows: libcamerify ./ustreamer -r 1920x1080 --encoder=m2m-image.

mkuf, 2023-06-09:
I don't think there is a way to get this work within the container, as the required packages are only available for raspberrypios.    
We could try to mount the `libcamerify` binary and necessary libraries into the container.

I do not own a raspberry pi nor a camera 3 where I could verify if this approach works, so you'd have to do some testing if you want this to work with prind. 

Alternatively, you may install ustreamer to your host and remove the webcam service from your override file.

boehser-enkel, 2023-07-19:
I ran another container to stream the camera signal (so it's possible at all?)
I can open the stream via VLC in my network but not inside the klipper container. Why is that?

mkuf, 2023-07-19:
Hey, 
I'm not familiar with the Image you mention.  
What do you mean that klipper can not access it? Usually, you'll have to add the streaming link to the frontend (Mainsail, Fluidd, etc.) to view the stream. Maybe the container is able to output different formats, from which one is compatible with mjpeg?

Crowsnest currently does not provide a docker image and prind already provides a solution for multiple webcams described here: https://github.com/mkuf/prind#multiple-webcams.

boehser-enkel, 2023-07-19:
Yes i treid to configure it inside mainsail. No picture is shown
I tried with one methid via rtsp and one via http link

mkuf, 2023-07-24:
You'll have to be able to access the WebRTC stream via your browser as described here: 
https://github.com/bluenviron/mediamtx#web-browsers-1
If this does not work, you'll have to get in touch with the mediamtx project, I can't assist you with that. 

Also, make sure to correctly set up the webcam in your frontend, using the WebRTC profile and the port/path combination of the container
[image]

boehser-enkel, 2023-07-26:
> You'll have to be able to access the WebRTC stream via your browser as described here: https://github.com/bluenviron/mediamtx#web-browsers-1 If this does not work, you'll have to get in touch with [...]

Yes i am able to access the stream (Browser or VLC). 
Only Mainsail is not able to access it (via the same URL)

boehser-enkel, 2023-09-11:
I Installed https://github.com/ayufan/camera-streamer/ on my host an can access it via browser on my RPI (http://IP:Port & /stream, /snapshot) but i am not able to access it inside mainsail for streaming

mkuf, 2023-09-13:
You'll have to get in touch with the maintail community/devs to get this sorted out.  
At this point, prind is not involved anymore and I can't help you with troubleshooting external software.