mkuf/prind #82: moonraker complains about dbus_manager

stockholmux, 2023-06-27:
The Fluidd UI gives an error message of:

```
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
dbus_manager
```

The logs show this:

```
023-06-27 23:27:14,117 [server.py:_initialize_component()] - Component [dbus_manager] failed post init
Traceback (most recent call last):
  File "/opt/moonraker/moonraker/server.py", line 222, in _initialize_component
    await ret
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 65, in component_init
    self.polkit = await self.get_interface(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 109, in get_interface
    ret = await self.get_interfaces(bus_name, bus_path,
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker/moonraker/components/dbus_manager.py", line 121, in get_interfaces
    introspection = await self.bus.introspect(bus_name, bus_path)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 216, in introspect
    return await asyncio.wait_for(future, timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/message_bus.py", line 195, in reply_notify
    BaseMessageBus._check_method_return(reply, err, 's')
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/message_bus.py", line 611, in _check_method_return
    raise err
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 365, in _message_reader
    if self._unmarshaller.unmarshall():
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 304, in unmarshall
    self._unmarshall()
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 243, in _unmarshall
    self.read(16, prefetch=True)
  File "/opt/venv/lib/python3.11/site-packages/dbus_next/_private/unmarshaller.py", line 91, in read
    raise EOFError()
EOFError
2023-06-27 23:27:14,117 [server.py:add_warning()] - Component 'dbus_manager' failed to load with error: 
```

Seems like others are having the same problem: https://github.com/Arksine/moonraker/issues/653 and Arksine indicates it's possibly the newer version of Python.

mkuf, 2023-07-01:
Could you provide arksine with a full log file that was requested in the issue you linked?

From the top of my head, apart from the python version mentioned in the issue, there could be multiple other causes for this behaviour

* dbus not accessible via moonraker, despite being `/run/dbus` being mounted in the container
* missing dependencies in the moonraker image
* ...

mkuf, 2023-07-24:
This should be fixed by upstream https://github.com/Arksine/moonraker/issues/653#issuecomment-1646681131, can you confirm this?

mkuf, 2023-08-06:
Closing because of missing activity. 
Feel free to reach out if you have any new insights.

closed 2023-08-06 by mkuf