mkuf/prind #86: Pulling containers does not update Mainsail.

GAZ082, 2023-07-29:
Hello!

Doing docker compose pull and then restarting does not seem to update Mainsail (2.5.1 is the one in this stack. There are two new versions since).

Should i do anything else?

mkuf, 2023-07-29:
What exactly did you execute to pull the new images? 2.5.1 is not set in the stack but was the current version, when you first pulled the images/started the stack.
Make sure to add the approriate profiles while pulling the new images. As the compose file is using the `edge` tag for mainsail, you'll always get the latest version, unless you explicitly pinned that image to a specific version in your override file.  

```
root@sparkcube:/opt/prind# docker compose --profile mainsail --profile klipperscreen --profile mobileraker_companion pull
[+] Running 27/27
 ⠿ klipper Pulled                                                                                                                                                                                                                                                       1.9s 
 ⠿ webcam Pulled                                                                                                                                                                                                                                                        1.8s 
 ⠿ moonraker Pulled                                                                                                                                                                                                                                                     1.8s 
 ⠿ mainsail Pulled                                                                                                                                                                                                                                                      6.8s 
   ⠿ edb6bdbacee9 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ 4c7f12338fe3 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ 002a136ea5c5 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ 6d407d2ad632 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ d1543f6e84d3 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ ad428fb17e98 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ bacb1bf71fa0 Already exists                                                                                                                                                                                                                                        0.0s 
   ⠿ 05fb56ef846f Pull complete                                                                                                                                                                                                                                         0.9s 
   ⠿ 7f339b6686e4 Pull complete                                                                                                                                                                                                                                         4.9s 
 ⠿ init Pulled                                                                                                                                                                                                                                                          7.1s 
   ⠿ 8a0af25e8c2e Pull complete                                                                                                                                                                                                                                         4.6s 
 ⠿ traefik Pulled                                                                                                                                                                                                                                                       1.7s 
 ⠿ klipperscreen Pulled                                                                                                                                                                                                                                                99.9s 
   ⠿ 3ae0c06b4d3a Pull complete                                                                                                                                                                                                                                        29.6s 
[... cut ...]
 ⠿ Container prind-klipper-1                Started                                                                                                                                                                                                                    18.5s
 ⠿ Container prind-moonraker-1              Started                                                                                                                                                                                                                     9.8s
```

[image]

GAZ082, 2023-07-29:
I just ran:
docker compose pull

Will try specifying mainsail.

GAZ082, 2023-07-29:
Setting the profile mainsail made the trick. Thanks.

closed 2023-07-29 by GAZ082