mkuf/prind #87: Klipper unable to connect to printer

rafael-aba, 2023-08-06:
Hi, 

I've followed everything seemly correctly however I'm still unable to connect to the printer via usb cable. 

After connecting the usb cable, I don't see the /dev/serial 
Tried restarting both printer and board.

Info dump:
System: Orange pi 5 plus
Connection via USB
Printer: Ender 3 v2
os: https://github.com/Joshua-Riek/ubuntu-rockchip

[image]

get-info.sh:
[info.txt](https://github.com/mkuf/prind/files/12268835/info.txt)

____
I`m still trying to re-do some steps like generating the klipper firmware, but I do think I am doing everything correctly.
Thanks,
Me.

mkuf, 2023-08-06:
Hi there, 

If you don't see a klipper device, the firmware flashing probably did not work.  
Please share the full commands and their output that you used to build the firmware and flash your board. 

Also, please upload the tar.gz file generated by `get-info.sh`, the scripts output alone does not show your full setup. 

-Markus

rafael-aba, 2023-08-06:
Alright, this is the .tar.gz file: 

https://u.pcloud.link/publink/show?code=XZMhTNVZQLKlbdU8VLYhaFsXVJDvmpSR7EtX

And these are the commands for creating the .bin file: 

sudo docker compose -f docker-compose.extra.make.yaml run --rm make menuconfig
--this guides me to the menuconfig, I select the processor, bootloader and communication interface described in the config file for my printer as described at [Klipper example config](https://github.com/Klipper3d/klipper/blob/master/config/printer-creality-ender3-v2-2020.cfg)

output: 
WARN[0000] Found orphan containers ([prind-moonraker-1 prind-klipper-1 prind-webcam-1 prind-mainsail-1 prind-init-1 prind-traefik-1]) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up. 
Loaded configuration '/opt/klipper/.config'
No changes to save (for '/opt/klipper/.config')

after that, this ran 
sudo docker compose -f docker-compose.extra.make.yaml run --rm make 

output: 
make: Nothing to be done for 'all'.

running containers with mainsail return the expected: 
[+] Running 6/6
 ✔ Container prind-init-1       Exited                                                                                                                                                 9.9s 
 ✔ Container prind-webcam-1     Started                                                                                                                                                1.6s 
 ✔ Container prind-traefik-1    Started                                                                                                                                                1.7s 
 ✔ Container prind-mainsail-1   Started                                                                                                                                                1.5s 
 ✔ Container prind-klipper-1    Started                                                                                                                                                9.0s 
 ✔ Container prind-moonraker-1  Started  

With that runinng, I'm fully able to access mainsail from my machine. 

If this is a klipper issue I'll try to bring it to them, so no worries!

Thanks a lot Markus!

--
me

mkuf, 2023-08-06:
Looking at the commands you provided, it seems you successfully built the firmware but did not actually flash it to the printers mainboard, skipping `make flash [...]`.

https://github.com/mkuf/prind/blob/21cdf7efe22f10930764fdd95f9e9baf2b9fb583/README.md?plain=1#L32-L36

I'd assume your printer can be flashed using the `make flash [...]` command, but I'm not familiar with the ender 3 series printers, which might require a different procedure. 

I suggest using the alias for the make command so that you can follow along the official klipper docs. 

-Markus

rafael-aba, 2023-08-06:
Flashing the firmware is via bringing the .bin file to a sd card and turning the printer on with it there. The sd card is working fine for as long as I can test it. 

I saw a thread on another site of someone with a similar issue for a different project, and that it worked with a debian image. So I'm trying that.

rafael-aba, 2023-08-07:
And trying a different iso worked. 

As a said at the first post, I was using https://github.com/Joshua-Riek/ubuntu-rockchip
I switched to their debian iso from [their site](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-plus.html).

I even used the exact same firmware I had flashed to the printer before, but this time the /dev/serial appeared just fine. 

Thank you so much for your help Markus! In the end it was a distro/iso bug. 
Your prind is amazing.

Closing thread.

Thanks,
Me.

closed 2023-08-07 by rafael-aba