mkuf/prind #88: Spaghetti detection

pelaxa, 2023-08-14:
I could not find any related issues to this, but is there a way to enable spaghetti detection in Klipper/Moonraker with this stack?

I found this link (https://obico.io/docs/user-guides/klipper-setup/) but it requires installation which will not work for the docker images, or will it work?

mkuf, 2023-08-15:
Looking at https://github.com/TheSpaghettiDetective/moonraker-obico, the application is just talking to the moonraker api, so if obico would provide a docker image, it could be implemented into the stack. 

Until then, you can probably install the moonraker-obico client directly on your host system, following the instructions you linked.

-Markus

mkuf, 2023-08-20:
Upstream has accepted the proposed changes and is building Container Images.  
PR #89  has been merged and a new Release has been created that contains the new profile: https://github.com/mkuf/prind/releases/tag/v1.9.0  

You may follow the Instructions at https://github.com/mkuf/prind#moonraker-obico to get the service up and running.

closed 2023-08-20 by mkuf

zefir-o, 2024-09-07:
I was trying to use the moonraker-obico profile, however, on the second run, I'm getting the error on klipper startup. 
> Unable to open config file printer_data/config/moonraker_obico_macros.cfg
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

I'm using the following command to start docker containers:
docker compose --profile fluidd --profile moonraker-obico up -d

I also observed, that after klipper restart the link is broken:
lrwxrwxrwx  1 user user 71 Sep  7 21:00 moonraker_obico_macros.cfg -> /opt/moonraker-obico/scripts/../include_cfgs/moonraker_obico_macros.cfg

mkuf, 2024-09-08:
@zefir-o

That sounds to me like you're trying to symlink a file between two containers, which is not possible.   
Klipper can't access any path in the moonraker-obico container and therefore refuses to start.  

To add the macros to klipper, you'll have to copy that file to the `./config` directory instead of creating a symlink.

zefir-o, 2024-09-08:
Hi @mkuf ,

Thank you for suggestion, however I just followed the documentation and run the container specifying obico profile. I don't add any symlink manually, assume the container is adding it. 

My point that the setup described in the documentation doesn't work unfortunately.

mkuf, 2024-09-08:
@zefir-o please run `./scripts/get-info.sh` and share the generated archive, so I can have a closer look.

zefir-o, 2024-09-09:
I think the issue is that the moonraker-obico creates a symlink.
https://github.com/TheSpaghettiDetective/moonraker-obico/blob/19167eba0487d1e6398331e8e39ba10b85da1827/scripts/ensure_include_cfgs.sh#L13

I just manually created a persistent moonraker_obico_macros.cfg file, copied the content to it and the setup works.