mkuf/prind #92: Unable to start simulavr on M1 Mac

Clon1998, 2023-09-01:
Hey,
I wanted to spin up a simulated printer for development. However, I am getting the following error:

```
patrick@MacBook-Pro-von-Patrick prind % docker compose \
  --profile mainsail \
  -f docker-compose.yaml \
  -f docker-compose.extra.simulavr.yaml \
  up -d
[+] Running 1/1
 ! simulavr Warning                                                                                                                                                   6.8s 
[+] Building 19.7s (22/22) FINISHED                                                                                                                                        
 => [internal] load build definition from Dockerfile                                                                                                                  0.0s
 => => transferring dockerfile: 3.30kB                                                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                                     0.0s
 => => transferring context: 2B                                                                                                                                       0.0s
 => [internal] load metadata for docker.io/library/debian:bookworm                                                                                                    2.7s
 => [internal] load metadata for docker.io/library/python:3-bookworm                                                                                                  2.6s
 => [build 1/4] FROM docker.io/library/python:3-bookworm@sha256:521ba7f72b6db00de67973cd478101877eb3c39fb7c07cfa0b7a257d80441b1c                                      0.0s
 => [internal] load build context                                                                                                                                     0.0s
 => => transferring context: 37B                                                                                                                                      0.0s
 => [tools 1/5] FROM docker.io/library/debian:bookworm@sha256:b91baba9c2cae5edbe3b0ff50ae8f05157e3ae6f018372dcfc3aba224acb392b                                        0.0s
 => CACHED [tools 2/5] WORKDIR /opt                                                                                                                                   0.0s
 => CACHED [tools 3/5] RUN apt update  && apt install -y       sudo       virtualenv python3-dev libffi-dev build-essential pkg-config      libncurses-dev       avr  0.0s
 => CACHED [build 2/4] WORKDIR /opt                                                                                                                                   0.0s
 => CACHED [build 3/4] RUN git clone https://github.com/Klipper3d/klipper klipper  && cd klipper  && git checkout master  && rm -rf .git                              0.0s
 => CACHED [build 4/4] RUN python -m venv venv  && venv/bin/pip install -r klipper/scripts/klippy-requirements.txt  && venv/bin/pip install numpy  && venv/bin/pytho  0.0s
 => CACHED [tools 4/5] COPY --from=build /opt/klipper ./klipper                                                                                                       0.0s
 => CACHED [tools 5/5] COPY --from=build /opt/venv ./venv                                                                                                             0.0s
 => CACHED [build-simulavr 1/8] RUN apt update  && apt install -y       git cmake swig       python3-dev  && apt clean                                                0.0s
 => CACHED [build-simulavr 2/8] WORKDIR /opt/klipper                                                                                                                  0.0s
 => CACHED [build-simulavr 3/8] COPY config.simulavr .config                                                                                                          0.0s
 => CACHED [build-simulavr 4/8] RUN make                                                                                                                              0.0s
 => CACHED [build-simulavr 5/8] WORKDIR /opt                                                                                                                          0.0s
 => CACHED [build-simulavr 6/8] RUN mkdir run                                                                                                                         0.0s
 => CACHED [build-simulavr 7/8] RUN groupadd simulavr --gid 1000  && useradd simulavr --uid 1000 --gid simulavr  && usermod simulavr --append --groups dialout        0.0s
 => ERROR [build-simulavr 8/8] RUN git clone git://git.savannah.nongnu.org/simulavr.git  && cd simulavr  && git checkout release-1.1.0  && make python  && make bui  16.9s
------                                                                                                                                                                     
 > [build-simulavr 8/8] RUN git clone git://git.savannah.nongnu.org/simulavr.git  && cd simulavr  && git checkout release-1.1.0  && make python  && make build  && chown -R simulavr:simulavr /opt/simulavr:                                                                                                                                          
#0 0.287 Cloning into 'simulavr'...                                                                                                                                        
#0 16.12 Note: switching to 'release-1.1.0'.                                                                                                                               
#0 16.12 
#0 16.12 You are in 'detached HEAD' state. You can look around, make experimental
#0 16.12 changes and commit them, and you can discard any commits you make in this
#0 16.12 state without impacting any branches by switching back to a branch.
#0 16.12 
#0 16.12 If you want to create a new branch to retain commits you create, you may
#0 16.12 do so (now or later) by using -c with the switch command. Example:
#0 16.12 
#0 16.12   git switch -c <new-branch-name>
#0 16.12 
#0 16.12 Or undo this operation with:
#0 16.12 
#0 16.12   git switch -
#0 16.12 
#0 16.12 Turn off this advice by setting config variable advice.detachedHead to false
#0 16.12 
#0 16.12 HEAD is now at e6eac87 Fix some warnings, clang compiler shows
#0 16.27 -- The C compiler identification is GNU 12.2.0
#0 16.35 -- The CXX compiler identification is GNU 12.2.0
#0 16.36 -- Detecting C compiler ABI info
#0 16.44 -- Detecting C compiler ABI info - done
#0 16.45 -- Check for working C compiler: /usr/bin/cc - skipped
#0 16.45 -- Detecting C compile features
#0 16.45 -- Detecting C compile features - done
#0 16.45 -- Detecting CXX compiler ABI info
#0 16.54 -- Detecting CXX compiler ABI info - done
#0 16.55 -- Check for working CXX compiler: /usr/bin/c++ - skipped
#0 16.55 -- Detecting CXX compile features
#0 16.55 -- Detecting CXX compile features - done
#0 16.57 -- version major 1 / minor 1 / patch 0
#0 16.60 -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.11.2", minimum required is "3.5") 
#0 16.62 -- Found SWIG: /usr/bin/swig4.0 (found version "4.1.0")  
#0 16.63 -- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.11.so (found version "3.11.2") 
#0 16.64 -- can't create man page because of:
#0 16.64 --   - help2man not found
#0 16.64 -- can't create info page because of:
#0 16.64 --   - makeinfo not found
#0 16.66 -- Could NOT find PY_SPHINX (missing: PY_SPHINX) 
#0 16.68 -- Could NOT find PY_RST2PDF (missing: PY_RST2PDF) 
#0 16.81 -- Found PY_BS4: /usr/lib/python3/dist-packages/bs4  
#0 16.83 -- Could NOT find PY_REQUESTS (missing: PY_REQUESTS) 
#0 16.84 -- gtest will be configured
#0 16.84 -- can build debian package
#0 16.84 CMake Error at debian/CMakeLists.txt:7 (message):
#0 16.84   unknown system architecture: aarch64
#0 16.84 
#0 16.84 
#0 16.84 -- package architecture: 
#0 16.85 -- Configuring incomplete, errors occurred!
#0 16.85 See also "/opt/simulavr/build/CMakeFiles/CMakeOutput.log".
#0 16.85 make: *** [Makefile:82: python] Error 1
------
failed to solve: process "/bin/sh -c git clone git://git.savannah.nongnu.org/simulavr.git  && cd simulavr  && git checkout release-1.1.0  && make python  && make build  && chown -R simulavr:simulavr /opt/simulavr" did not complete successfully: exit code: 2
```

mkuf, 2023-09-01:
Hey there, 

simulavr only works on amd64 or i386 machines, see the first 8 lines here as reference: https://git.savannah.nongnu.org/cgit/simulavr.git/tree/debian/CMakeLists.txt?h=release-1.1.0

As simulavr does not seem to be actively maintained anymore, I don't see a way to run this on a different Architecture.  

Maybe there is a more up to date simulator available but support for that would probably have to be implemented to klipper, as upstream is specifically referencing simulavr (https://github.com/Klipper3d/klipper/blob/5f990f93d533247d3a675e8c423280f4333ad8ce/docs/Debugging.md#testing-with-simulavr)

-Markus

closed 2023-09-01 by mkuf

Clon1998, 2023-09-02:
Hey Markus,
thanks for your quick reply.
That's to bad but also what I already though would be the root cause.

Thanks for looking into that!