mkuf/prind #98: [Errno 13] Permission denied: '/dev/pts/1' -> 'printer_data/run/klipper.tty' RPI4

Seredar, 2023-10-08:
docker-compose.yaml

```
services:
  klipper:
    image: mkuf/klipper:latest
    privileged: true
    volumes:
      - /dev:/dev
      - ./config/printer.cfg:/opt/printer_data/config/printer.cfg
      - ./run:/opt/printer_data/run
      - ./gcode:/opt/printer_data/gcodes

  moonraker:
    image: mkuf/moonraker:latest
    ports:
      - "7125:7125"
    volumes:
      - ./config/moonraker.conf:/opt/printer_data/config/moonraker.conf
      - ./run:/opt/printer_data/run
      - ./gcode:/opt/printer_data/gcodes

```
```
pi@mars:~ $ uname -a
Linux mars 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
pi@mars:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

pi-klipper-1    | Traceback (most recent call last):
pi-klipper-1    |   File "/opt/klipper/klippy/klippy.py", line 407, in <module>
pi-klipper-1    |     main()
pi-klipper-1    |   File "/opt/klipper/klippy/klippy.py", line 335, in main
pi-klipper-1    |     start_args['gcode_fd'] = util.create_pty(options.inputtty)
pi-klipper-1    |                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pi-klipper-1    |   File "/opt/klipper/klippy/util.py", line 42, in create_pty
pi-klipper-1    |     os.symlink(filename, ptyname)
pi-klipper-1    | PermissionError: [Errno 13] Permission denied: '/dev/pts/1' -> 'printer_data/run/klipper.tty'

any idea how to fix?

mkuf, 2023-10-08:
Please share the output of the following commands on your host and share the output here. 

```
docker system info
```

```
docker inspect --format '{{ index .Config.Labels "org.prind.image.version"}}' mkuf/klipper:latest
```
```
ls -lRn /dev
```

Seredar, 2023-10-08:
```
pi@mars:~ $ docker system info
Client: Docker Engine - Community
 Version:    24.0.6
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.11.2
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.21.0
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 2
  Running: 0
  Paused: 0
  Stopped: 2
 Images: 4
 Server Version: 24.0.6
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 61f9fd88f79f081d64d6fa3bb1a0dc71ec870523
 runc version: v1.1.9-0-gccaecfc
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 6.1.21-v8+
 Operating System: Debian GNU/Linux 11 (bullseye)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 3.705GiB
 Name: mars
 ID: a65d0650-e4e6-43b3-a7d0-2eeb5ea77085
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No memory limit support
WARNING: No swap limit support
```
```

docker inspect --format '{{ index .Config.Labels "org.prind.image.version"}}' mkuf/klipper:latest
83ef0e135ef820f73c8579069c347953b9d6d56e
```

```
[... 516 lines cut ...]
crw-rw---- 1 0 29 116, 16 Oct  8 14:45 pcmC0D0p
crw-rw---- 1 0 29 116, 48 Oct  8 14:45 pcmC1D0p
crw-rw---- 1 0 29 116, 80 Oct  8 14:45 pcmC2D0p
crw-rw---- 1 0 29 116,  1 Oct  8 14:45 seq
crw-rw---- 1 0 29 116, 33 Oct  8 14:45 timer

/dev/snd/by-path:
total 0
lrwxrwxrwx 1 0 0 12 Oct  8 14:45 platform-bcm2835_audio -> ../controlC0
lrwxrwxrwx 1 0 0 12 Oct  8 14:45 platform-fef00700.hdmi -> ../controlC1
lrwxrwxrwx 1 0 0 12 Oct  8 14:45 platform-fef05700.hdmi -> ../controlC2

/dev/v4l:
total 0
drwxr-xr-x 2 0 0 160 Oct  8 14:45 by-path

/dev/v4l/by-path:
total 0
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-bcm2835-codec-video-index0 -> ../../video31
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-bcm2835-isp-video-index0 -> ../../video20
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-bcm2835-isp-video-index1 -> ../../video21
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-bcm2835-isp-video-index2 -> ../../video22
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-bcm2835-isp-video-index3 -> ../../video23
lrwxrwxrwx 1 0 0 13 Oct  8 14:45 platform-feb10000.codec-video-index0 -> ../../video19
```

mkuf, 2023-10-09:
I just noticed you're binding a local directory to `/opt/printer_data/run`, which causes the hosts permissions on that directory to be also used within the container.  

Make sure that your local `./run` directory is owned by uid 1000 or use a named docker volume.

Seredar, 2023-10-09:
yeap thanks fixed with `sudo chown 1000:1000 run folder`