mkuf/prind #99: Enhancement: Adding a slicer to the stack

fneddy, 2023-10-22:
Don’t know if it is desirable for the project but personally I’d love to have slicer in the stack and be able to upload stls from my phone directly

mkuf, 2023-10-23:
Do you have a specific slicer in mind?  
The most common ones like Cura, prusaSlicer and its derivatives are all requiring a desktop environment to run, making it impossible to use them via the web. 

What comes to mind is https://grid.space/, which hosts tools for editing meshes and generating gcodes for various CNC machines.  These tools are free to use online and would probably fulfill your use case. 

As these apps are free to use on the web, I don't really see a use case to host them locally via prind. 
Nevertheless, if there is enough demand for such a addition, Upstream may be open to provide ready to use docker images, so they can be added to the stack.

A word of caution - high demanding workload like slicers may interfere with the functionality of klipper when run on the same host and may cause interruptions/stutters while printing. 

-Markus

mkuf, 2023-11-03:
I did some testing and kirimoto as well as the mesh tool can not be served under a subpath because they require assets from the root of the webservice. This means they can not be served via the traefik instance shipped with prind. Also, I'd like to refrain from opening extra ports on the host, as this adds complexity for the user.  

Advanced Users may follow the gridspace docs on [Testing locally with docker](https://github.com/GridSpace/grid-apps#testing-locally-with-docker) to set up a separate compose stack for hosting kirimoto and meshtool, but for the time being this will not be added to prind. 

-Markus

closed 2023-11-03 by mkuf