mkuf/prind #100: Configuring smart plug with mainsail

jhugs, 2023-10-27:
### Description

Trying the set up a tplink kasa smart plug with the following in `moonraker.conf` from the mainsail docs + some additional options I've seen in other people's configs:

```
[power printer]
type: tplink_smartplug
address: ...
bound_services: klipper
off_when_shutdown_delay: 0
off_when_shutdown: True
locked_while_printing: True
```

I get an error saying the bound service clipper is unavailable, I'm assuming because it's running in the container. The plug still works from the drop down menu, but what does _not_ work is the custom power UI when the printer is shut down. Any help / guidance is appreciated.

mkuf, 2023-10-27:
Hi there, 

looking at the moonraker config docs, you'll definitely have to remove the `bound_services` key from your moonraker config. This will get rid of the error message about the service being unavailable.  
When running in a container, there is no way for moonraker to control any service.  

Can you give more details about the *custom power UI* you mentioned?  
Where is this configured and how do you access it? Is it specific to the used frontend?

-Markus

jhugs, 2023-10-27:
Thanks for the quick response!

Ah, alright that makes sense, I'll get rid of that.

The UI I'm talking about (that I'm not seeing in my UI, but would like to) can be seen [here](https://docs.mainsail.xyz/overview/quicktips/printer-power-switch)

mkuf, 2023-10-29:
Looking at the docs you linked, there does not seem to be a way to get the prompt when `bound_devices: klipper` is not set. And setting it will cause an error so that the prompt also is not displayed.  

From the hosting side, there is nothing that can be done here.   
Maybe there is a setting in mainsail that enables this prompt even if the service is not defined. The Mainsail maintainers/Support group may be able to help you with that.

closed 2023-10-29 by mkuf