mkuf/prind #106: No webcam in OrcaSlicer

padcom, 2023-12-29:
When running [OrcaSlicer](https://github.com/SoftFever/OrcaSlicer) on the `Device` tab, the webcam preview is just blank. Refer to [this](https://github.com/SoftFever/OrcaSlicer/issues/2927) issue for details.

I think it might have something to do with the `moonraker-obico.cfg` since it is the only place where the section `[webcam ..]` is defined.

mkuf, 2023-12-29:
Hi there, 

`moonraker-obico.cfg` is only used for the moonraker-obico service, not moonraker itself.  

I just downloaded orcaslicer to verify this behavior and can confirm that the webcam image is not visible in the device tab.  
Looking at the traefik logs when opening the frontend via orcaslicer does show a successful request to the webcam endpoint, which does indicate that this is an issue with the in-app browser of orcaslicer, as it works fine in Chrome and Firefox for me. 

```
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:50 +0000] "GET / HTTP/1.1" 200 2123 "-" "-" 73 "mainsail@docker" "http://172.18.0.7:80" 4ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:11:29 +0000] "GET /websocket HTTP/1.1" 0 0 "-" "-" 43 "moonraker@docker" "http://172.18.0.4:7125" 81537ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:11:29 +0000] "GET /websocket HTTP/1.1" 0 0 "-" "-" 44 "moonraker@docker" "http://172.18.0.4:7125" 81350ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:50 +0000] "GET /assets/index-f84b6612.js HTTP/1.1" 304 0 "-" "-" 74 "mainsail@docker" "http://172.18.0.7:80" 1ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:50 +0000] "GET /assets/index-756e3396.css HTTP/1.1" 304 0 "-" "-" 75 "mainsail@docker" "http://172.18.0.7:80" 2ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:50 +0000] "GET /assets/overlayscrollbars-a16bc3d3.css HTTP/1.1" 304 0 "-" "-" 76 "mainsail@docker" "http://172.18.0.7:80" 4ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:50 +0000] "GET /assets/vuetify-950d1cb0.css HTTP/1.1" 304 0 "-" "-" 77 "mainsail@docker" "http://172.18.0.7:80" 4ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:51 +0000] "GET /fonts/roboto-regular.woff2 HTTP/1.1" 304 0 "-" "-" 78 "mainsail@docker" "http://172.18.0.7:80" 2ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:53 +0000] "GET /fonts/roboto-light.woff2 HTTP/1.1" 304 0 "-" "-" 82 "mainsail@docker" "http://172.18.0.7:80" 3ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:53 +0000] "GET /fonts/roboto-medium.woff2 HTTP/1.1" 304 0 "-" "-" 81 "mainsail@docker" "http://172.18.0.7:80" 3ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:53 +0000] "GET /fonts/roboto-bold.woff2 HTTP/1.1" 304 0 "-" "-" 83 "mainsail@docker" "http://172.18.0.7:80" 3ms
prind-traefik-1  | 172.18.0.1 - - [29/Dec/2023:23:12:54 +0000] "GET /webcam/stream?action=stream HTTP/1.1" 200 248125 "-" "-" 84 "webcam@docker" "http://172.18.0.2:8080" 107ms
```

I don't think there is anything that can be done on the (docker/)hosting side to fix this, this requires fixes in the affected Applications themselves. 

-Markus

padcom, 2023-12-30:
Thanks for checking it out. I was also thinking this might be Orca-related. Do you mind if we keep this issue open until it is fixed? I will monitor the situation, and if the problem is fixed there, I'll make a reference and close this ticket.

padcom, 2023-12-30:
Alternatively, maybe you could suggest some other way to expose the webcam that would result in the image being shown in Orca?

mkuf, 2023-12-30:
Do you by any chance use the appimage of orcaslicer?  
I found this issue https://github.com/SoftFever/OrcaSlicer/issues/920 , which showed issues with the Device tab when using the appimage.  

My testing yesterday was done on Ubuntu w/ the appimage for 1.8.1, which confirmed the described behavior.  
Testing today with the windows portable in version 1.8.0 and 1.8.1, the webcam was working in the device tab.

padcom, 2023-12-30:
Indeed, I am using the AppImage of OrcaSlicer. I run Linux Mint on all my computers, windows is essentially banned in my shop xD.

padcom, 2023-12-30:
I do, however, own a mac mini and the webcam does work there no problem. So it is definitely not a problem with the config provided by your composition. Closing it.

Thanks for digging in into the problem!

closed 2023-12-30 by padcom

padcom, 2023-12-30:
Interesting fact: I just changed the service from MJPEG-Streamer to Adaptive MJPEG-Streamer and look:

[image]

I've done the change to limit FPS to 5 so that the stream runs smoother. The default 15 fps is saturating the printer's WiFi - so not even to fix the webcam issue :)

bwees, 2024-11-17:
I am still having this issue, even when using MJPEG-adaptive. WebRTC works but my host is not powerful enough to run it and stutters. Is there some debug mode I can go into to see why the WebView isn't able to load it?