mkuf/prind #108: klippain support

zackbcom, 2024-01-20:
I am wondering if anyone has setup [Klippain](https://github.com/Frix-x/klippain) in the docker setup?
The setup script looks like it might not work with the custom configs here.

mkuf, 2024-01-23:
I'm not aware of anyone using klippain with prind. 

I'm not familliar with klippain and the repository structure but the install script will certainly not work, as it is using hard coded paths for the config directories, that can not be changed. 

Something along the lines of this example override could be used to achieve that functionality. 
The init container of prind is utilized to clone klippain into a docker volume and may then replicate parts of the install script, mainly the symlinking of files between `/opt/klippain` and `/opt/prinder_data/config`.  
Furthermore, the image for klipper is built ad-hoc to add a symlink in the `klippy/extras` directory. 

This is just a proof of concept and is missing all of the symlinking commands.  
Furthermore, its not possible to run the init container again, as the script is not idempotent. 

```yaml
## Add your personal config here
services:
  init:
    image: alpine/git
    volumes:
      - klippain:/opt/klippain
      - ./config:/opt/printer_data/config
    entrypoint: ["ash", "-c"]
    command: 
      - |
        # Clone the repo
        git clone https://github.com/Frix-x/klippain /opt/klippain || true
        # create symlinks for directories between /opt/klippain and /opt/printer_data/config
        # do other linking, etc

  klipper:
    image: klipper:klippain
    build:
      dockerfile_inline: |
        FROM mkuf/klipper:latest
        USER root
        RUN ln -s /opt/klippain/scripts/gcode_shell_command.py /opt/klipper/klippy/extras
        USER klipper
    volumes:
      - /dev:/dev
      - ./config:/opt/printer_data/config
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - klippain:/opt/klippain

  moonraker:
    volumes:
      - /dev/null:/opt/klipper/config/null
      - /dev/null:/opt/klipper/docs/null
      - /run/dbus:/run/dbus
      - /run/systemd:/run/systemd
      - run:/opt/printer_data/run
      - gcode:/opt/printer_data/gcodes
      - log:/opt/printer_data/logs
      - moonraker-db:/opt/printer_data/database
      - ./config:/opt/printer_data/config
      - klippain:/opt/klippain

volumes:
  klippain:
```

mkuf, 2024-02-08:
I'll close this issue for now, as demand for klippain support was very low.  
If more users request support, I'd be willing to spend some time on implementing this.  
I'm also happy to accept Pull Requests to add this functionality. 

-Markus

closed 2024-02-08 by mkuf