mkuf/prind #113: TIMELAPSE not working due to local webcam url not being accessable inside moonraker container (probably)

raphaeleduardo42, 2024-02-22:
I've been trying to make timelapse to work for a couple of days.
Gathering more information, I'd maybe found why but my lack of knowledge stop me trying to fix it.

The first error message I recieved was, when trying to render:
```bash
2024-02-22 18:29:11,012 [timelapse.py:render()] - Rendering Video failed: /usr/bin/ffmpeg -r 5 -i '/tmp/timelapse/frame%6d.jpg' -threads 2 -g 5 -crf 23 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/timelapse__20240222_1829.mp4' -y : Output file #0 does not contain any stream
```

Going trough, it showed me that the TIMELAPSE_TAKE_FRAME gcode lead to this erro at the logs:

```bash
2024-02-22 18:29:07,009 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost/webcam?action=snapshot -O /tmp/timelapse/frame000001.jpg
```

The error could be replicated when trying to get frame inside the container:

```bash
$ docker exec -it prind-moonraker-1 wget http://localhost/webcam?action=snapshot -O /tmp/timelapse/frame000001.jpg
--2024-02-22 18:37:04--  http://localhost/webcam?action=snapshot
Resolving localhost (localhost)... 127.0.0.1, ::1
Connecting to localhost (localhost)|127.0.0.1|:80... failed: Connection refused.
Connecting to localhost (localhost)|::1|:80... failed: Cannot assign requested address.
Retrying.

--2024-02-22 18:37:05--  (try: 2)  http://localhost/webcam?action=snapshot
Connecting to localhost (localhost)|127.0.0.1|:80... failed: Connection refused.
Connecting to localhost (localhost)|::1|:80... failed: Cannot assign requested address.
Retrying.

--2024-02-22 18:37:07--  (try: 3)  http://localhost/webcam?action=snapshot
Connecting to localhost (localhost)|127.0.0.1|:80... failed: Connection refused.
Connecting to localhost (localhost)|::1|:80... failed: Cannot assign requested address.
Retrying.
```

Even though the webcam is working, and accessable outside the container.
When trying to access using the machine IP it worked flawless, so maybe this is related on how the containers are interacting with themselves.
```bash
$ docker exec -it prind-moonraker-1 wget http://192.168.0.156/webcam?action=snapshot -O /tmp/timelapse/frame000001.jpg
--2024-02-22 18:41:39--  http://192.168.0.156/webcam?action=snapshot
Connecting to 192.168.0.156:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 35136 (34K) [image/jpeg]
Saving to: ‘/tmp/timelapse/frame000001.jpg’

/tmp/timelapse/frame000001.jp 100%[=================================================>]  34.31K  --.-KB/s    in 0.005s

2024-02-22 18:41:39 (6.18 MB/s) - ‘/tmp/timelapse/frame000001.jpg’ saved [35136/35136]
```

When trying to acess the webcam inside the moonraker container trough curl it does work. Also using `http://webcam:8080/snapshot` as snapshot url from webcam on mainsail make work the `TIMELAPSE_TAKE_FRAME` but also vanish the stream.

mkuf, 2024-02-22:
Hi there, 

Containers in the stack can reach each other by their name, so you'll have to set the snapshot url in `moonraker.conf` to `http://webcam:8080/snapshot`.   
The webcam URLs in mainsail do not need to be changed. 

-Markus

raphaeleduardo42, 2024-02-22:
Just worked @mkuf really thx

closed 2024-02-22 by raphaeleduardo42